shmuelzon/esp32-ble2mqtt #1: Unable to compile

dvzkul, 2018-03-20:
I am very interested in your code, but am a novice in C (I am a .NET developer).  When I run make, it appears to compile fine until ble_utils.o

In function 'ble_atotype':

/home/ubuntu/projects/esp32-ble2mqtt/main/./ble_utils.c:178:22: error: 'CHAR_TYPE_BOOLEAN' undeclared (first use in this function)
         { "boolean", CHAR_TYPE_BOOLEAN },

There is a similar error for each item in the struct.

shmuelzon, 2018-03-20:
Thanks for pointing this out to me. I wonder how long this has been going on.
It seems that the Bluetooth SIG has changed the way their site works which broke my GATT scraper.
Until I have a solution in place, you can use the attached files (extract them to the main/ directory) and build as usual.

[gatt.zip](https://github.com/shmuelzon/esp32-ble2mqtt/files/1830762/gatt.zip)

dvzkul, 2018-03-20:
Thank you!  That did the trick.  I did need to comment out everything in main/component.mk, otherwise it was being overwritten.

commit 24a3e391 (2018-03-21): Use cookies when scraping bluetooth.org. Fixes #1
closed 2018-03-21 by shmuelzon