shmuelzon/esp32-ble2mqtt #2: Indication and notification support

DurandA, 2018-04-10:
Is it possible to use indications and/or notifications with the current release?

Your project is awesome. This is a nice open-source alternative to the [Mongoose BLE-to-WiFi gateway](https://mongoose-os.com/ble-wifi-gateway/).

shmuelzon, 2018-04-10:
Thanks!

If the characteristic supports notifications, the app will [register for notifications](https://github.com/shmuelzon/esp32-ble2mqtt/blob/f478b2dbb7647e6f4b5a1b6faf3b3d427983c932/main/ble2mqtt.c#L282-L286) and an MQTT topic should be published with the new data.

Haven't tried/tested indications.

shmuelzon, 2018-04-11:
Are you having any issues with receiving notifications or is this just a general question?
If it's the former, could you provide more details? The device you're using, logs, etc.
Otherwise, can I close this issue?

DurandA, 2018-04-11:
Thank you very much for the quick answer! It was a general question.

closed 2018-04-11 by DurandA

bilogic, 2019-04-22:
Hi,

I can confirm that indications are not being received from my mi_scale.

serviceUUID is 0x181D
characteristicsUUID is 0x2A9D

If it helps, I have working code for the platformIO Arduino framework.

shmuelzon, 2019-04-22:
Hey,

To be honest I don’t have and devices that make use of indications, only notifications, so I’ve never really tested indications. Anything you can provide (serial log with debug level, code examples, etc.) would be great so I can try to figure out what’s missing.

Thanks!

bilogic, 2019-04-22:
Ok, on closer look, some of the services are not detected by the current code.
This is the full logs
```
I (94468) BLE2MQTT: Discovered BLE device: c8:0f:10:c7:6f:e1, connecting
I (98248) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
I (98878) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (98878) BLE2MQTT: Got characterisic: 00002a00-0000-1000-8000-00805f9b34fb (8)
I (98888) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (98888) BLE2MQTT: Copying payload to: 0x3ffeb7bc
I (98898) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (98898) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (98908) BLE2MQTT: Parsing characterisic: 00002a00-0000-1000-8000-00805f9b34fb
I (98918) BLE2MQTT: Payload at 0x3ffeb7bc (8)
I (98918) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (98928) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/DeviceName = MI_SCALE
I (98938) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (98938) BLE2MQTT: Got characterisic: 00002a01-0000-1000-8000-00805f9b34fb (2)
I (98948) BLE2MQTT: 80 0c
I (98948) BLE2MQTT: Copying payload to: 0x3ffecac0
I (98958) BLE2MQTT: 80 0c
I (98958) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (98968) BLE2MQTT: Parsing characterisic: 00002a01-0000-1000-8000-00805f9b34fb
I (98978) BLE2MQTT: Payload at 0x3ffecac0 (2)
I (98978) BLE2MQTT: 80 0c
I (98988) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/Appearance = 3200
I (99028) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (99028) BLE2MQTT: Got characterisic: 00002a02-0000-1000-8000-00805f9b34fb (1)
I (99038) BLE2MQTT: 00
I (99038) BLE2MQTT: Copying payload to: 0x3ffecac0
I (99038) BLE2MQTT: 00
I (99048) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (99058) BLE2MQTT: Parsing characterisic: 00002a02-0000-1000-8000-00805f9b34fb
I (99058) BLE2MQTT: Payload at 0x3ffecac0 (1)
I (99068) BLE2MQTT: 00
I (99068) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPrivacyFlag = false
I (99088) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (99088) BLE2MQTT: Got characterisic: 00002a04-0000-1000-8000-00805f9b34fb (8)
I (99098) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (99098) BLE2MQTT: Copying payload to: 0x3ffeb7bc
I (99108) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (99108) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (99118) BLE2MQTT: Parsing characterisic: 00002a04-0000-1000-8000-00805f9b34fb
I (99128) BLE2MQTT: Payload at 0x3ffeb7bc (8)
I (99128) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (99138) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPreferredConnectionParameters = 100,200,0,2000
I (99148) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (99158) BLE2MQTT: Got characterisic: 00002a03-0000-1000-8000-00805f9b34fb (0)
I (99168) BLE2MQTT: Copying payload to: 0x0
I (99168) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (99178) BLE2MQTT: Parsing characterisic: 00002a03-0000-1000-8000-00805f9b34fb
I (99188) BLE2MQTT: Payload at 0x0 (0)
I (99188) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/ReconnectionAddress =
I (99258) BLE2MQTT: Got service: 00001801-0000-1000-8000-00805f9b34fb
I (99258) BLE2MQTT: Got characterisic: 00002a05-0000-1000-8000-00805f9b34fb (4)
I (99258) BLE2MQTT: 01 00 ff ff
I (99258) BLE2MQTT: Copying payload to: 0x3ffecac0
I (99268) BLE2MQTT: 01 00 ff ff
I (99278) BLE2MQTT: Parsing service: 00001801-0000-1000-8000-00805f9b34fb
I (99278) BLE2MQTT: Parsing characterisic: 00002a05-0000-1000-8000-00805f9b34fb
I (99288) BLE2MQTT: Payload at 0x3ffecac0 (4)
I (99288) BLE2MQTT: 01 00 ff ff
I (99298) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAttribute/ServiceChanged = 1,65535
```

bilogic, 2019-04-22:
I have found the location to insert the indication codes, however, the bigger issue now is, why doesn't characteristic 0x181D show up?

shmuelzon, 2019-04-22:
Hard to tell from here :)
Have you tried to connect from you mobile phone and view the available services?
I, personally, use the LightBlue app. If your phone is already paired with the scale, try to remove it so it will start with a clean slate as the ESP32 will be.

bilogic, 2019-04-22:
Yes, I use nRF, I also have working code for platformIO Arduino framework. The reason I'm looking at your repo is to skip re-writing the MQTT part.

I enabled DEBUG level logs and 0x181D showed up, apparently there are no INFO lines for notifications.

shmuelzon, 2019-04-22:
Does it show up only as a part of the `Found new characteristic!` log?
If it’s readable, it should still be published after a successful read, regardless of the indications. If it’s not readable, writeable nor supports notifications then it will be ignored, I need to add support for indications once I figure out what I need to do with them.

bilogic, 2019-04-22:
You can test whether to send Indications inside ```ble_characteristic_notify_register()``` with ```if (characteristic->properties & CHAR_PROP_INDICATE)```
If true, just set ```notify_en = 2```

I added INFO lines for Indications and Notifications registration

Can you make sense of this? All seems fine, except that I don't get any callbacks when I step on the scale.
```
D (123072) BLE2MQTT: Found new characteristic!
D (123072) MQTT_CLIENT: Subscribe successful
D (123082) BLE2MQTT:   Service: 0000181d-0000-1000-8000-00805f9b34fb
D (123092) BLE2MQTT:   Characteristic: 00002a9d-0000-1000-8000-00805f9b34fb
I (123102) BLE: Indication registration to: 0000181d-0000-1000-8000-00805f9b34fb
D (123102) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
I (123102) BLE: Indication registration to: 00002a9d-0000-1000-8000-00805f9b34fb
D (123122) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a9d-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffec33c
D (123132) BLE2MQTT: Found new characteristic!
D (123142) BLE2MQTT:   Service: 0000181d-0000-1000-8000-00805f9b34fb
D (123142) BLE2MQTT:   Characteristic: 00002a2f-0000-3512-2118-0009af100700
D (123152) MQTT: Subscribing to c8:0f:10:c7:6f:e1/WeightScale/00002a2f-0000-3512-2118-0009af100700/Set
D (123162) MQTT_CLIENT: mqtt_enqueue id: 63194, type=8 successful
D (123172) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/WeightScale/00002a2f-0000-3512-2118-0009af100700/Set, id: 35979, type=8 successful
D (123182) MQTT_CLIENT: msg_type=9, msg_id=35979
D (123182) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
I (123182) BLE: Notification registration to: 0000181d-0000-1000-8000-00805f9b34fb
D (123182) MQTT_CLIENT: pending_id=35979, pending_msg_count = 1
D (123212) MQTT_CLIENT: Subscribe successful
I (123212) BLE: Notification registration to: 00002a2f-0000-3512-2118-0009af100700
D (123222) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a2f-0000-3512-2118-0009af100700, len: 2, val: 0x3ffec3d8
D (123232) BLE2MQTT: Found new characteristic!
D (123232) BLE2MQTT:   Service: 00001530-0000-3512-2118-0009af100700
D (123242) BLE2MQTT:   Characteristic: 00001531-0000-3512-2118-0009af100700
D (123252) MQTT: Subscribing to c8:0f:10:c7:6f:e1/00001530-0000-3512-2118-0009af100700/00001531-0000-3512-2118-0009af100700/Set
D (123262) MQTT_CLIENT: mqtt_enqueue id: 35979, type=8 successful
D (123272) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/00001530-0000-3512-2118-0009af100700/00001531-0000-3512-2118-0009af100700/Set, id: 62353, type=8 successful
D (123272) MQTT_CLIENT: msg_type=9, msg_id=62353
D (123292) MQTT_CLIENT: pending_id=62353, pending_msg_count = 1
D (123292) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
```

shmuelzon, 2019-04-22:
Why is there an indication registration on the service uuid? Shouldn’t be only for characteristics? Also, indications require the master to acknowledge the indication, do you know if this is this something the ESP’s BLE library handles by itself or if we should do that manually?
Did it work as expected with your Implementation?

bilogic, 2019-04-22:
1. The registration line is just the log to show me what service UUID was received inside ```ble_characteristic_notify_register()```, ```ble_operation_enqueue()``` sends only characteristics.
2. In my working code derived from https://github.com/wakwak-koba/ESP32_BLE_Arduino, I could not find any acknowledgement.
3. Also could not find any ESP-IDF function documention for acknowledgement, so I'm assuming the ESP-IDF functions are taking care of it.
4. My working code is able to register for indication and receive callbacks

bilogic, 2019-04-22:
Can you point out to me the code that receives notification callbacks?

bilogic, 2019-04-22:
I found ```case ESP_GATTC_NOTIFY_EVT:``` inside ```ble.c```

shmuelzon, 2019-04-22:
Yes, that would be the place...
I've create a [various-fixes](https://github.com/shmuelzon/esp32-ble2mqtt/tree/various-fixes) branch for fixing both the zero-length characteristic values and also added support for indications.
Would you mind trying it out?
I also took a look at the repo you linked to and couldn't find any special handling for indications as well, so it should work, in theory.

shmuelzon, 2019-04-22:
BTW - Is the characteristic you're interested in readable or only supports indications?
(I've also added a debug print to show the characteristic properties)

bilogic, 2019-04-22:
> BTW - Is the characteristic you're interested in readable or only supports indications?
nRF says INDICATE only.

shmuelzon, 2019-04-22:
OK, that would, at least, explain why it wasn’t seen at all.
Now, if only indications would work... :)

bilogic, 2019-04-22:
After I did a whitelist on the service and characteristics on the main branch, I'm getting INDICATIONS via ```ESP_GATTC_NOTIFY_EVT``` now. Maybe reading some attribute is messing with the scale indications.

bilogic, 2019-04-22:
I got a ```topic: "c8:0f:10:c7:6f:e1/Connected"``` on my nodered MQTT logs.
However, even though I'm also seeing the following, nothing pops up on nodered
```
I (29732) BLE: Getting ESP_GATTC_NOTIFY_EVT
I (29732) BLE2MQTT: Got service: 0000181d-0000-1000-8000-00805f9b34fb
I (29742) BLE2MQTT: Got characterisic: 00002a9d-0000-1000-8000-00805f9b34fb (10)
I (29752) BLE2MQTT: a2 d4 35 b2 08 01 03 00 16 31
I (29752) BLE2MQTT: Copying payload to: 0x3ffec684
I (29762) BLE2MQTT: a2 d4 35 b2 08 01 03 00 16 31
I (29762) BLE2MQTT: Parsing service: 0000181d-0000-1000-8000-00805f9b34fb
I (29772) BLE2MQTT: Parsing characterisic: 00002a9d-0000-1000-8000-00805f9b34fb
I (29782) BLE2MQTT: Payload at 0x3ffec684 (10)
I (29782) BLE2MQTT: a2 d4 35 b2 08 01 03 00 16 31
I (29792) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/WeightScale/WeightMeasurement = 162,13780,2226,1,3,12566,16382
```

bilogic, 2019-04-22:
Ok, here's the conclusion:

1. Whitelisting just serviceUUID 0x181D and characteristicsUUID 0x2A9D fixed the ```ESP_GATTC_NOTIFY_EVT``` not being raised issue
2. ```ESP_GATTC_NOTIFY_EVT``` is raising INDICATIONS, all our earlier discussions are valid
3. node-RED issues was due to my config
4. I will find some time to test the ```various-fixes``` in the next few days

Thank you!

bilogic, 2019-04-22:
Have you tried registering for multiple callbacks before?
I have a BC85 blood pressure monitor, it's callbacks are not working as well. 

It seems that when BLE2MQTT registers for both the scale and BP monitor, callbacks don't work, i.e. don't see any ```ESP_GATTC_NOTIFY_EVT```
It was very similar to when I had to whitelist only 1 callback for the scale.

Will continue again another time.

bilogic, 2019-04-22:
Forgot to add that I have working code for the BC85 as well, so that rules out any device issues. I'm currently inclined to think issue lies inside BLE2MQTT.

shmuelzon, 2019-04-22:
The [device I use](https://github.com/shmuelzon/ble-peripherals) all use notifications (not indications) and I have no issues with receiving notifications from multiple characteristics and multiple devices on the same ESP32 module.

Technically speaking, I'm only registering a single callback function and only enabling notifications for specific devices/characteristics. I don't see any errors from writing to the client config characteristic so it should work.
I will add some more debug and error messages so it will be easier to trace in the future.

BTW - which ESP-IDF version are you using?

bilogic, 2019-04-22:
I'm on the latest ESP-IDF (from just days ago)
Is there a way I can tell exactly which version?

bilogic, 2019-04-22:
Just curious, how is the value of ```BLE_OPERATION_TYPE_WRITE_CHAR``` in ```ble.c``` assigned?
From my working code, it needs to be 0x2902

shmuelzon, 2019-04-23:
Could you please switch to ESP-IDF v3.2 and try again? That’s the (latest) release I’m using.

bilogic, 2019-04-28:
@shmuelzon I tried v3.2, same result. 
I changed ```enqueue``` and ```dequeue``` from DEBUG to WARNING and set loglevel to INFO
I'm targeting registration for characteristic.
Apparently, there is an enqueue for type 0x2 ```BLE_OPERATION_TYPE_WRITE_CHAR```, but no matching dequeue. 

How do I lengthen the queue timer or something? Thank you.

```
...
I (2675) BLE2MQTT: Connected to WiFi, connecting to MQTT
I (2675) MQTT: Connecting MQTT client
I (2675) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2695) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (2705) MQTT: MQTT client connected
I (2705) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (2725) BLE2MQTT: Discovered BLE device: 3f:e8:5d:63:23:b2, not connecting
I (3025) BLE2MQTT: Discovered BLE device: f2:d7:58:34:22:40, not connecting
I (3155) BLE2MQTT: Discovered BLE device: c8:0f:10:c7:6f:e1, connecting
I (7965) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
W (7975) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (7975) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (7985) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a02-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8005) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8015) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a03-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8025) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (8035) BLE: Indication registration for characteristic: 00002a05-0000-1000-8000-00805f9b34fb
W (8045) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbb6d0
W (8055) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a25-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8065) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a28-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8085) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a23-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8095) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a50-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8105) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a2b-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (8115) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a9e-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (8125) BLE: Indication registration for characteristic: 00002a9d-0000-1000-8000-00805f9b34fb
W (8135) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a9d-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbaa44
I (8145) BLE: Notification registration for characteristic: 00002a2f-0000-3512-2118-0009af100700
W (8155) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a2f-0000-3512-2118-0009af100700, len: 2, val: 0x3ffbaa4c
I (8165) BLE: Notification registration for characteristic: 00001531-0000-3512-2118-0009af100700
W (8175) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00001531-0000-3512-2118-0009af100700, len: 2, val: 0x3ffd3b20
W (8195) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (8205) BLE: Notification registration for characteristic: 00002a04-0000-1000-8000-00805f9b34fb
W (8205) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffd3b48
W (8225) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00001542-0000-3512-2118-0009af100700, len: 0, val: 0x0
I (8235) BLE: Notification registration for characteristic: 00001542-0000-3512-2118-0009af100700
W (8245) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00001542-0000-3512-2118-0009af100700, len: 2, val: 0x3ffbaa20
W (8255) BLE: Enqueue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00001543-0000-3512-2118-0009af100700, len: 0, val: 0x0
I (8265) BLE: Notification registration for characteristic: 00001543-0000-3512-2118-0009af100700
W (8275) BLE: Enqueue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00001543-0000-3512-2118-0009af100700, len: 2, val: 0x3ffd3af8
W (8785) BLE: Dequeue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (8815) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (8825) BLE2MQTT: Got characterisic: 00002a00-0000-1000-8000-00805f9b34fb (8)
I (8825) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (8825) BLE2MQTT: Copying payload to: 0x3ffd3b00
I (8835) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
W (8835) BLE: Dequeue: type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (8835) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (8855) BLE2MQTT: Parsing characterisic: 00002a00-0000-1000-8000-00805f9b34fb
I (8865) BLE2MQTT: Payload at 0x3ffd3b00 (8)
I (8875) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (8875) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/DeviceName = MI_SCALE
I (8875) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (8895) BLE2MQTT: Got characterisic: 00002a01-0000-1000-8000-00805f9b34fb (2)
I (8895) BLE2MQTT: 80 0c
I (8905) BLE2MQTT: Copying payload to: 0x3ffd3b00
I (8905) BLE2MQTT: 80 0c
[... cut ...]
W (9235) BLE: Dequeue: type: 2, device: c8:0f:10:c7:6f:e1, char: 00001801-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbb6d0
I (9235) BLE2MQTT: Parsing service: 00001801-0000-1000-8000-00805f9b34fb
I (9245) BLE: BLE_OPERATION_TYPE_WRITE_CHAR: 0xf
I (9245) BLE2MQTT: Parsing characterisic: 00002a05-0000-1000-8000-00805f9b34fb
I (9265) BLE2MQTT: Payload at 0x3ffd3b00 (4)
I (9265) BLE2MQTT: 01 00 ff ff
I (9275) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAttribute/ServiceChanged = 1,65535
I (9815) BLE2MQTT: Discovered BLE device: a0:18:28:f3:d6:4f, not connecting
...
```

bilogic, 2019-04-28:
Hi,

I added an ID to each ```operation``` that is queued. The problem is clear now.

1. 22 operations queued, only 7 dequeued aka performed
2. After the 7th operation, ```esp32-ble2mqtt``` becomes idle until mi_scale disconnects (usually about 30 secs) and the whole cycle starts again.
4. The setup never gets configured properly as there are operations which were not performed.
5. What is the logic for dequeuing with ```need_dequeue``` in ```esp_gattc_cb()```? 
6. Shouldn't it be based on whether there are queued operations rather than incoming events?

Thank you.

```
I (2640) BLE2MQTT: Connected to WiFi, connecting to MQTT
I (2640) MQTT: Connecting MQTT client
I (2640) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2670) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (2680) MQTT: MQTT client connected
I (2680) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (2900) BLE2MQTT: Discovered BLE device: f2:d7:58:34:22:40, not connecting
I (3400) BLE2MQTT: Discovered BLE device: a0:18:28:f3:d6:4f, not connecting
I (5020) BLE2MQTT: Discovered BLE device: 38:b0:26:5e:0e:4a, not connecting
I (15880) BLE2MQTT: Discovered BLE device: c8:0f:10:c7:6f:e1, connecting
I (20600) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
W (20610) BLE: Enqueue: id: 1, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20620) BLE: Enqueue: id: 2, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20630) BLE: Enqueue: id: 3, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a02-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20640) BLE: Enqueue: id: 4, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20660) BLE: Enqueue: id: 5, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a03-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20670) BLE: Enqueue: id: 6, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (20680) BLE: Indication registration for characteristic: 00002a05-0000-1000-8000-00805f9b34fb
W (20690) BLE: Enqueue: id: 7, type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbaa04
W (20700) BLE: Enqueue: id: 8, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a25-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20710) BLE: Enqueue: id: 9, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a28-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20730) BLE: Enqueue: id: 10, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a23-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20740) BLE: Enqueue: id: 11, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a50-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20750) BLE: Enqueue: id: 12, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a2b-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (20760) BLE: Enqueue: id: 13, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a9e-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (20770) BLE: Indication registration for characteristic: 00002a9d-0000-1000-8000-00805f9b34fb
W (20780) BLE: Enqueue: id: 14, type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a9d-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbb6e0
I (20800) BLE: Notification registration for characteristic: 00002a2f-0000-3512-2118-0009af100700
W (20800) BLE: Enqueue: id: 15, type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a2f-0000-3512-2118-0009af100700, len: 2, val: 0x3ffba5cc
I (20820) BLE: Notification registration for characteristic: 00001531-0000-3512-2118-0009af100700
W (20830) BLE: Enqueue: id: 16, type: 2, device: c8:0f:10:c7:6f:e1, char: 00001531-0000-3512-2118-0009af100700, len: 2, val: 0x3ffba5d4
W (20840) BLE: Enqueue: id: 17, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (20850) BLE: Notification registration for characteristic: 00002a04-0000-1000-8000-00805f9b34fb
W (20860) BLE: Enqueue: id: 18, type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffba9b4
W (20880) BLE: Enqueue: id: 19, type: 0, device: c8:0f:10:c7:6f:e1, char: 00001542-0000-3512-2118-0009af100700, len: 0, val: 0x0
I (20890) BLE: Notification registration for characteristic: 00001542-0000-3512-2118-0009af100700
W (20890) BLE: Enqueue: id: 20, type: 2, device: c8:0f:10:c7:6f:e1, char: 00001542-0000-3512-2118-0009af100700, len: 2, val: 0x3ffba9bc
W (20910) BLE: Enqueue: id: 21, type: 0, device: c8:0f:10:c7:6f:e1, char: 00001543-0000-3512-2118-0009af100700, len: 0, val: 0x0
I (20920) BLE: Notification registration for characteristic: 00001543-0000-3512-2118-0009af100700
W (20930) BLE: Enqueue: id: 22, type: 2, device: c8:0f:10:c7:6f:e1, char: 00001543-0000-3512-2118-0009af100700, len: 2, val: 0x3ffba9c4
W (21440) BLE: Queue timer expired
W (21440) BLE: Dequeue: id: 1, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (21500) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (21500) BLE2MQTT: Got characterisic: 00002a00-0000-1000-8000-00805f9b34fb (8)
I (21510) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (21510) BLE2MQTT: Copying payload to: 0x3ffec688
I (21520) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
W (21520) BLE: Dequeue: id: 2, type: 0, device: c8:0f:10:c7:6f:e1, char: 00001800-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
[... cut ...]
I (21960) BLE2MQTT: Parsing characterisic: 00002a05-0000-1000-8000-00805f9b34fb
I (21970) BLE2MQTT: Payload at 0x3ffba9f4 (4)
I (21980) BLE2MQTT: 01 00 ff ff
I (21980) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAttribute/ServiceChanged = 1,65535
I (25030) BLE2MQTT: Discovered BLE device: 63:53:6a:5e:8b:85, not connecting
W (48980) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
I (48980) BLE: Connection closed, reason = 0x13
I (48980) BLE2MQTT: Disconnected from device: c8:0f:10:c7:6f:e1
```

bilogic, 2019-04-28:
More issues:

1. characteristics does not always match, e.g. ID 13, queued as 0x2a35, but dequeued as 0x180a

```
D (254340) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (254360) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (254470) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (254530) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (254570) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (255260) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (255990) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (256170) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (256280) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (256290) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (256990) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257090) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257400) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257510) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257640) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257720) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257830) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (257930) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (258140) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (258350) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (258460) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (258850) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (259370) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (259470) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (260390) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (261420) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (261430) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (261840) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262210) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262420) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262440) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262520) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262620) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (262640) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (262640) BLE2MQTT: Discovered BLE device: 28:fd:80:07:b5:25, connecting
D (262650) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
D (262880) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1
D (262880) BLE: GATTC event 40 wasn't handled
D (262880) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
D (262890) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (262940) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263140) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263150) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263260) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263370) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263470) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263580) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263680) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (263880) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (264090) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (264100) BLE: Received GATTC event 18 (ESP_GATTC_CFG_MTU_EVT), gattc_if 1
I (264100) BLE2MQTT: Connected to device: 28:fd:80:07:b5:25, scanning
D (264100) MQTT: Subscribing to 28:fd:80:07:b5:25/Connected
D (264100) MQTT_CLIENT: mqtt_enqueue id: 19111, type=8 successful
D (264110) MQTT_CLIENT: Sent subscribe topic=28:fd:80:07:b5:25/Connected, id: 29165, type=8 successful
D (264120) MQTT_CLIENT: msg_type=9, msg_id=29165
D (264120) MQTT_CLIENT: msg_type=9, msg_id=29165
D (264130) MQTT_CLIENT: pending_id=29165, pending_msg_count = 1
D (264140) MQTT_CLIENT: Subscribe successful
D (264140) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (264150) BLE: GATTC event 7 wasn't handled
D (264150) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (264160) BLE: GATTC event 7 wasn't handled
D (264160) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
[... 315 lines cut ...]
D (268680) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (268780) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (268890) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (269010) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
D (269020) BLE: Received GATTC event 41 (ESP_GATTC_DISCONNECT_EVT), gattc_if 1
D (269020) BLE: GATTC event 41 wasn't handled
D (269020) BLE: Received GATTC event 5 (ESP_GATTC_CLOSE_EVT), gattc_if 1
I (269030) BLE: Connection closed, reason = 0x13
I (269030) BLE2MQTT: Disconnected from device: 28:fd:80:07:b5:25
D (269040) MQTT: Unsubscribing from 28:fd:80:07:b5:25/Connected
D (269050) MQTT_CLIENT: mqtt_enqueue id: 26795, type=8 successful
D (269050) MQTT_CLIENT: unsubscribe, topic"28:fd:80:07:b5:25/Connected", id: 44575
D (269060) MQTT_CLIENT: Sent Unsubscribe topic=28:fd:80:07:b5:25/Connected, id: 44575, successful
D (269070) MQTT_CLIENT: msg_type=11, msg_id=44575
D (269080) MQTT_CLIENT: msg_type=11, msg_id=44575
[... cut ...]
D (269090) MQTT_CLIENT: UnSubscribe successful
D (269090) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269190) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269200) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269310) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269410) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269510) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (269620) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
```

shmuelzon, 2019-04-28:
Hey,

> After the 7th operation, esp32-ble2mqtt becomes idle until mi_scale disconnects (usually about 30 secs) and the whole cycle starts again.

True, and it fails right after writing to the client configuration descriptor for enabling indications.

> What is the logic for dequeuing with need_dequeue in esp_gattc_cb()?
> Shouldn't it be based on whether there are queued operations rather than incoming events?

Yes and no. I only queue read and write operations. Once they are completed, I dequeue the next one. However, not all events received are relevant, for example scanning events, received notifications/indications, etc. Only after a read or write event has completed, I want to dequeue the next operation. After the `need_dequeue` flag is set, I check if there is anything to dequeue.

With the current log, I see two options:
1. The write operation to enable indications succeeds and:
    1. No event is sent to the app so it doesn't know it can dequeue the next operation
    2. An event that's not the `ESP_GATTC_WRITE_DESCR_EVT` it expects is send and is ignored
2. The write operation fails and/or hangs. Maybe the scale closes the connection and it takes a while to the ESP to detect that or there is no confirmation for the write event

If I may suggest the following:
1. Remove the previous logs added for debugging the crash that was already solved in `main/ble2mqtt.c`. They just add clutter
2. Raise the `Received GATTC event` and `GATTC event %d wasn't handled` log levels in `main/ble.c:esp_gattc_cb()` so we can see which events are received and maybe understand what happens after enabling indications

BTW - I want to thank you for your work and helping me debug this. It's not obvious and I appreciate it

Good day

shmuelzon, 2019-04-28:
> characteristics does not always match, e.g. ID 13, queued as 0x2a35, but dequeued as 0x180a

I think this is related to the fact that the main application thread also writes UUIDs to the log and it messes things up (a static buffer when converting UUIDs to string), that's why I suggested to remove those logs

bilogic, 2019-04-28:
It takes 2 hands to clap. I'm sharing all this info because you are responding and helping promptly. :)
I'm going to revert to your master branch and do the following:

> 2. Raise the `Received GATTC event` and `GATTC event %d wasn't handled` log levels in `main/ble.c:esp_gattc_cb()` so we can see which events are received and maybe understand what happens after enabling indications

bilogic, 2019-04-28:
Ok, I did not revert, otherwise had to rewrite my ID codes. Just removed the earlier codes in ```ble.c```. Here are the logs, it seems no further events received.

```
D (8920) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (9020) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (9230) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (9330) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (9540) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (9620) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (9620) BLE2MQTT: Discovered BLE device: c8:0f:10:c7:6f:e1, connecting
D (9630) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
W (13780) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1
W (13780) BLE: GATTC event 40 wasn't handled
W (13780) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
D (13790) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (13930) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (14140) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (14150) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (14560) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (14870) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (15350) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (15630) BLE: Received GATTC event 18 (ESP_GATTC_CFG_MTU_EVT), gattc_if 1
I (15640) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
D (15640) MQTT: Subscribing to c8:0f:10:c7:6f:e1/Connected
D (15640) MQTT_CLIENT: mqtt_enqueue id: 56922, type=8 successful
D (15650) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/Connected, id: 1773, type=8 successful
W (15660) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (15670) BLE: GATTC event 7 wasn't handled
W (15670) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (15680) BLE: GATTC event 7 wasn't handled
W (15680) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (15690) BLE: GATTC event 7 wasn't handled
W (15700) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (15700) MQTT_CLIENT: msg_type=9, msg_id=1773
W (15700) BLE: GATTC event 7 wasn't handled
D (15710) MQTT_CLIENT: msg_type=9, msg_id=1773
W (15710) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (15720) MQTT_CLIENT: pending_id=1773, pending_msg_count = 1
W (15730) BLE: GATTC event 7 wasn't handled
D (15730) MQTT_CLIENT: Subscribe successful
W (15740) BLE: Received GATTC event 6 (ESP_GATTC_SEARCH_CMPL_EVT), gattc_if 1
D (15750) BLE2MQTT: Services discovered on device: c8:0f:10:c7:6f:e1
D (15760) BLE2MQTT: Found new characteristic!
D (15760) BLE2MQTT:   Service: 00001800-0000-1000-8000-00805f9b34fb
D (15770) BLE2MQTT:   Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
D (15770) MQTT: Subscribing to c8:0f:10:c7:6f:e1/GenericAccess/DeviceName/Get
D (15780) MQTT_CLIENT: mqtt_enqueue id: 1773, type=8 successful
D (15790) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/GenericAccess/DeviceName/Get, id: 55009, type=8 successful
W (15800) BLE: Enqueue: id: 1, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
D (15800) MQTT_CLIENT: msg_type=9, msg_id=55009
D (15810) MQTT_CLIENT: msg_type=9, msg_id=55009
D (15820) MQTT_CLIENT: pending_id=55009, pending_msg_count = 1
D (15820) MQTT: Subscribing to c8:0f:10:c7:6f:e1/GenericAccess/DeviceName/Set
D (15820) MQTT_CLIENT: Subscribe successful
D (15840) MQTT_CLIENT: mqtt_enqueue id: 55009, type=8 successful
D (15840) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/GenericAccess/DeviceName/Set, id: 32586, type=8 successful
D (15850) MQTT_CLIENT: msg_type=9, msg_id=32586
D (15860) MQTT_CLIENT: msg_type=9, msg_id=32586
D (15860) BLE2MQTT: Found new characteristic!
D (15870) MQTT_CLIENT: pending_id=32586, pending_msg_count = 1
D (15870) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (15870) BLE2MQTT:   Service: 00001800-0000-1000-8000-00805f9b34fb
D (15870) MQTT_CLIENT: Subscribe successful
D (15890) BLE2MQTT:   Characteristic: 00002a01-0000-1000-8000-00805f9b34fb
D (15900) MQTT: Subscribing to c8:0f:10:c7:6f:e1/GenericAccess/Appearance/Get
D (15900) MQTT_CLIENT: mqtt_enqueue id: 32586, type=8 successful
D (15910) MQTT_CLIENT: Sent subscribe topic=c8:0f:10:c7:6f:e1/GenericAccess/Appearance/Get, id: 16970, type=8 successful
D (15920) MQTT_CLIENT: msg_type=9, msg_id=16970
D (15930) MQTT_CLIENT: msg_type=9, msg_id=16970
W (15930) BLE: Enqueue: id: 2, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
[... 245 lines cut ...]
W (18270) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (18270) BLE: Dequeue: id: 3, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (18280) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/Appearance = 3200
W (18320) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (18320) BLE: Dequeue: id: 4, type: 0, device: c8:0f:10:c7:6f:e1, char: 00002a02-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (18320) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPrivacyFlag = false
D (18330) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (18350) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
[... cut ...]
W (18530) BLE: Dequeue: id: 7, type: 2, device: c8:0f:10:c7:6f:e1, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbb6e0
I (18530) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAttribute/ServiceChanged = 1,65535
I (18540) BLE: BLE_OPERATION_TYPE_WRITE_CHAR: 0xf
D (18550) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (18750) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (18860) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (19060) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (19170) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
```

bilogic, 2019-04-28:
Actually, is it necessary responses (A) to be received before the next operation (Q)? e.g. Q1, A1, Q2, A2?
What can't we go Q1, Q2, Q3, Q4, A1, A2, A3, A4? Or even Q1, Q2, Q3, Q4, A1, A4, A2, A3?

shmuelzon, 2019-04-28:
The BLE queue was introduced in #3 and was meant to solve stability issues with BLE peripheral devices which exposed many services/characteristics like multi-sensors (TI Sensor Tag, Nordic Thingy:52, etc.). Overwhelming the BT stack with too many operations at once would cause issues so I decided to queue them at the application level and execute them one-by-one. Even if we decided to have 2/3/4 operation concurrently, we'd still be left with an operation (enabling indications) that doesn't let us know when it was completed.

At the very least, is seems that it should work since all the sanity checks in `ble_characteristic_notify_register()` have passed as the operation was queued.

Now, BLE GATT does define a write-without-response property for a characteristic but I think that's not relevant to the client configuration, only to the characteristic value itself as the client configuration might be writeable to enable notifications/indications, the characteristic value itself might be read-only.
Anyways, the 2a05 is standard and should behave correctly, but who knows. What if you add `00002a05-0000-1000-8000-00805f9b34fb` to the characteristics blacklist? Will the other characteristics behave as expected or also not send a response?

BTW - I don't suppose you have the means to sniff BLE traffic, do you? Would be nice to see of the write request is sent and if a response is received

bilogic, 2019-04-28:
Ok, it works. Added 0x2a05 to the blacklist and ```dequeue``` is no longer stuck. What does it take to sniff BLE traffic? I probably don't have it. The BP monitor has another indication issue which I want to ask about after this stuck queue is resolved. Thank you.

shmuelzon, 2019-04-28:
So, the rest of the dequeued `type: 2` operations work as expected? I wonder if those are notifications or indications. If they're indications then I have no idea why some would work and not others, If they're notifications then there's something problematic with indications and I'll need to see how I can debug that.

For sniffing, I have a [TI CC2540 USB dongle](https://www.aliexpress.com/item/Bluetooth-4-0-BLE-CC2540-USB-Dongle-Protocol-Analysis-BTool-Packet-Sniffer-Board-Debug-Pin-1Mbps/32769663761.html?spm=a2g0s.9042311.0.0.27424c4dxinbmg) and loaded it with TI's firmware for packet sniffing.

As for the BP monitor, could you open a separate issue? I don't mind looking at it simultaneously.

bilogic, 2019-04-28:
1. I don't have packet sniffing equipment
2. These are confirmed indications, I have modified ```ble_characteristic_notify_register``` to set ```notify_en = 2``` if ```CHAR_PROP_INDICATE``` bit is present
3. At our level, I don't think there is much difference between notification/indications, the only changes required are as per .2
4. I see the issue as the queue getting stuck while handling 0x2905 of the MI_SCALE.. perhaps we should dequeue based on a timeout OR response, which ever comes first. I could not make out how the current timeout is supposed to behave