shmuelzon/esp32-ble2mqtt #7: Error with the config_initialize()

DarvosBR, 2018-12-13:
Hello and thank you for this nice work!
I've used the esp-idf to flash the code into the esp32, i'm facing this error at the monitor once I open the Esp32 Terminal.

> [0;32mI (544) Config: Initializing configuration[0m
> ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d48c9
> file: "C:/Users/Nathan/msys32/home/Nathan/esp/esp32-ble2mqtt/main/ble2mqtt.c" line 376
> func: app_main
> expression: config_initialize()
> 
> Backtrace: 0x400935cc:0x3ffbb3d0 0x400939f8:0x3ffbb3f0 0x400d48c9:0x3ffbb410 0x400d27ce:0x3ffbb430
> 
> Rebooting...
> 

ok, I presume this error msg has something to do with the config file inside of the data folder, I've found a error on my previous config.json file, therefore I let that file just like the one on this git.
The other question I have is, I belive even after the alteration of the config.json file, the esp-idf does not rebuild the code! you know any command to rebuild the code? 

Ty very much!

DarvosBR, 2018-12-13:
Hey,
Thank you for the quick anwser!, it was just a few wrong coding on the config.json file! I've manage to connect either to the Wi-fi and to the MQTT and BLE ! preatty nice!

Unfortunately, after publishing a few infos on the MQTT, the device disconect either from the MQTT and from the BLE device.

here is the issue:

[0;32mI (15483) BLE2MQTT: Publishing: 4c:65:a8:d7:f4:05/226c0000-6476-4566-7562-66734470666d/226caa55-6476-4566-7562-66734470666d = 84,61,50,57,46,49,32,72,61,55,49,46,51,0[0m
[0;32mI (15493) BLE2MQTT: Publishing: 4c:65:a8:d7:f4:05/226c0000-6476-4566-7562-66734470666d/226caa55-6476-4566-7562-66734470666d = 84,61,50,57,46,49,32,72,61,55,49,46,50,0[0m
I (16013) wifi: bcn_timout,ap_probe_send_start
[0;33mW (16383) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013[0m
[0;32mI (16383) BLE: Connection closed, reason = 0x13[0m
[0;32mI (16383) BLE2MQTT: Disconnected from device: 4c:65:a8:d7:f4:05[0m
I (18523) wifi: ap_probe_send over, resett wifi status to disassoc
I (18523) wifi: state: run -> init (100)
I (18523) wifi: pm stop, total sleep time: 4482196 us / 16313479 us

I (18523) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
[0;31mE (18533) MQTT_CLIENT: Read error or end of stream[0m
[0;32mI (18533) WiFi: Disconnected[0m
[0;32mI (18533) MQTT_CLIENT: Reconnect after 10000 ms[0m
[0;32mI (18533) BLE2MQTT: Disonnected from WiFi, stopping MQTT[0m
[0;32mI (18543) MQTT: MQTT client disconnected[0m
[0;32mI (18553) MQTT: Disconnecting MQTT client[0m
[0;32mI (18553) BLE2MQTT: Disonnected from MQTT, stopping BLE[0m
[0;32mI (18563) MQTT: Connecting MQTT client[0m

Any idea? its a normal behavior? 
Other thing I forgot to mention, the broker keeps receiving PINGREQ and PINGRESP.

Client mosqsub|6140-hassbian received PUBLISH (d0, q0, r0, m0, '4c:65:a8:d7:f4:05/BatteryService/BatteryLevel', ... (3 bytes))
100
Client mosqsub|6140-hassbian sending PINGREQ
Client mosqsub|6140-hassbian received PINGRESP
Client mosqsub|6140-hassbian sending PINGREQ

Any idea how to prevent those things?

shmuelzon, 2018-12-13:
Well, regarding the disconnections, which ESP-IDF are you using? Did you try with the version described in the README file? I've had varying degrees of success with ESP-IDF versions but the one described in the README is stable for me.

The BLE disconnection reason is 0x13 which, according to the documentation, means `Connection terminate by peer user`. Supposedly the BLE peripheral (your Mijia temperature+humidity sensor) caused the disconnection, but I wouldn't trust that just yet.

As for the PINGREQ and PINGRESP, those are a part of the MQTT protocol and must be sent after some keep-alive period if no other control traffic was exchanged between the client and broker. Those are not topics being published and no other client will them, so I don't think you should worry about them or try to prevent them.

DarvosBR, 2018-12-13:
I'm using the version : v3.2-dev-708-ga557e8cc, gonna try the one you mentioned!
Thank you for the quick support and nice code you have here! let me ask you something, there is an easy way to send the packages in HEX? or at least to print them in HEX on the terminal?

DarvosBR, 2018-12-13:
Sir, sorry again to bother you but i'm having trouble downloading the old version of the esp-idf on git, the version 139d49894c473bd34ab6e9515e85d3f6871a30c7 must be git cloned in orther to work properly.

I already tried git hist, git reset and no success. do you know how can I clone an older version? whats the command? 

Ty for ur help!

DarvosBR, 2018-12-13:
Ok, everything seems to be working great now... Ty.
As for the Hex value, you are saying i can change the value type on the config.json...
Like this:

"characteristics": {
  "definitions": {
    "00002f01-0000-1000-8000-00805f9b34fb": {
      "name": "Relay State",
      "types": [
        **"HEX"**
      ]

DarvosBR, 2018-12-14:
> However, please note that assuming the value types of these characteristic is standard, you can add their definition to the configuration file and it will convert the byte array to a human-readable string.

Teach me  Master!

shmuelzon, 2018-12-14:
> As for the Hex value, you are saying i can change the value type on the config.json...

No, if you want the characteristic values to be HEX, you will need to modify the 2 functions described above.
What I meant was that if your characteristics are strings, 32-bit integers, boolean, or any other [standard type](https://www.bluetooth.com/specifications/assigned-numbers/format-types), the code can parse it and convert it to a human-readable string. This is done automatically for all [standard characteristics](https://www.bluetooth.com/specifications/gatt/characteristics) that are defined by the Bluetooth SIG.

If your device has non-standard characteristics, you can manually add the definitions in your configuration file so it will parse them correctly and won't default to byte-arrays. You can find the list of supported formats [here](https://github.com/shmuelzon/esp32-ble2mqtt/blob/50623cd24a9461d6bccc553bfdbe48b27d5bae67/main/ble_utils.c#L178).

As far as I can tell, the `226caa55-6476-4566-7562-66734470666d` characteristic is a string, so you should be able to use the `utf8s` type. It should look something like:
```json
"characteristics": {
  "definitions": {
    "226caa55-6476-4566-7562-66734470666d": {
      "name": "TemperatureAndHumidity",
      "types": [
        "utf8s"
      ]
    }
  }
}
```

DarvosBR, 2018-12-14:
I finaly made it ^^

"Utf8s" was all i need it, found it at the GATT files of bt site

shmuelzon, 2018-12-14:
Glad to hear it worked out!
Can we close this issue or is there anything else?

shmuelzon, 2018-12-14:
> Now I just need to think in a way to conver this msgs into Json format for better use!

Since you're using Home Assistant, you could just make use of its templates to parse the string directly. I don't think you have to convert it to JSON first.