shmuelzon/esp32-ble2mqtt #10: make: xtensa-esp32-elfgcc: Command not found

salarelv, 2019-01-15:
This isnt probably the problem with this library but I didn't find any other place.
So it says xtensa-esp32-elfgcc not found but the file in ztensa-esp32-elf is xtensa-esp32-elf-gcc not xtensa-esp32-elfgcc.

All examples run fine. 

taavi@taavi-ThinkPad-T430:~/projects/oss/esp32-ble2mqtt$ make
make: xtensa-esp32-elfgcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elfgcc: not found
/bin/sh: 1: xtensa-esp32-elfgcc: not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elfgcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elfgcc: not found
/bin/sh: 1: xtensa-esp32-elfgcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/bootloader_support/src/bootloader_flash.o
make[2]: xtensa-esp32-elfgcc: Command not found
/media/taavi/Apps/applications/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 127
/media/taavi/Apps/applications/esp/esp-idf/make/project.mk:527: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/media/taavi/Apps/applications/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/taavi/projects/oss/esp32-ble2mqtt/build/bootloader/bootloader.bin' failed
make: *** [/home/taavi/projects/oss/esp32-ble2mqtt/build/bootloader/bootloader.bin] Error 2

shmuelzon, 2019-01-16:
Hey,

The toolchain prefix (`xtensa-esp32-elf-`) resides in the sdkconfig file.
I would guess that you inadvertently removed the trailing dash from there.
You can either run `make menuconfig`, go into "SDK tool configuration" and add the trailing dash there. Or simply edit the `sdkconfig` file. It's the first configuration option there.

Good day

salarelv, 2019-01-16:
Thanks that was it.

closed 2019-01-16 by salarelv