shmuelzon/esp32-ble2mqtt #15: 433Mhz Radio and other custom developed module support?

robertcsakany, 2019-01-23:
I'm planning that using these chips also for receiving / transmitting radio codes over MQTT. Maybe I will use that as a module in other project - but on that case some refactor needed to able to use as an external module in other projects. Is there any recommended practice how to do that?

shmuelzon, 2019-01-23:
Well, this project wasn't written as a separate component. It's not just a library with external headers that you can add and call.
You can always just fork it and add another module, like ble.c, that the main app will initialise and act upon events (callbacks) from it.

robertcsakany, 2019-01-28:
Make sense, but on that case the maintenance of fork can get a lot of time.  Lot of functions - the BLE processing functions - in ble2mqtt.c too - can be modularised. I understand that this project base goal is the ble 2 mqtt gateway functions, but some specific implementations can reuse the most of the functions as module - I think the BLE functions are implemented well and other projects can reuse it. 

I will make some POC about it sometimes. (after some more important issue for me)

shmuelzon, 2019-01-29:
OK, we can discuss again once you have something ready. I'm not sure separating the BLE wrappers as a separate component is really worth while so it would either need to be a fork or somehow generalize the BLE actions in ble2mqtt.c in way that still leaves the implementation simple without loosing too much.

closed 2019-01-29 by shmuelzon