shmuelzon/esp32-ble2mqtt #16: MAC Whitelist / Blacklist / Passkey regexp like support

robertcsakany, 2019-01-23:
I have a lot of devices and the whitelist / blacklist function administration by address can cause lot of pain. It can be easier if I can specify that per manufacturer. For example: 

`
"whitelist": [
  "aa:bb:cc:*:*:*"
]
`

robertcsakany, 2019-01-23:
Yes. That was my "regex like" function, the * or ?? does not matter me. Better the use of Wildcard as name. I fix the issue topic.

robertcsakany, 2019-01-23:
It may an easy implementation that replace the characters on the source mac addresses to ? on positions where the configurations contains them and only strcmp is required.  Does it makes sense?

robertcsakany, 2019-01-23:
There is a proposal for wildcard support:

```
// The main function that checks if two given strings 
// match. The first string may contain wildcard characters 
int match(char *first, char * second) 
{ 
    // If we reach at the end of both strings, we are done 
    if (*first == '\0' && *second == '\0') 
        return 1; 
  
    // Make sure that the characters after '*' are present 
    // in second string. This function assumes that the first 
    // string will not contain two consecutive '*' 
    if (*first == '*' && *(first+1) != '\0' && *second == '\0') 
        return 0; 
  
    // If the first string contains '?', or current characters 
    // of both strings match 
    if (*first == '?' || *first == *second) 
        return match(first+1, second+1); 
  
    // If there is *, then there are two possibilities 
    // a) We consider current character of second string 
    // b) We ignore current character of second string. 
    if (*first == '*') 
        return match(first+1, second) || match(first, second+1); 
    return 0; 
}
```

shmuelzon, 2019-01-23:
I saw that implementation when I was looking before. It seemed cumbersome and I'm not sure if it will handle multiple wildcards as expected. That's why I opted for a single-character wildcard. much more simple to implement.
The only reason I suggested the `?` is because it usually denotes a single character where `*` is for zero-or-more characters.

commit f8d28bcb (2019-01-27): Support wildcards in MAC+UUID filters and passkeys

Fixes #16
closed 2019-01-27 by shmuelzon