shmuelzon/esp32-ble2mqtt #17: Remote UDP Logging

robertcsakany, 2019-01-23:
I would like to log to remote syslog server over UDP.  The ESP-IDF gives API to be able to set it up. I found a  github project to achieve it:

https://github.com/MalteJ/embedded-esp32-component-udp_logging

I suggest to make config entry, when server is set, the init is run (after WIFI connection is made)
'
{
  "log": {
         "server" : {
               "host" : "hoostname",
               "port" : port
          }            
  }
'

What do you think?

shmuelzon, 2019-01-23:
I like this idea very much, not sure regarding this specific implementation though.
I would also suggest using multicast and then the configuration option would only be to enable or disable it. Perhaps that would even be a build-time configuration and not part of the configuration file.
The (Python?) code on the PC that would listen to the logs, could do a reverse DNS lookup and prefix the log messages with the instance name (BLE2MQTT-XXXX).

robertcsakany, 2019-01-23:
About multicast:

The address space is used for multicast is:

Relative | Description | References | Change Controller | Date Registered | Last Reviewed
-- | -- | -- | -- | -- | --
232.0.0.0 | Reserved | [RFC4607] |   |   |  
232.0.0.1-232.0.0.255 | Reserved for IANA allocation | [RFC4607] |   |   |  
232.0.1.0-232.255.255.255 | Reserved for local host allocation | [RFC4607] |   |   |  

Just an idea: Does it makes sense when the host address in multicast space uses multicast send to log, when normal address space it uses normal UDP socket transmit?

shmuelzon, 2019-01-23:
> Just an idea: Does it makes sense when the host address in multicast space uses multicast send to log, when normal address space it uses normal UDP socket transmit?

Sounds reasonable to me.
If you want to support both IPv4 and IPv6 then the code needs identify the IP family from the destination IP string. Maybe something like that already exists in ESP-IDF, need to check.
Whatever script the listens to these logs would need to do the same and if it's a multicast/broadcast IP listen on it and if it's a unicast IP, verify that it has that IP on one of its interfaces and display an error if not.

robertcsakany, 2019-01-28:
I've checked the official ESP-IDF multicast example, there is mechanism which can determinate that a given address is a multicast family address, so no need special utility function for that.

### Ipv4 version

```
if (!IP_MULTICAST(ntohl(s_addr))) {
        ESP_LOGW(V4TAG, "Configured IPV4 multicast address '%s' is not a valid multicast address. This will probably not work.", MULTICAST_IPV4_ADDR);
    }
```

### Ipv6 version

```
    if (!ip6_addr_ismulticast(&multi_addr)) {
        ESP_LOGW(V6TAG, "Configured IPV6 multicast address '%s' is not a valid multicast address. This will probably not work.", MULTICAST_IPV6_ADDR);
    }
```

commit 2cdbde35 (2019-01-31): Add support for remote UDP logging

Fixes #17
closed 2019-01-31 by shmuelzon