shmuelzon/esp32-ble2mqtt #18: Add IPV6 Support

robertcsakany, 2019-01-23:
For my IOT network I'm using dual stack network. Is there any reason that IPV6 support is not enabled (size, stability etc.)? I've made some modification in WIFI.C and it works well.

shmuelzon, 2019-01-23:
No specific reason. I don't use IPv6 in my network so I never even thought about it.
Does it affect image size? Free RAM?

robertcsakany, 2019-01-23:
I will check it soon.

robertcsakany, 2019-01-28:
I've made the checking, and it seems there is not any impact at all, because the version of lwip (which is included in ESP-IDF) is Ipv6 compliant by defult.
But for the record there is the statistics:

### Image size
#### Ipv6 not initialized:
Wrote 1280928 bytes (793164 compressed) at 0x00010000 in 12.6 seconds (effective 815.7 kbit/s).

#### Ipv6 initialized:
Wrote 1281616 bytes (793621 compressed) at 0x00010000 in 12.6 seconds (effective 816.3 kbit/s)...

### Heap size
I made a timered task to determinate every 5 sec the actual free heap size. I've waited for every BLE advertisment / discovery happens and that free heap value is used.

#### Ipv6 not initialized:
I (55351) STAT: free DRAM 65056

#### Ipv6 initialized:
I (80351) STAT: free DRAM 65052

So I think there is no risk.
I create a pull request with the patch soon.