shmuelzon/esp32-ble2mqtt #23: build error in log.c

knopserl, 2019-02-10:
I get a compile error. Not sure if I'm doing something wrong?

Downloaded org.bluetooth.characteristic.plx_spot_check_measurement
Generating source code
CC build/main/ble_utils.o
CC build/main/log.o
D:/mcdev/msys32/home/esp/esp32-ble2mqtt/main/log.c: In function 'log_start':
D:/mcdev/msys32/home/esp/esp32-ble2mqtt/main/log.c:43:5: error: implicit declaration of function 'memset' [-Werror=implicit-function-declaration]
     memset(&dst, 0, sizeof(dst));
     ^
D:/mcdev/msys32/home/esp/esp32-ble2mqtt/main/log.c:43:5: warning: incompatible implicit declaration of built-in function 'memset'
D:/mcdev/msys32/home/esp/esp32-ble2mqtt/main/log.c:43:5: note: include '<string.h>' or provide a declaration of 'memset'
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/esp/esp-idf/make/component_wrapper.mk:292: log.o] Fehler 1

knopserl, 2019-02-10:
ok, solved,it requires to include <string.h> in log.c then the error is gone.

shmuelzon, 2019-02-10:
Fixed, thanks!

commit 532242ac (2019-02-10): Added missing include

Fixes #23
closed 2019-02-10 by shmuelzon