shmuelzon/esp32-ble2mqtt #24: Restart loop after flashing

eercanayar, 2019-02-26:
Hey everyone,

## How to reproduce

- I've cloned the repo with its submodules
- Configured serial flasher port via `make menuconfig`
- Edited config.json for just wifi credentials and mqtt.host

After first `make flash`, I've faced is there was no mkspiffs executable in it's folder.

```
make: /Users/emir.ayar.empa/esp/esp32-ble2mqtt/mkspiffs/mkspiffs: No such file or directory
make: *** [/Users/emir.ayar.empa/esp/esp32-ble2mqtt/build/spiffs.bin] Error 1
```

Then, I couldn't figure out to build it. The instruction on the README of its repo (https://github.com/igrr/mkspiffs) says;

```
$ git submodule update --init
$ make dist
```

But it didn't worked. I've gcc-4.9 and everything seems okay but It gave something ends with this;

```
      ...
  "_strncpy", referenced from:
      _spiffs_stat_pix in spiffs_hydrogen.o
      _spiffs_object_create in spiffs_nucleus.o
      _spiffs_object_update_index_hdr in spiffs_nucleus.o
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [mkspiffs] Error 1
```

So, I downloaded the latest release from its Releases; `mkspiffs-0.2.3-esp-idf-osx.tar.gz` and copied it to mkspiffs folder. I was feeling ready to `make flash` the project. 

## Problem

The project is flashed successfully.

```
Emirs-MacBook-Pro:esp32-ble2mqtt emir.ayar.empa$ make flash
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x10000 )...
esptool.py v2.5.0
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: 30:ae:a4:15:29:dc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 20928 bytes to 12379...
Wrote 20928 bytes (12379 compressed) at 0x00001000 in 1.1 seconds (effective 153.1 kbit/s)...
Hash of data verified.
Compressed 1185488 bytes to 723300...
Wrote 1185488 bytes (723300 compressed) at 0x00010000 in 63.8 seconds (effective 148.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1310.6 kbit/s)...
Hash of data verified.
Compressed 524288 bytes to 1203...
Wrote 524288 bytes (1203 compressed) at 0x00310000 in 0.1 seconds (effective 37784.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

But it have been failed into a reset loop. Here is the some of logs of `make monitor`

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (405) BLE2MQTT: Version: v0.6.0-7-g532242a
I (405) Config: Initializing configuration
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d46f7
0x400d46f7: app_main at /Users/emir.ayar.empa/esp/esp32-ble2mqtt/main/ble2mqtt.c:402 (discriminator 1)

file: "/Users/emir.ayar.empa/esp/esp32-ble2mqtt/main/ble2mqtt.c" line 402
func: app_main
expression: config_initialize()

Backtrace: 0x40093830:0x3ffd9350 0x40093ce0:0x3ffd9370 0x400d46f7:0x3ffd9390 0x400d24da:0x3ffd93b0
0x40093830: invoke_abort at /Users/emir.ayar.empa/esp/esp-idf/components/esp32/panic.c:649

0x40093ce0: _esp_error_check_failed at /Users/emir.ayar.empa/esp/esp-idf/components/esp32/panic.c:665

0x400d46f7: app_main at /Users/emir.ayar.empa/esp/esp32-ble2mqtt/main/ble2mqtt.c:402 (discriminator 1)

0x400d24da: main_task at /Users/emir.ayar.empa/esp/esp-idf/components/esp32/cpu_start.c:470

Rebooting...
ets Jun  8 2016 00:22:57
```

Do you have any ideas?

shmuelzon, 2019-02-26:
Hey,

First of all, spiffs should have been compiled during along with everything else. I worked fine on my Mac, so I'm not sure why it failed.
Anyways, the reason your ESP32 keeps rebooting is because it couldn't load the configuration file.
One reason might be that your JSON file isn't valid. Could you please try and pass it through a [JSON validator](https://jsonlint.com/) first? If your configuration file is valid, another issue might be a mismatched SPIFFS version since I'm not sure what you compiled is the same (and configured the same) as what ESP-IDF is using.

Let me know...

shmuelzon, 2019-03-04:
@eercanayar, any updates?

shmuelzon, 2019-03-08:
@eercanayar, I'm closing this issue for now as I haven't heard back from you. If you're still experiencing trouble, let me know.

closed 2019-03-08 by shmuelzon

fabiosoft, 2019-10-29:
I have the same issue... no `spiffs.bin` file is generated while building... so flashing without gives me a bunch of error before reboot.

I'm using Mac os Catalina.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6152
load:0x40078000,len:10180
load:0x40080400,len:6660
entry 0x40080764
.[0;32mI (29) boot: ESP-IDF v3.2.2 2nd stage bootloader.[0m
.[0;32mI (29) boot: compile time 23:59:42.[0m
.[0;32mI (29) boot: Enabling RNG early entropy source....[0m
.[0;32mI (34) boot: SPI Speed      : 40MHz.[0m
.[0;32mI (38) boot: SPI Mode       : DIO.[0m
.[0;32mI (42) boot: SPI Flash Size : 4MB.[0m
.[0;32mI (46) boot: Partition Table:.[0m
.[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length.[0m
.[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000.[0m
.[0;32mI (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000.[0m
.[0;32mI (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000.[0m
.[0;32mI (79) boot:  3 ota_0            OTA app          00 10 00010000 00180000.[0m
.[0;32mI (87) boot:  4 ota_1            OTA app          00 11 00190000 00180000.[0m
.[0;32mI (94) boot:  5 fs_0             Unknown data     01 82 00310000 00040000.[0m
.[0;32mI (101) boot:  6 fs_1             Unknown data     01 82 00350000 00040000.[0m
.[0;32mI (109) boot: End of partition table.[0m
.[0;32mI (113) boot: No factory image, trying OTA 0.[0m
.[0;32mI (118) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x37e6c (228972) map.[0m
.[0;32mI (207) esp_image: segment 1: paddr=0x00047e94 vaddr=0x3ffbdb60 size=0x05934 ( 22836) load.[0m
.[0;32mI (217) esp_image: segment 2: paddr=0x0004d7d0 vaddr=0x40080000 size=0x00400 (  1024) load.[0m
.[0;32mI (217) esp_image: segment 3: paddr=0x0004dbd8 vaddr=0x40080400 size=0x02438 (  9272) load.[0m
.[0;32mI (229) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xf0ef4 (986868) map.[0m
.[0;32mI (580) esp_image: segment 5: paddr=0x00140f14 vaddr=0x40082838 size=0x13e48 ( 81480) load.[0m
.[0;32mI (628) boot: Loaded app from partition at offset 0x10000.[0m
.[0;32mI (628) boot: Disabling RNG early entropy source....[0m
.[0;32mI (628) cpu_start: Pro cpu up..[0m
.[0;32mI (632) cpu_start: Starting app cpu, entry point is 0x40081020.[0m
.[0;32mI (624) cpu_start: App cpu up..[0m
.[0;32mI (642) heap_init: Initializing. RAM available for dynamic allocation:.[0m
.[0;32mI (649) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM.[0m
.[0;32mI (655) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM.[0m
.[0;32mI (661) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM.[0m
.[0;32mI (667) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM.[0m
.[0;32mI (674) heap_init: At 3FFD33B8 len 0000CC48 (51 KiB): DRAM.[0m
.[0;32mI (680) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM.[0m
.[0;32mI (686) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM.[0m
.[0;32mI (692) heap_init: At 40096680 len 00009980 (38 KiB): IRAM.[0m
.[0;32mI (699) cpu_start: Pro cpu start user code.[0m
.[0;32mI (46) cpu_start: Starting scheduler on PRO CPU..[0m
.[0;32mI (0) cpu_start: Starting scheduler on APP CPU..[0m
.[0;32mI (131) BLE2MQTT: Version: v0.11.0-1-gea67a23.[0m
.[0;32mI (131) Config: Initializing configuration.[0m
.[0;31mE (171) Config: Failed loading partition 0, falling back to 1.[0m
.[0;31mE (201) Config: Failed loading partition 1 as well.[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40090424
file: "/Users/fabio/Downloads/esp32-ble2mqtt/main/ble2mqtt.c" line 753
func: app_main
expression: config_initialize()

Backtrace: 0x4008ffdc:0x3ffbb450 0x40090427:0x3ffbb470 0x40104dc1:0x3ffbb490 0x400d0e36:0x3ffbb4d0 0x4008c939:0x3ffbb4f0

Rebooting...
```

shmuelzon, 2019-10-30:
Yes, without a valid configuration, you can't run the application
```
(131) Config: Initializing configuration
(171) Config: Failed loading partition 0, falling back to 1
(201) Config: Failed loading partition 1 as well
```
As for why `spiffs.bin` isn't generated, we'll discuss on #48.

fabiosoft, 2019-10-31:
after compiled mkspiffs, it worked.
Thanks