shmuelzon/esp32-ble2mqtt #26: Roadmap and Lates IDF and WiFi issues

jamesfcurran, 2019-03-20:
Hi Assaf,
Not sure if this is the best way to contact you but here goes..

Was wondering if you have a roadmap for this project as it seems to fill a valuable niche.

Also, I am having significant troubles even with recent IDF releases with WiFi dropping out every few minutes.   Seems like people attribute this to the shared WiFi/Bluetooth hardware.  Not sure if you have seen this or have advice.

Thanks,
Jim

shmuelzon, 2019-03-20:
Hey Jim,

I don't have a specific roadmap per-se but there are a few requested features in the currently open issues. Sadly, I haven't gotten around to addressing them yet but hope to get around to it in the near future.
If you have a suggestion, feel free to open an issue about it and/or submit a pull-request.

Regarding WiFi issues, I'm personally working against ESP-IDF tag v3.1.2 and don't see many disconnections. I used to see them in previous versions when WiFi/BLE coexistence was managed in software and so the default settings for this project were to use hardware coexistence but I reverted back to software management with the recent versions of ESP-IDF.

If you're using a different ESP-IDF version, I suggest you test with 3.1.2, maybe 3.1.3 (I haven't tried it yet). If it doesn't improve your situation, you can try to use the hardware coexistence variant in the configuration menu (Component config -> Wi-Fi) and turn off "Software controls WiFi/Bluetooth coexistence".

Let me know how it works out for you.

jamesfcurran, 2019-03-20:
Hi Assaf,
I'm using v3.1.3 and am seeing frequent instances of "wifi: bcn_timout,ap_probe_send_start".  

Upgrading to a newer version of the SDK made a huge difference in my case.  But these messages are pretty frequent (every 2-3 minutes) and usually signal an impending WiFi disconnect which leads to a gap in BLE scanning.   I'll review the settings you mentioned.

One feature I am wondering about is supporting use of the adv_filter_policy parameter to enable whitelisting that I noticed in the SDK.  Another is passive scan support.   I anticipate trying to track a lot of devices so anything that minimizes overhead in ble2mqtt is interesting.   Am just starting out with BLE so I hope I am making sense.

Thanks,
Jim

shmuelzon, 2019-03-20:
Hey Jim,

AFAIK, adv_filter_policy is only relevant for the BLE peripherals, i.e. the devices that send the advertising packets, so that wouldn’t apply for this project.
As for passive scanning, should be as simple as changing `BLE_SCAN_TYPE_ACTIVE` to `BLE_SCAN_TYPE_PASSIVE` in main/ble.c. Techinally, a passive scan might be all we need in the first place as we don’t really make use of the information in the scan result.

I don’t know what your use-case is and if you’re interested more in the advertising data or connecting to the devices. In case of the former, using passive scan and an empty whitelist should keep you only in RX but the constant checks of supported broadcasters might be too much. In case of the latter, you’re limited to 7 simultaneous connections anyways. Here the bottleneck might be converting UUIDs and the payload to names and strings. You might want to think about using multiple ble2mqtt’s with overlapping coverage to act as sort-of load balancer/backup.

Good day,
Assaf

jamesfcurran, 2019-03-22:
Thanks Assaf,
I replaced BLE_SCAN_TYPE_ACTIVE with BLE_SCAN_TYPE_PASSIVE in ble.c
and
*enabled* "Software controls WiFi/Bluetooth coexistence.." 

This stabilized my WiFi connections !  Again, am using IDF v3.1.3.  I see there is a newer one in beta.

Thanks for the suggestions on managing volume.   Am looking forward to scaling up the number of BLE iBeacons in this project.

Jim

closed 2019-03-22 by jamesfcurran