shmuelzon/esp32-ble2mqtt #27: How can I upload this to my ESP32 board

hapklaar, 2019-03-21:
Excuse my ignorance, but how can I compile and upload this to my ESP32 board? I'd like to use this program to monitor my Tilt hydrometer and send it's major/minor data to an MQTT gateway.

Up until now I've only used Arduino IDE and am helpless without an .ino file...

thanks!

closed 2019-03-21 by hapklaar

DurandA, 2019-03-21:
Can you tell us more about your setup? What operating system do you use? The first step is to [set up the toolchain](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-1-set-up-the-toolchain) and then follow the steps until step 9.

Once this is done, you should update the [configuration file](https://github.com/shmuelzon/esp32-ble2mqtt#configuration) according to your device. I personally use [nRF Connect](https://www.nordicsemi.com/Software-and-Tools/Development-Tools/nRF-Connect-for-mobile) to explore the services and characteristics (e.g. hydrometer value) that are exposed by the BLE device before configuring.

The last step is to flash the ESP32 firmware using the `make flash` command.

hapklaar, 2019-03-22:
Hi DurandA,

Thanks for responding. I was able to get it flashed and can even do OTA updates, after following the guide.

Didn't know about nRF connect, will check it out! Have been using BLE Scanner so far.

shmuelzon, 2019-03-22:
Hey @hapklaar,

Glad to hear you were able to set everything up!
The next step would be to get the information from the Tilt sensor. As far as I can tell, the Tilt uses the iBeacon standard to transmit its sensor readings with the major and minor values corresponding to temperature and gravity respectively, see [here](https://kvurd.com/blog/tilt-hydrometer-ibeacon-data-format/).
Since we already support iBeacons, this means that you should be able to get the readings without any modifications.

Let me know if you run into any issues.

Good day!

hapklaar, 2019-03-22:
Hi @shmuelzon , good day to you too!

Had no issues receiving data from the Tilt>ESP>Pi using mosquitto. Already graphing it in Zabbix on my Pi.

Setup seems to be a little unstable however. Every few hours I have to reset the ESP to get new readings. Also haven't been able to get the log monitoring part working. After 'make remote-monitor' in msys32 shell no data is received. It just sits there at 

```
hapkl@SLOEBER-10 MINGW32 ~/esp/esp32-ble2mqtt
$ make remote-monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0

```

shmuelzon, 2019-03-22:
Hey,

Regarding setup stability, have a serial connection open would be the best way to figure out what's going on but we can start with remote logging. Next time it fails, could you please also try to ping the ESP32 so we know if it's still connected to the WiFi network or not?
Also, different ESP-IDF versions give very different results. I'm personally using v3.1.2 and don't see these issues. If you're using something else, please try to switch to v3.1.2 first.

As for the remote logging feature, once it starts listening for log messages, is should print `Listening on port xxxx`. Since I don't see that message, either the Python script is not running or something else might be wrong. Could you please try to run the script manually with `python remote_log.py`?

Let me know...

hapklaar, 2019-03-22:
I seem to be using a newer version:
```
hapkl@SLOEBER-10 MINGW32 ~/esp/esp-idf
$ git describe --tags --dirty
v4.0-dev-137-gce07a1057
```
When I run the python script directly, there is also no output.

Shall try 3.1.2 later today. Guess a make clean / make flash after changing $IDF_PATH to the 3.1.2 folder is required?

shmuelzon, 2019-03-22:
> Guess a make clean / make flash after changing $IDF_PATH to the 3.1.2 folder is required?

I would suggest a `rm -rf build sdkconfig sdkconfig.old` to make sure nothing is left from the newer version as the configuration options would also be different and `make clean` doesn't remove that.

hapklaar, 2019-03-22:
I redid everything with v3.1.2 and also new clone of your repo, but still no joy on remote-monitor. It's just sitting there doing nothing apparent. A 'make remote-monitor -d' show nothing special either, other than that its spawning a process after compiling:

```
Considering target file 'remote-monitor'.
 File 'remote-monitor' does not exist.
 Finished prerequisites of target file 'remote-monitor'.
Must remake target 'remote-monitor'.
Putting child 0x200570b0 (remote-monitor) PID 3676 on the chain.
Live child 0x200570b0 (remote-monitor) PID 3676
Live child 0x200570b0 (remote-monitor) PID 3676

```

shmuelzon, 2019-03-22:
I'm guessing `make` is doing its job but Python is giving us some trouble.

Which version of Python are you using? I've tested on both Python 2.7 and 3.7 but on macOS and Linux, haven't tried on Windows.
Maybe we can try to trace where the script is failing by running it as: `python -m trace --trace remote_log.py`?

Could you also share the "log" portion of your configuration?

hapklaar, 2019-03-22:
Python version is 2.7.15

It's acting very strange looking at the trace, it seems to hang then continue at the same stages every time I execute `python -m trace --trace remote_log.py`. Eventually some logdata gets through though in big chunks. Apparently I just didn't wait long enough before.

Both trace and log output seems to halt for 20 seconds then another chunk is shown upon where output halts again.. Hard to explain, uploaded a vid of what is happening [here](https://we.tl/t-jJ0ujukALH)

shmuelzon, 2019-03-22:
Hey,

Sorry for the back-and-forth but since the outout was buffered, I can’t tell exactly where it really hangs. Could you please run `python -u -m trace --trace remote_log.py` so it will be unbuffered?

Thanks

hapklaar, 2019-03-22:
Not a problem, really appreciate your help!

The buffering was actually the reason for the 'halts' in output. With -u it works just fine and shows trace + log without issue.

Without option -u it takes a while for log output to start, with -u I now have wireless realtime logging. :)

I suspect, because logs are transmitted using UDP and WIFI on my ESP32 seems flaky at best, and the device was not near my access point, I was just missing packets.

shmuelzon, 2019-03-23:
This is probably just a difference in how output buffers are managed in Windows. I changed `make remote-monitor` to use `-u` be default.