shmuelzon/esp32-ble2mqtt #28: Whitelisting mac address

hapklaar, 2019-03-22:
I'm only interested in receiving broadcasts from one single mac address, so I changed config.json to reflect this:
```
	"ble": {
		"whitelist": [
			"c5:11:8c:ba:55:ef"],
...
```
But when using serial monitor I still see a lot more action going on where BLE2MQTT is trying to connect all sorts of BLE devices after scanning. Is it possible to only listen for broadcasts, to save on power usage?

shmuelzon, 2019-03-22:
Hey @hapklaar,

Could you please post an example of what you're seeing in the log?

Also, can you verify that the configuration file on your host matches the one on the ESP? To do that, look at the MD5 hash calculated by the ESP during boot (you should see a "version: xxxxxx" message) and compare it to one calculated on your host.

There is no setting for disabling connections, and when a new device is found it will always do something about that, but it should still not connect to devices that aren't in the whitelist.

Good day

hapklaar, 2019-03-22:
Hey @shmuelzon :)

Please find my serial output here: 
[serialoutput.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/2997636/serialoutput.txt)
I see this version line during boot: `I (351) BLE2MQTT: Version: v0.6.0-9-gcc7fbae`

I'm not sure how to get the MD5 hash calculated on my host though... Where do I calculate this from?

shmuelzon, 2019-03-22:
Hey,

Sorry, should have been clearer.
The configuration that’s on the ESP is:
> I (391) Config: version: a538d55078fe24957f28c7c2e4501480

To calculate the hash of the configuration on the host, execute `md5sum data/config.json`.

The snippet of the configuration file looks OK. If the configuration hashes match, you can send me your entire configuration (after removing sensative information) and I’ll try it on my setup.

Thanks

hapklaar, 2019-03-22:
It seems the configs didn't match
`$ md5sum data/config.json
331381d494cb463ae3347f9ea56e9902 *data/config.json
`

and the reason is that I failed to see the config had to be uploaded using `make upload-config` instead of `make upload`... With this new config it is indeed working just fine.

Thanks!

closed 2019-03-22 by hapklaar