shmuelzon/esp32-ble2mqtt #29: ESP32 wifi power save

hapklaar, 2019-03-22:
The ESP32 apparently suffers from some WiFi power save bug, the issue is described [here](https://github.com/espressif/arduino-esp32/issues/1484)

This seems to impacts UDP logging and MQTT data for me, as RTT is bad and I think I also have packet loss because of this.

Do you think the fix discussed there (`Esp_wifi_set_ps (WIFI_PS_NONE);`) could help? Might be nice to be able to enable and disable this via config, to also have the option to keep it on if power consumption is an issue.

shmuelzon, 2019-03-23:
Hey @hapklaar,

When the radio sleeps it won't be able to receive any packets but the access point will queue them and send them when it wakes up again. This will add latency (hence the higher round-trip-time) but should no affect packets sent by the ESP (logs, MQTT publishes, etc.).
It will affect MQTT topics you want to send to the ESP when you want to write to a BLE characteristic but you don't mention those as a problem.

The WiFi issues @jamesfcurran mentioned were mostly remedied by using the suggested ESP-IDF version in the README file.

Anyway, please try adding that line in your setup and let me know if it really does make a difference for you for anything other than pings.

Thanks

hapklaar, 2019-03-23:
Hi Assaf,

Can you tell me where I should add this to test? It seems to also need esp_wifi.h.

Thanks,
Erwin

shmuelzon, 2019-03-23:
Hey,

I would probably add it at the end of [wifi_initialize()](https://github.com/shmuelzon/esp32-ble2mqtt/blob/96ef229491d32e487e4d17d0dcb53b136f239062/main/wifi.c#L91).
esp_wifi.h is already included.

Good day

hapklaar, 2019-03-23:
This certainly reduced RTT tenfold, but I guess you have a point about packets being initiated from the esp32 should not be affected. At least I learned something as I'm a complete n00b in programming.

I'd like to play around a little more to see if I can reduce power consumption, as I run this on an 18650 cell. Would it be an easy change to just listen for Beacons and publish their values to MQTT server? Ideally it would only listen for this, store the last value it got which could then be pulled by a client of some sorts. The module then only would have to listen and not transmit and thus use less power. But I guess this is not in the scope of your program.

shmuelzon, 2019-03-23:
Hey,

Making the following changes in main/ble.c would avoid sending out any BLE packets:
```patch
diff --git a/main/ble.c b/main/ble.c
@@ -17,7 +17,7 @@
 
 static const char *TAG = "BLE";
 static esp_ble_scan_params_t ble_scan_params = {
-    .scan_type = BLE_SCAN_TYPE_ACTIVE,
+    .scan_type = BLE_SCAN_TYPE_PASSIVE,
     .own_addr_type = BLE_ADDR_TYPE_RANDOM,
     .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
     /* Scan interval. This is defined as the time interval from
@@ -559,6 +559,7 @@ static void gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
                 param->scan_rst.rssi, broadcaster_ops);
         }
 
+        break;
         device = ble_device_find_by_mac(devices_list, param->scan_rst.bda);
 
         /* Device already discovered, nothing to do*/
```
The first change would change the BLE scanning mode to passive (doesn't send out scan requests) and the second will not notify the application when a new BLE device is discovered (and so it won't try to connect to it).

Event with these changes I'm not sure how long you can run off batteries as the entire setup isn't that geared toward low power, but it's a step in the right direction.

Good day

closed 2019-03-23 by shmuelzon

hapklaar, 2019-03-26:
Hi Assaf,

fyi setting scan type to passive gave me one hour extra on battery, now 18 hours. This isn't much but  nice to have!

thanks,
Erwin

shmuelzon, 2019-03-26:
Nice!

You might be able to scrape a bit more by lowering the BLE transmit power (depending on the distance between the ESP32 and your BLE peripheral) by removing the call to `esp_ble_tx_power_set()` in [main/ble.c](https://github.com/shmuelzon/esp32-ble2mqtt/blob/671f084b02329196cb7d2e72383486912bcffc58/main/ble.c#L790).

commit 671f084b (2019-03-23): Disable WiFi modem sleep

This will lower the latency a bit when sending commands (read/write) to the
ESP32.

Fixes #29