shmuelzon/esp32-ble2mqtt #30: Crash while connecting to miscale

bilogic, 2019-04-21:
Hi,

I'm getting this from my MI_SCALE. Any idea?
Thank you.

```
I (12533) BLE2MQTT: Discovered BLE device: c8:0f:10:c7:6f:e1, connecting
I (18433) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
I (19003) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/DeviceName = MI_SCALE
I (19063) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/Appearance = 3200
I (19093) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPrivacyFlag = false
I (19183) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPreferredConnectionParameters = 100,200,0,2000
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d66ca  PS      : 0x00060430  A0      : 0x800d4f05  A1      : 0x3ffe9ad0
0x400d66ca: chartoa at E:/projectx/site/iot/esp32-ble2mqtt/main/ble_utils.c:765

A2      : 0x00000000  A3      : 0x3ffcc4e8  A4      : 0x400d66c7  A5      : 0x00000000
0x400d66c7: chartoa at E:/projectx/site/iot/esp32-ble2mqtt/main/ble_utils.c:765

A6      : 0x3ffc0e28  A7      : 0x00000000  A8      : 0x800d64fc  A9      : 0x3ffe9ab0
A10     : 0x3ffc0e28  A11     : 0x00000018  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x3ffcc910  A15     : 0x3f406040  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000005  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x8017ea37

Backtrace: 0x400d66ca:0x3ffe9ad0 0x400d4f02:0x3ffe9b00 0x400d4fd9:0x3ffe9b30 0x400d5009:0x3ffe9b50 0x40093d55:0x3ffe9b80
0x400d66ca: chartoa at E:/projectx/site/iot/esp32-ble2mqtt/main/ble_utils.c:765

0x400d4f02: ble_on_device_characteristic_value at E:/projectx/site/iot/esp32-ble2mqtt/main/ble2mqtt.c:420

0x400d4fd9: ble2mqtt_handle_event at E:/projectx/site/iot/esp32-ble2mqtt/main/ble2mqtt.c:537

0x400d5009: ble2mqtt_task at E:/projectx/site/iot/esp32-ble2mqtt/main/ble2mqtt.c:559

0x40093d55: vPortTaskWrapper at E:/portable/esp32/esp-idf/components/freertos/port.c:403

Rebooting...
```

shmuelzon, 2019-04-21:
Hey,

It seems there is a problem converting the data received from the scale to a human-readable string before publishing it via MQTT.
In order to try and figure out what's wrong, I would need to know which service/characteristic was received and the payload. Would you mind adding the following 3 lines to the beginning of the `ble_on_device_characteristic_value()` in `main/ble2mqtt.c` and post the output just before the crash?
```patch
diff --git a/main/ble2mqtt.c b/main/ble2mqtt.c
index 6853642..7ffedcb 100644
--- a/main/ble2mqtt.c
+++ b/main/ble2mqtt.c
@@ -416,6 +416,9 @@ static void ble_on_device_characteristic_value(mac_addr_t mac,
     ble_uuid_t service, ble_uuid_t characteristic, uint8_t *value,
     size_t value_len)
 {
+    ESP_LOGI(TAG, "Parsing service: %s", uuidtoa(service));
+    ESP_LOGI(TAG, "Parsing characterisic: %s", uuidtoa(characteristic));
+    ESP_LOG_BUFFER_HEX(TAG, value, value_len);
     char *topic = ble_topic(mac, service, characteristic);
     char *payload = chartoa(characteristic, value, value_len);
     size_t payload_len = strlen(payload);
```

Thanks

bilogic, 2019-04-21:
1 moment please :)

bilogic, 2019-04-21:
This is what I got.
```
I (23961) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
I (24611) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (24611) BLE2MQTT: Parsing characterisic: 00002a00-0000-1000-8000-00805f9b34fb
I (24611) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (24621) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/DeviceName = MI_SCALE
I (24701) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (24701) BLE2MQTT: Parsing characterisic: 00002a01-0000-1000-8000-00805f9b34fb
I (24701) BLE2MQTT: 80 0c
I (24711) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/Appearance = 3200
I (24771) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (24771) BLE2MQTT: Parsing characterisic: 00002a02-0000-1000-8000-00805f9b34fb
I (24781) BLE2MQTT: 00
I (24781) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPrivacyFlag = false
I (24801) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (24801) BLE2MQTT: Parsing characterisic: 00002a04-0000-1000-8000-00805f9b34fb
I (24811) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (24811) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPreferredConnectionParameters = 100,200,0,2000
I (24891) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (24891) BLE2MQTT: Parsing characterisic: 00002a03-0000-1000-8000-00805f9b34fb
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d682e  PS      : 0x00060430  A0      : 0x800d5069  A1      : 0x3ffe9080
```

bilogic, 2019-04-21:
Does it help? It seems that ```ESP_LOG_BUFFER_HEX``` for ```GenericAccess/PeripheralPreferredConnectionParameters``` did not manage to print.

shmuelzon, 2019-04-21:
Actually, `GenericAccess/PeripheralPreferredConnectionParameters` did manage to print, it was `GenericAccess/ReconnectionAddress` (00002a03-0000-1000-8000-00805f9b34fb) that failed.
It should be a uint48 (6 bytes).
It's very odd though, it's as if the memory is invalid, but we were able to copy the payload to it just a moment earlier...
Would you mind applying the following changes? (Sorry in advance for the ping-pong)
```patch
diff --git a/main/ble2mqtt.c b/main/ble2mqtt.c
index 6853642..0ac093a 100644
--- a/main/ble2mqtt.c
+++ b/main/ble2mqtt.c
@@ -416,6 +416,10 @@ static void ble_on_device_characteristic_value(mac_addr_t mac,
     ble_uuid_t service, ble_uuid_t characteristic, uint8_t *value,
     size_t value_len)
 {
+    ESP_LOGI(TAG, "Parsing service: %s", uuidtoa(service));
+    ESP_LOGI(TAG, "Parsing characterisic: %s", uuidtoa(characteristic));
+    ESP_LOGI(TAG, "Payload at %p (%u)", value, value_len);
+    ESP_LOG_BUFFER_HEX(TAG, value, value_len);
     char *topic = ble_topic(mac, service, characteristic);
     char *payload = chartoa(characteristic, value, value_len);
     size_t payload_len = strlen(payload);
@@ -701,6 +705,10 @@ static void _ble_on_device_characteristic_value(mac_addr_t mac,
 {
     event_t *event = malloc(sizeof(*event));
 
+    ESP_LOGI(TAG, "Got service: %s", uuidtoa(service));
+    ESP_LOGI(TAG, "Got characterisic: %s (%u)", uuidtoa(characteristic), value_len);
+    ESP_LOG_BUFFER_HEX(TAG, value, value_len);
+
     event->type = EVENT_TYPE_BLE_DEVICE_CHARACTERISTIC_VALUE;
     memcpy(event->ble_device_characteristic_value.mac, mac, sizeof(mac_addr_t));
     memcpy(event->ble_device_characteristic_value.service, service,
@@ -708,8 +716,10 @@ static void _ble_on_device_characteristic_value(mac_addr_t mac,
     memcpy(event->ble_device_characteristic_value.characteristic,
         characteristic, sizeof(ble_uuid_t));
     event->ble_device_characteristic_value.value = malloc(value_len);
+    ESP_LOGI(TAG, "Copying payload to: %p", event->ble_device_characteristic_value.value);
     memcpy(event->ble_device_characteristic_value.value, value, value_len);
     event->ble_device_characteristic_value.value_len = value_len;
+    ESP_LOG_BUFFER_HEX(TAG, event->ble_device_characteristic_value.value, event->ble_device_characteristic_value.value_len);
 
     xQueueSend(event_queue, &event, portMAX_DELAY);
 }
```

bilogic, 2019-04-21:
Hi,

Here you go! Btw can you give me a quick primer on how to apply your patch using git?
Currently, I'm manually updating the lines. 

The other thing is that the scale is known to disconnect by itself after some 20 seconds or so, if it helps.

Thank you.
```
I (10090) BLE2MQTT: Connected to device: c8:0f:10:c7:6f:e1, scanning
I (10720) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (10720) BLE2MQTT: Got characterisic: 00002a00-0000-1000-8000-00805f9b34fb (8)
I (10730) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (10730) BLE2MQTT: Copying payload to: 0x3ffba904
I (10740) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (10740) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (10750) BLE2MQTT: Parsing characterisic: 00002a00-0000-1000-8000-00805f9b34fb
I (10760) BLE2MQTT: Payload at 0x3ffba904 (8)
I (10760) BLE2MQTT: 4d 49 5f 53 43 41 4c 45
I (10770) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/DeviceName = MI_SCALE
I (10800) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (10800) BLE2MQTT: Got characterisic: 00002a01-0000-1000-8000-00805f9b34fb (2)
I (10800) BLE2MQTT: 80 0c
I (10810) BLE2MQTT: Copying payload to: 0x3ffba900
I (10810) BLE2MQTT: 80 0c
I (10820) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (10820) BLE2MQTT: Parsing characterisic: 00002a01-0000-1000-8000-00805f9b34fb
I (10830) BLE2MQTT: Payload at 0x3ffba900 (2)
I (10840) BLE2MQTT: 80 0c
I (10840) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/Appearance = 3200
I (10840) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (10850) BLE2MQTT: Got characterisic: 00002a02-0000-1000-8000-00805f9b34fb (1)
I (10860) BLE2MQTT: 00
I (10870) BLE2MQTT: Copying payload to: 0x3ffba8c8
I (10870) BLE2MQTT: 00
I (10870) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (10880) BLE2MQTT: Parsing characterisic: 00002a02-0000-1000-8000-00805f9b34fb
I (10890) BLE2MQTT: Payload at 0x3ffba8c8 (1)
I (10890) BLE2MQTT: 00
I (10900) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPrivacyFlag = false
I (10950) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (10950) BLE2MQTT: Got characterisic: 00002a04-0000-1000-8000-00805f9b34fb (8)
I (10950) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (10960) BLE2MQTT: Copying payload to: 0x3ffba904
I (10960) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (10970) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (10980) BLE2MQTT: Parsing characterisic: 00002a04-0000-1000-8000-00805f9b34fb
I (10980) BLE2MQTT: Payload at 0x3ffba904 (8)
I (10990) BLE2MQTT: 64 00 c8 00 00 00 d0 07
I (10990) BLE2MQTT: Publishing: c8:0f:10:c7:6f:e1/GenericAccess/PeripheralPreferredConnectionParameters = 100,200,0,2000
I (11020) BLE2MQTT: Got service: 00001800-0000-1000-8000-00805f9b34fb
I (11020) BLE2MQTT: Got characterisic: 00002a03-0000-1000-8000-00805f9b34fb (0)
I (11030) BLE2MQTT: Copying payload to: 0x0
I (11030) BLE2MQTT: Parsing service: 00001800-0000-1000-8000-00805f9b34fb
I (11040) BLE2MQTT: Parsing characterisic: 00002a03-0000-1000-8000-00805f9b34fb
I (11050) BLE2MQTT: Payload at 0x0 (0)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d68ce  PS      : 0x00060430  A0      : 0x800d5109  A1      : 0x3ffe8fd0
```

shmuelzon, 2019-04-22:
Hey,

So the issue is that the application receives a zero-length value for that characteristic and I never accounted for that scenario. I'll need to come up with a nice solution but, in the meantime, you can make the following changes just to get yourself started:
```patch
diff --git a/main/ble_utils.c b/main/ble_utils.c
index 45db0a5..6db52de 100644
--- a/main/ble_utils.c
+++ b/main/ble_utils.c
@@ -238,6 +238,11 @@ char *chartoa(ble_uuid_t uuid, const uint8_t *data, size_t len)
     char *p = buf;
     int i = 0;
 
+    if (len == 0)
+    {
+        buf[0] = '\0';
+        return buf;
+    }
     /* A note from the Bluetooth specification:
      * If a format is not a whole number of octets, then the data shall be
      * contained within the least significant bits of the value, and all other
```

As for applying the patches, as far as I know, applying a patch using git (`git am /path/to/file.patch`) would generate a commit and probably expects a commit message in the patch file and everything.
You can use the `patch` command line utility directly, though: `patch -p1 < /path/to/file.patch` or just `patch -p1`, paste the contents of the patch file shown above and hit ctrl+d to signify the end of file. The `-p1` flag tells `patch` to strip 1 component from the path (e.g. `a/main/ble_utils.c`) so it would look for a file called `main/ble_utils.c`.

bilogic, 2019-04-22:
Alright, it does not hang any more and thanks for the patch tips too!

bilogic, 2019-04-22:
I have another issue on #2, not sure if you can see it as it is closed.
Thank you.

shmuelzon, 2019-04-22:
Reopening until a fix is committed.

commit 43d233b5 (2019-04-23): Verify data length when parsing characteristic

Fixes #30
closed 2019-04-23 by shmuelzon