shmuelzon/esp32-ble2mqtt #32: Queued and dequeued operations mismatch

bilogic, 2019-04-28:
Hi,

1. I'm trying to troubleshoot indications callback and have added an ```id``` field to ```struct operation```.
2. I used a global static ```queue_id``` which is incremented inside ```ble_operation_enqueue()```
3. Notice how the characteristics don't match up in the Enqueue vs Dequeue

```
W (23350) BLE: Enqueue: id: 2, type: 0, device: 28:fd:80:07:b5:25, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
```
vs
```
W (24700) BLE: Dequeue: id: 2, type: 0, device: 28:fd:80:07:b5:25, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
```
Could you verify this at your end? Thank you.

Full logs:
```
D (17220) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (17320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (17430) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (17530) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (17530) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (18940) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (20780) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (21700) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (21800) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (21810) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (21910) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (21910) BLE2MQTT: Discovered BLE device: 28:fd:80:07:b5:25, connecting
D (21920) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
W (22120) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1
W (22120) BLE: GATTC event 40 wasn't handled
W (22120) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
D (22130) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (22250) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22430) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22540) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22850) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (22960) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (23060) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (23110) BLE: Received GATTC event 18 (ESP_GATTC_CFG_MTU_EVT), gattc_if 1
I (23110) BLE2MQTT: Connected to device: 28:fd:80:07:b5:25, scanning
D (23120) MQTT: Subscribing to 28:fd:80:07:b5:25/Connected
D (23120) MQTT_CLIENT: mqtt_enqueue id: 33995, type=8 successful
D (23130) MQTT_CLIENT: Sent subscribe topic=28:fd:80:07:b5:25/Connected, id: 53441, type=8 successful
W (23140) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (23140) BLE: GATTC event 7 wasn't handled
W (23150) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (23160) BLE: GATTC event 7 wasn't handled
W (23160) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
W (23170) BLE: GATTC event 7 wasn't handled
D (23170) MQTT_CLIENT: msg_type=9, msg_id=53441
W (23170) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (23180) MQTT_CLIENT: msg_type=9, msg_id=53441
W (23180) BLE: GATTC event 7 wasn't handled
D (23190) MQTT_CLIENT: pending_id=53441, pending_msg_count = 1
W (23190) BLE: Received GATTC event 6 (ESP_GATTC_SEARCH_CMPL_EVT), gattc_if 1
D (23200) MQTT_CLIENT: Subscribe successful
D (23210) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (23210) BLE2MQTT: Services discovered on device: 28:fd:80:07:b5:25
D (23220) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (23230) BLE2MQTT: Found new characteristic!
D (23240) BLE2MQTT:   Service: 00001800-0000-1000-8000-00805f9b34fb
D (23240) BLE2MQTT:   Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
D (23250) MQTT: Subscribing to 28:fd:80:07:b5:25/GenericAccess/DeviceName/Get
D (23260) MQTT_CLIENT: mqtt_enqueue id: 53441, type=8 successful
D (23260) MQTT_CLIENT: Sent subscribe topic=28:fd:80:07:b5:25/GenericAccess/DeviceName/Get, id: 19104, type=8 successful
D (23270) MQTT_CLIENT: msg_type=9, msg_id=19104
[... 181 lines cut ...]
D (28140) MQTT_CLIENT: msg_type=11, msg_id=58361
D (28140) MQTT_CLIENT: pending_id=58361, pending_msg_count = 1
D (28150) MQTT_CLIENT: UnSubscribe successful
D (28160) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28370) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28480) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28580) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28670) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28690) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28780) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28790) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (28900) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29000) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29110) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29220) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29420) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29520) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29620) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29720) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29800) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (29930) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (30030) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (30130) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
```

bilogic, 2019-04-28:
I added another ESP_LOGW inside ```ble_operation_perform()```, to my surprise, it matches Enqueue.
And somehow, it is only the characteristics that is wrong, the other values, e.g. type, is correct.
Basically,

Enquene11: 0x2935
Dequeue11: 0x2950
Perform11: 0x2935

This makes debugging (this and other issues) very challenging as I don't know when the logs can be trusted.

```
W (40300) BLE: Enqueue: id: 10, type: 0, device: 28:fd:80:07:b5:25, char: 00002a50-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
D (40320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (40300) MQTT_CLIENT: msg_type=9, msg_id=47957
D (40320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (40320) BLE2MQTT: Found new characteristic!
D (40330) MQTT_CLIENT: pending_id=47957, pending_msg_count = 1
D (40340) BLE2MQTT:   Service: 00001810-0000-1000-8000-00805f9b34fb
D (40350) MQTT_CLIENT: Subscribe successful
D (40350) BLE2MQTT:   Characteristic: 00002a35-0000-1000-8000-00805f9b34fb
W (40360) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
I (40360) BLE: Indication registered: 00002a35-0000-1000-8000-00805f9b34fb
W (40380) BLE: Enqueue: id: 11, type: 2, device: 28:fd:80:07:b5:25, char: 00002a35-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3ffbab28
D (40390) BLE2MQTT: Found new characteristic!
D (40390) BLE2MQTT:   Service: 00001810-0000-1000-8000-00805f9b34fb
D (40400) BLE2MQTT:   Characteristic: 00002a49-0000-1000-8000-00805f9b34fb
D (40840) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (40890) BLE: Queue timer expired
W (40890) BLE: Dequeue: id: 1, type: 0, device: 28:fd:80:07:b5:25, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (40890) BLE: Perform: id: 1, type: 0, device: 28:fd:80:07:b5:25, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
D (40950) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (40950) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (40950) BLE: Dequeue: id: 2, type: 0, device: 28:fd:80:07:b5:25, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (40950) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/GenericAccess/DeviceName = BC85
W (40960) BLE: Perform: id: 2, type: 0, device: 28:fd:80:07:b5:25, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (41130) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41130) BLE: Dequeue: id: 3, type: 0, device: 28:fd:80:07:b5:25, char: 00002a01-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41130) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/GenericAccess/Appearance = 0
W (41140) BLE: Perform: id: 3, type: 0, device: 28:fd:80:07:b5:25, char: 00002a02-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (41250) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41250) BLE: Dequeue: id: 4, type: 0, device: 28:fd:80:07:b5:25, char: 00002a02-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41250) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/GenericAccess/PeripheralPrivacyFlag = false
W (41260) BLE: Perform: id: 4, type: 0, device: 28:fd:80:07:b5:25, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
D (41280) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (41350) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (41360) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41360) BLE: Dequeue: id: 5, type: 0, device: 28:fd:80:07:b5:25, char: 00002a04-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41360) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/GenericAccess/PeripheralPreferredConnectionParameters = 8,16,0,100
W (41370) BLE: Perform: id: 5, type: 0, device: 28:fd:80:07:b5:25, char: 00002a29-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
D (41390) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
W (41460) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41460) BLE: Dequeue: id: 6, type: 0, device: 28:fd:80:07:b5:25, char: 00002a29-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41460) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/DeviceInformation/ManufacturerNameString = Dialog Semi
W (41470) BLE: Perform: id: 6, type: 0, device: 28:fd:80:07:b5:25, char: 00002a24-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (41570) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41570) BLE: Dequeue: id: 7, type: 0, device: 28:fd:80:07:b5:25, char: 00002a24-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41570) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/DeviceInformation/ModelNumberString = DA14580
W (41580) BLE: Perform: id: 7, type: 0, device: 28:fd:80:07:b5:25, char: 00002a26-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (41750) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41750) BLE: Dequeue: id: 8, type: 0, device: 28:fd:80:07:b5:25, char: 00002a26-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41750) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/DeviceInformation/FirmwareRevisionString = v8.1
W (41760) BLE: Perform: id: 8, type: 0, device: 28:fd:80:07:b5:25, char: 00002a28-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
W (41850) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
W (41850) BLE: Dequeue: id: 9, type: 0, device: 28:fd:80:07:b5:25, char: 00002a28-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
I (41850) BLE2MQTT: Publishing: 28:fd:80:07:b5:25/DeviceInformation/SoftwareRevisionString = v_3.150.1.5
[... cut ...]
D (45090) MQTT_CLIENT: unsubscribe, topic"28:fd:80:07:b5:25/Connected", id: 20069
D (45100) MQTT_CLIENT: Sent Unsubscribe topic=28:fd:80:07:b5:25/Connected, id: 20069, successful
D (45120) MQTT_CLIENT: msg_type=11, msg_id=20069
D (45120) MQTT_CLIENT: msg_type=11, msg_id=20069
D (45120) MQTT_CLIENT: pending_id=20069, pending_msg_count = 1
D (45120) MQTT_CLIENT: UnSubscribe successful
D (45140) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (45240) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
```

bilogic, 2019-04-28:
This is very troubling, I disabled ```Enqueue```, now ```Perform``` is wrong.

Earlier
Enqueue: 0x2935
Dequeue: 0x2950
Perform: 0x2935

Now
Enqueue: 0x2935
// Dequeue: N/A
Perform: 0x2950

I have no idea what is being actually sent.

My real issue with a blood pressure device is:

1. The device stores values in its memory and can be retrieved by registering for indications from characteristics 0x2A35 and sending 0x2902
2. On another piece of BLE ESP32 code, after doing .1, I get a series of callback via ```ESP_GATTC_NOTIFY_EVT```
3. However, in ```esp32-ble2mqtt```, after doing .1, there are no ```ESP_GATTC_NOTIFY_EVT``` events received

shmuelzon, 2019-04-28:
Hey,

So, first off, the thing to note about the `uuidtoa()` function is that it's not really thread-safe. It returns a pointer to a static buffer so each call won't allocate and the caller won't need to free anything. This is usually not a problem since it's mainly used for logging and usually most logs are disabled. If you want to make sure you print what you expect then you'll need the following replacement:
```c
    ESP_LOGD(TAG, "  Service: %s", uuidtoa(service_uuid));
```
With:
```c
    ESP_LOGD(TAG, "  Service: %02x%02x%02x%02x-%02x%02x-%02x%02x-%02x%02x-%02x%02x%02x%02x%02x%02x", uuid[15], uuid[14], uuid[13], uuid[12], uuid[11], uuid[10], uuid[9], uuid[8], uuid[7], uuid[6], uuid[5], uuid[4], uuid[3], uuid[2], uuid[1], uuid[0]);
```
If you're going to do this multiple times, you should probably do the following:
```c
#define UUID_FMT "%02x%02x%02x%02x-%02x%02x-%02x%02x-%02x%02x-%02x%02x%02x%02x%02x%02x"
#define UUID_PARAM(uuid) uuid[15], uuid[14], uuid[13], uuid[12], uuid[11], uuid[10], uuid[9], uuid[8], uuid[7], uuid[6], uuid[5], uuid[4], uuid[3], uuid[2], uuid[1], uuid[0]
ESP_LOGD(TAG, "  Service " UUID_FMT, UUID_PARAM(service_uuid));
```

I think (and hope) you'll find that they will match.
Let me know if not...

Good day!

bilogic, 2019-04-28:
Ohh.. ok let me try thanks!

bilogic, 2019-04-28:
Ok, it works. I will open another issue for my actual problem. Thank you.

commit 9aaac59e (2019-04-29): Don't use uuidtoa/mactoa with formatting functions

Fixes #32
closed 2019-04-28 by bilogic