shmuelzon/esp32-ble2mqtt #34: Compiler errors with newer version of ESP-IDF

qanh96, 2019-05-03:
Hello and many thanks for this quality project
With the slight older ESP-IDF from Espressif everything work fine, but i'm having compiler Errors with the newer version when try to build a new clone of the project
Ich have already tried things like fresh installation of IDF or fresh clone of project and confirmed the same problems on multiple devices.

Thanks for your hardwork!
[Build-logs.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/3143486/Build-logs.txt)

shmuelzon, 2019-05-04:
Hey,

It seems that commit https://github.com/espressif/esp-idf/commit/d15e18aa5db53f55344bbb4a7623d3dd46b6ad0f (changes done on components/wpa_supplicant/port/include/endian.h) is what broke things. It removed the defines needed to actually map the endianess conversion functions to something useful.
You could apply the following patch and the project should compile:
```patch
diff --git a/main/ble_utils.h b/main/ble_utils.h
index 6c6c181..88b1d03 100644
--- a/main/ble_utils.h
+++ b/main/ble_utils.h
@@ -4,6 +4,10 @@
 #include <esp_gap_ble_api.h>
 #include <esp_gattc_api.h>
 
+#define __bswap16     __bswap_16
+#define __bswap32     __bswap_32
+#define __bswap64     __bswap_64
+
 #define MAC_FMT "%02x:%02x:%02x:%02x:%02x:%02x"
 #define MAC_PARAM(mac) mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]
 
```
I, generally, try to stay on official releases so these kinds of things won't happen. Currently, as is described in the README file, I'm working against ESP-IDF v3.2 and that's what I can really support.
I've seen unstable versions come with all sorts of weird behavior and stability problems.

Out of curiosity, do you have any reason to use ESP-IDF's top-of-branch instead of the latest release?

Good day!

qanh96, 2019-05-04:
Thank you for your help, i will stick to the stable v3.2 to build the project.

As for reson to use the lastest Version, we want to try out the some of the new functions that supposed to give us more control to the energie consumption of the ESP32 and see how can we create sth that consume as little energie as possible. 

Thanks again for this well made and active supported Projects
Best regard!

closed 2019-05-04 by qanh96