shmuelzon/esp32-ble2mqtt #35: TLS support

plett, 2019-05-16:
How feasible would it be to add TLS support for the connection to the MQTT server?

I see that ESP-IDF has [example code for it](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl) but I don't know if it's just a case of copying the code over and adding some configuration entries for it, or if it's more involved than that.

Is it likely to grow the image size and RAM footprint beyond what an ESP32 can cope with?

shmuelzon, 2019-05-17:
Hey,

> How feasible would it be to add TLS support for the connection to the MQTT server?

Enabling SSL is just a matter of configuration and could be achieved by applying the following changes:
```patch
diff --git a/main/mqtt.c b/main/mqtt.c
--- a/main/mqtt.c
+++ b/main/mqtt.c
@@ -260,6 +260,7 @@ int mqtt_connect(const char *host, uint16_t port, const char *client_id,
         .client_id = client_id,
         .username = username,
         .password = password,
+        .transport = MQTT_TRANSPORT_OVER_SSL,
     };
 
     ESP_LOGI(TAG, "Connecting MQTT client");
```

Doing it properly would include adding this option to the configuration file and also adding support for loading certificates. The later would take a bit more time since it would also require changing OTA updates to include the entire flash file system and not just the configuration file.
I'll try to get around to implementing this in the near future...

> Is it likely to grow the image size and RAM footprint beyond what an ESP32 can cope with?

I don't think there will be a big increase since all the TLS support is already a part of the image.

Good day!

shmuelzon, 2019-05-31:
I've pushed initial support of SSL, without supporting certificates yet.
I'll keep this issue open until that is ready as well.

commit 9678de0e (2019-05-31): Add support for MQTT of SSL

Fixes #35
commit 165b5bba (2019-06-14): Add support for MQTT certificates

Fixes #35
closed 2019-06-14 by shmuelzon