shmuelzon/esp32-ble2mqtt #36: Is it possible compile by Mongoose-OS?

lourivalbarreto, 2019-05-23:
Hi,

First of all, congratulations for Ble2Mqtt bridge. Its really nice!

As I have others IoT firmwares using Mongoose-OS connected directly to AWS IoT Mqtt, I would like to know if is possible I compile your esp32-ble2mqtt firmware by Mongoose-OS and connect it to AWS IoT mqtt server?

Thanks a lot!
Jose Lima

shmuelzon, 2019-05-23:
Hey Jose,

Thanks for the kind words.
From a quick look at Mongoose OS, I don’t think the code will run as is via Mongoose OS.
As for the second part, what’s stopping you from connecting this project to your AWS MQTT broker? Should be just a matter of configuration, no?

Good day

lourivalbarreto, 2019-05-23:
I didn't try yet to connect it to AWS IoT.
But I really would like to compile it from Mongoose-OS to maintain my development pattern. Could you help me with this deal?

Thanks a lot!
Jose

shmuelzon, 2019-05-26:
Hey Jose,

As far as I can tell, using Mongoose OS would be a complete rewrite of this project so it's not something I can help with.

Sorry

closed 2019-05-26 by shmuelzon