shmuelzon/esp32-ble2mqtt #37: Can't find spiffs.bin

louielong, 2019-06-05:
Hi,
 
I saw the image download command is 

`python /root/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 /root/esp/esp32-ble2mqtt/build/ota_data_initial.bin 0x1000 /root/esp/esp32-ble2mqtt/build/bootloader/bootloader.bin 0x10000 /root/esp/esp32-ble2mqtt/build/ble2mqtt.bin 0x8000 /root/esp/esp32-ble2mqtt/build/partitions.bin 0x310000 /root/esp/esp32-ble2mqtt/build/spiffs.bin`

But, I cann't found spiffs.bin file, is it correctly？

shmuelzon, 2019-06-05:
Hey,

`spiffs.bin` should be built when running `make flash`.

Good day

shmuelzon, 2019-06-08:
@louielong Does that help? Were you able to locate it?

louielong, 2019-06-09:
Hi,

Sorry for response you so late, the 'make flash' does works. 
And I have check this on ubuntu and Macos.

Finally, thank you very much for your response.

shmuelzon, 2019-06-09:
Great, glad to hear it worked out for you!

closed 2019-06-09 by shmuelzon