shmuelzon/esp32-ble2mqtt #38: Support for e-ink Xiaomi BLE Thermometer LYWSD02 Model?

jamesyym, 2019-06-10:
Actually ESPHome does not support Xiaomi BLE Thermometer LYWSD02 Model

There is a custom_components that works for this model: https://github.com/h4/LYWSD02-home-assistant. And also be supported with OMG: https://github.com/1technophile/OpenMQTTGateway

It would be great if there is official support for this model

Thank you.

shmuelzon, 2019-06-10:
Hey,

Not a lot to go on here but, as far as I can tell, the device provides a characteristic with the temperature and humidity values. It should work as is as far as connecting, reading and listening on notifications.

What you can do is add the definitions for the characteristic so they are published in a more readable format on the MQTT bus.

It's hard to tell without investigating a bit more (and having the device to test) but the following might be enough:
```json
{
  "ble": {
    "characteristics": {
      "definitions": {
        "ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6": {
          "name": "TemperatueHumidity",
          "types": [
            "sint16",
            "uint16",
          ]
        }
    }
  }
}
```

Note that I'm guessing regarding the actual signed/unsigned types.

Let me know...

jamesyym, 2019-06-10:
James@Chromebook MINGW32 ~/esp/esp32-ble2mqtt
$ make upload-config
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Uploading configuration /home/James/esp/esp32-ble2mqtt/data/config.json to BLE2MQTT
Listening on port 8000
11/Jun/2019 02:15:49 - (BLE2MQTT-F224: BLE2MQTT/v0.10.0-2-g9678de0) GET /
11/Jun/2019 02:15:49 - (BLE2MQTT-F224: BLE2MQTT/v0.10.0-2-g9678de0) Done: 200 OK
No new connection requests, shutting down

James@Chromebook MINGW32 ~/esp/esp32-ble2mqtt
$ make remote-monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Listening on port 5000
2019-06-11 02:16:10.524640 (BLE2MQTT-F224): I (19512) BLE2MQTT: Connected to device: 3f:59:c8:82:88:29, scanning
2019-06-11 02:16:11.449334 (BLE2MQTT-F224): I (20392) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/GenericAccess/DeviceName = LYWSD02
2019-06-11 02:16:11.449334 (BLE2MQTT-F224): I (20512) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/GenericAccess/Appearance = 0
2019-06-11 02:16:11.746332 (BLE2MQTT-F224): I (20712) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/GenericAttribute/ServiceChanged = 1,65535

shmuelzon, 2019-06-11:
This looks similar to #33. In short, for some reason, the application doesn't get a response from the ESP BLE library to one of the read/write operations and it doesn't continue issuing BLE commands.
Could you please enable the debug log level and post the resulting logs?
If it is something similar to #33, then you can apply the same workaround and blacklist the offending characteristic.

Where can I find one of these LYWSD02? How much do they cost? If this is indeed the same issue, getting one would help a lot with fixing the root cause.

shmuelzon, 2019-06-11:
I think that, for now, blacklisting the `00002a05-0000-1000-8000-00805f9b34fb` (Service Changed) characteristic would make it workable for you. Add the following to your configuration file:

```json
{
  "ble": {
    "characteristics": {
      "blacklist": [ "00002a05-0000-1000-8000-00805f9b34fb" ]
    }
  }
}
```

jamesyym, 2019-06-12:
> I think that, for now, blacklisting the `00002a05-0000-1000-8000-00805f9b34fb` (Service Changed) characteristic would make it workable for you. Add the following to your configuration file:
>
> ```json [...]

I've tried this first :)

James@Chromebook MINGW32 ~/esp/esp32-ble2mqtt
$ make upload-config
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Uploading configuration /home/James/esp/esp32-ble2mqtt/data/config.json to BLE2MQTT
Listening on port 8000
12/Jun/2019 16:26:48 - (BLE2MQTT-F224: BLE2MQTT/v0.10.0-2-g9678de0) GET /
12/Jun/2019 16:26:48 - (BLE2MQTT-F224: BLE2MQTT/v0.10.0-2-g9678de0) Done: 200 OK
No new connection requests, shutting down

James@Chromebook MINGW32 ~/esp/esp32-ble2mqtt
$ make remote-monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Listening on port 5000
2019-06-12 16:27:01.968024 (BLE2MQTT-F224): I (12372) BLE2MQTT: Discovered BLE device: 98:bb:99:68:c9:fb, not connecting
2019-06-12 16:27:07.815407 (BLE2MQTT-F224): I (18092) BLE2MQTT: Connected to device: 3f:59:c8:82:88:29, scanning
2019-06-12 16:27:08.728485 (BLE2MQTT-F224): I (19012) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/GenericAccess/DeviceName = LYWSD02
2019-06-12 16:27:08.728485 (BLE2MQTT-F224): I (19132) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/GenericAccess/Appearance = 0
2019-06-12 16:27:09.046202 (BLE2MQTT-F224): I (19292) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/ManufacturerNameString = miaomiaoce.com
2019-06-12 16:27:09.046202 (BLE2MQTT-F224): I (19412) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/ModelNumberString = LYWSD02
2019-06-12 16:27:09.345362 (BLE2MQTT-F224): I (19612) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/SerialNumberString = F200FFFF0077C7A15C
2019-06-12 16:27:09.345362 (BLE2MQTT-F224): I (19812) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/HardwareRevisionString = F2_WF
2019-06-12 16:27:09.657904 (BLE2MQTT-F224): I (19932) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/FirmwareRevisionString = 1.1.2_0019
2019-06-12 16:27:09.657904 (BLE2MQTT-F224): I (20052) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/SoftwareRevisionString = 0019
2019-06-12 16:27:09.957328 (BLE2MQTT-F224): I (20252) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/DeviceInformation/SystemID = 18446744072778582057,63
2019-06-12 16:27:10.270329 (BLE2MQTT-F224): I (20532) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/8082caa8-41a6-4021-91c6-56f9b954cc34 = 18446744072778582057,63
2019-06-12 16:27:10.270329 (BLE2MQTT-F224): I (20732) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/724249f0-5ec3-4b5f-8804-42345af08651 = 18446744072778582057,63
2019-06-12 16:27:10.582834 (BLE2MQTT-F224): I (20852) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/6c53db25-47a1-45fe-a022-7c92fb334fd4 = 18446744072778582057,63
2019-06-12 16:27:10.582834 (BLE2MQTT-F224): I (21052) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/9d84b9a3-000c-49d8-9183-855b673fda31 = 18446744072778582057,63
2019-06-12 16:27:10.878408 (BLE2MQTT-F224): I (21252) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/457871e8-d516-4ca1-9116-57d0b17b9cb2 = 18446744072778582057,63
2019-06-12 16:27:11.183816 (BLE2MQTT-F224): I (21452) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/5f78df94-798c-46f5-990a-b3eb6a065c88 = 0
2019-06-12 16:27:11.496320 (BLE2MQTT-F224): I (21772) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/64b4e8b5-0de5-401b-a21d-acc8db3b913a = 13
2019-06-12 16:27:11.496320 (BLE2MQTT-F224): I (21892) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/42c3dfdd-77be-4d9c-8454-8f875267fb3b = 244,0
2019-06-12 16:27:11.808823 (BLE2MQTT-F224): I (22092) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/0000fef5-0000-1000-8000-00805f9b34fb/b7de1eea-823d-43bb-a3af-c4903dfce23c = 200,0
2019-06-12 16:27:11.808823 (BLE2MQTT-F224): I (22292) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccb7-7a0a-4b0c-8a1a-6ff2997da3a6 = 95,183,0,93,8
2019-06-12 16:27:12.118876 (BLE2MQTT-F224): I (22412) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccb9-7a0a-4b0c-8a1a-6ff2997da3a6 = 127,1,0,0,128,1,0,0
2019-06-12 16:27:12.118876 (BLE2MQTT-F224): I (22572) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccba-7a0a-4b0c-8a1a-6ff2997da3a6 = 55,1,0,0
2019-06-12 16:27:12.425667 (BLE2MQTT-F224): I (22692) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccbb-7a0a-4b0c-8a1a-6ff2997da3a6 = 0,0,0,0,224,75,164,92,160,9,21,67,9,21,0,0
2019-06-12 16:27:12.425667 (BLE2MQTT-F224): I (22892) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccbc-7a0a-4b0c-8a1a-6ff2997da3a6 = 0,0,0,0,224,75,164,92,160,9,21,67,9,21,0,0
2019-06-12 16:27:12.722545 (BLE2MQTT-F224): I (23212) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccbc-7a0a-4b0c-8a1a-6ff2997da3a6 = 0,0,0,0,224,75,164,92,160,9,21,67,9,20
[... cut ...]
2019-06-12 16:28:42.124914 (BLE2MQTT-F224): I (112492) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6 = 55,10,65
2019-06-12 16:28:48.269450 (BLE2MQTT-F224): I (118532) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6 = 56,10,65
2019-06-12 16:29:03.018476 (BLE2MQTT-F224): I (133492) BLE2MQTT: Publishing: 3f:59:c8:82:88:29/ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6/ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6 = 58,10,65

shmuelzon, 2019-06-12:
That looks good, it connects and get's all of the data. Also registers for notifications to get new sensor readings.
I can't account for all of the data there but the majority of values are probably historic values (notice the incrementing values in the first bytes). The current readings seem to come from the `ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6` characteristic but I looks like it's 3 bytes and not 4 as I original read somewhere.
Looking at the last value (58,10,65). The first two bytes are a signed 16 bit integers (in little-endian). `58` is `0x3a`, `10` is `0x0a`, so `0x0a3a` is `2618` meaning 26.18 degrees. The last byte is probably the humidity in an unsigned 8-bit integer, simply 65% humidity. Does that seem to correspond with what's displayed on the device?
If so, the following should be the characteristic definition:
```json
{
  "ble": {
    "characteristics": {
      "definitions": {
        "ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6": {
          "name": "TemperatureHumidity",
          "types": [
            "sint16",
            "uint8",
          ]
        }
    }
  }
}
```
With that configuration, the MQTT payload would be: `2618,58`.
You should also provide a more readable name for the `ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6` service just to make the MQTT topic nicer.

I'm guessing you're using Home Assistant and probably want to display the readings there.
Just create a MQTT sensor and use a value template to parse the actual value. In my case, I have a characteristic that only includes the temperature also in 0.01 units, and I have the following in my configuration:
```yaml
sensor:
  - platform: mqtt
    name: "Temperature Sensor"
    device_class: temperature
    state_topic: "xx:xx:xx:xx:xx:xx/EnvironmentalSensing/Temperature"
    unit_of_measurement: °C
    value_template: '{{ value | multiply(0.01) | round(1) }}'
```
In your case, you would need to separate the temperature from the humidity first.

Let me know if there is anything else you need help with, otherwise, could you close this issue?

Good day!

jamesyym, 2019-06-12:
Thank you so so so much!!! I will try tomorrow and post here: )

Now I have a new question that is how to add two different devices to the config and whitelist with its own mac? Following is my config for another BLE Temp&Humid meter:

{
  "wifi": {
    "ssid": "James",
    "password": "pass"
  },
  "mqtt": {
    "server": {
      "host": "192.168.2.8",
      "port": 1883,
      "ssl": false,
      "username": "mqtt",
      "password": "pass",
      "client_id": "esp32"
    },
    "publish": {
      "qos": 0,
      "retain": true
    },
    "topics" :{
      "prefix": "",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "whitelist": [
      "68:3e:34:cc:d7:0e"
    ],
    "services": {
      "definitions": {
        "000016f0-0000-1000-8000-00805f9b34fb": {
          "name": "SensorService"
        }
      },
      "whitelist": [
        "000016f0-0000-1000-8000-00805f9b34fb"
      ]
    },
    "characteristics": {
      "definitions": {
        "000016f2-0000-1000-8000-00805f9b34fb": {
          "name": "SensorValue",
          "types": []
        }
      },
      "whitelist": [
        "000016f2-0000-1000-8000-00805f9b34fb"
      ]
    },
    "passkeys": {}
  },
  "log": {
    "host": "224.0.0.200",
    "port": 5000
  }
}

shmuelzon, 2019-06-12:
I'm not sure I follow. You want to add two BLE devices to the whitelist? So it will connect only to them?
If so, then the MAC address whitelist is a JSON array, just add more.
```json
{
  "ble": {
    "whitelist": [
      "68:3e:34:cc:d7:0e",
      "68:3e:34:cc:d7:0f",
      "68:3e:34:cc:d7:10",
    ],
  }
}
```

jamesyym, 2019-06-12:
Yes. I've added service and characteristic with the whitelist MAC address "68:3e:34:cc:d7:0e".

But I don't know how to add the characteristic of LYWSD02 of its MAC address "3f:59:c8:82:88:29" with the oirg service and characteristics I post above.

Because I replaced the ble part everytime when I try the test of LYWSD02😅

shmuelzon, 2019-06-12:
Something like this?
```json
{
    "wifi": {
        "ssid": "James",
        "password": "pass"
    },
    "mqtt": {
        "server": {
            "host": "192.168.2.8",
            "port": 1883,
            "ssl": false,
            "username": "mqtt",
            "password": "pass",
            "client_id": "esp32"
        },
        "publish": {
            "qos": 0,
            "retain": true
        },
        "topics": {
            "prefix": "",
            "get_suffix": "/Get",
            "set_suffix": "/Set"
        }
    },
    "ble": {
        "whitelist": [
            "68:3e:34:cc:d7:0e",
            "3f:59:c8:82:88:29"
        ],
        "services": {
            "definitions": {
                "000016f0-0000-1000-8000-00805f9b34fb": {
                    "name": "SensorService"
                },
                "ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6": {
                    "name": "SensorService"
                }
            },
            "whitelist": [
                "000016f0-0000-1000-8000-00805f9b34fb",
                "ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"
            ]
        },
        "characteristics": {
            "definitions": {
                "000016f2-0000-1000-8000-00805f9b34fb": {
                    "name": "SensorValue",
                    "types": []
                },
                "ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6": {
                    "name": "SensorValue",
                    "types": [
                        "sint16",
                        "uint8"
                    ]
                }
            },
            "whitelist": [
                "000016f2-0000-1000-8000-00805f9b34fb",
                "ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"
            ]
        },
        "passkeys": {}
    },
    "log": {
        "host": "224.0.0.200",
        "port": 5000
    }
}
```

jamesyym, 2019-06-12:
How kind of you, bro!

Should I add the blacklist 00002a05-0000-1000-8000-00805f9b34fb (Service Changed) as well?

shmuelzon, 2019-06-12:
Since you decided to use a whitelist, anything except those in the list are ignored, so the Service Changed characteristic is ignored as well.

It’s up to you if you prefer to whitelist the characteristics you are interested in or blacklist the ones you aren’t interested in.

jamesyym, 2019-06-13:
> Something like this?
>
> ```json
> {
>     "wifi": {
>         "ssid": "James",
>         "password": "pass"
>     },
>     "mqtt": {
>         "server": {
>             "host": "192.168.2.8",
>             "port": [...]

This runs perfect well. But in HA, it show "2555,68 °C" with the configuration you provide to me.
I tried to split the temperature and the humidity with the script like
```javascript
value_template: '{{ int(value.split(',')[0])  | multiply(0.01) | round(1) }}'
```
but always shows unknown.
Could you please help me with it?
```javascript
sensor:
  - platform: mqtt
    name: "Xiaomi Temperature"
    device_class: temperature
    state_topic: "3f:59:c8:82:88:29/SensorService/SensorValue"
    unit_of_measurement: °C
    value_template: '{{ value | multiply(0.01) | round(1) }}'

  - platform: mqtt
    name: "Xiaomi Humidity"
    device_class: humidity
    state_topic: "3f:59:c8:82:88:29/SensorService/SensorValue"
    unit_of_measurement: "%"
    value_template: '{{ value | multiply(0.01) | round(1) }}'

group:
  Xiaomi:
    name: Xiaomi
    view: no
    entities:
      - sensor.xiaomi_temperature
      - sensor.xiaomi_humidity
```
Thank you very much.

shmuelzon, 2019-06-13:
`value_template: '{{ value.split(",")[0] | multiply(0.01) | round(1) }}'` seems to work fine...

jamesyym, 2019-06-13:
> `value_template: '{{ value.split(",")[0] | multiply(0.01) | round(1) }}'` seems to work fine...
> 
> > This runs perfect well. BTW, is there any option for the interval scan of BLE?
> 
> No, the current implementation scans continuously.

We got it. Thank you, bro.
```javascript
sensor:
  - platform: mqtt
    name: "Xiaomi Temperature"
    device_class: temperature
    state_topic: "3f:59:c8:82:88:29/SensorService/SensorValue"
    unit_of_measurement: °C
    value_template: '{{ value.split(",")[0] | multiply(0.01) | round(1) }}'

  - platform: mqtt
    name: "Xiaomi Humidity"
    device_class: humidity
    state_topic: "3f:59:c8:82:88:29/SensorService/SensorValue"
    unit_of_measurement: "%"
    value_template: '{{ value.split(",")[1] }}'

group:
  Xiaomi:
    name: Xiaomi
    view: no
    entities:
      - sensor.xiaomi_temperature
      - sensor.xiaomi_humidity
```
[image]
[image]

closed 2019-06-13 by jamesyym

shmuelzon, 2019-06-13:
Cool! Glad to see you got it working!