shmuelzon/esp32-ble2mqtt #39: How to disconnect from BLE devices?

tiandeyu, 2019-06-14:
Hi 
It seems esp32 will try to connect BLE devices in the config.json once esp32 startup. It there any way to disconnect from BLE devices after "Get" data from BLE devices. Next time when I need to fetch data connects first. That for better energy saving.

Thank,
Natic

shmuelzon, 2019-06-14:
Hey Natic,

There's currently no way to temporarily disconnect from a device and there's also no way to request the ESP32 to connect to it at a later time. This project was meant to hide all of the BLE-specifics and just expose the data. This, IMHO, makes it easier to use as you don't need another entity to manage the devices, connect/disconnect, etc.

As for energy savings, I'm not convinced it's more efficient. It all depends on how the peripheral is configured but when connected, the peripheral can make use of slave latency to keep itself in sleep instead of constantly advertising. Connection setup and teardown also take considerable resources and, in general, push (assuming the peripheral supports notifications/indications) is considered a better approach than fetch/polling.

Good day

tiandeyu, 2019-06-14:
Thanks shmuelzon,

Very appreciate for your explaining. I got more knowledge about how peripheral works.

closed 2019-06-14 by tiandeyu