shmuelzon/esp32-ble2mqtt #40: CmakeLists.txt is missing

europacafe, 2019-07-20:
Hi,

When I tried to idf.py menuconfig, it reported error that CmakeLists.txt is missing. Am I supposed to create one? If so, what are supposed to be inside the file? and where to put it. I saw some sample project coding, CmakeLists.txt appears in two places, one in the "main" folder, and the other one outside the "main" folder.

After I created (by imitating coding from some sample project and not sure it was correct. In the CmakeLists.txt, I use ble2mqtt.c as main source, correct?), I can enter into the config, then issue command "idf.py build". It was compiling a lot before displaying errors about missing "ble2mqtt_ver". I don't know how to proceed.

Thanks in advance for your help.

shmuelzon, 2019-07-20:
Hey,

I don’t use CMake in this project, only ‘regular’ make.
So, you would need to run: `make menuconfig` and `make`. See the README for more info.
Also, please make sure to use the supported ESP-IDF version also documented inthe README file, currently 3.2 or 3.2.2.

Let me know if you have any issues with that.

europacafe, 2019-07-20:
Thanks. I use make menuconfig and got the error:
makefile:12: /make/project.mk: No such file or directory
make: *** No rule to make target `/make/project.mk'.  Stop.

shmuelzon, 2019-07-20:
It seems that the IDF_PATH environment variable, which points to where the ESP-IDF is installed, isn’t defined.

europacafe, 2019-07-20:
Thanks. I remove my esp-idf and restart over with 3.2.2. Path is now set properly. I copy your ble2mqtt master folder into /msys32/home/bthoven/esp/ folder. Run command make menuconfig. There was error message: fatal: not a git repository (or any of the parent directories): .git. Though I still see the menu and change serial to COM3.
Then make flash, but the same git error display again...though it continued compiling a lot of stuff; but eventually stopped with
......
.....
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/ble2mqtt.elf
esptool.py v2.6
make: *** No targets specified and no makefile found.  Stop.
make: Entering directory /home/bthoven/esp/esp32-ble2mqtt-master/mkspiffs make: Leaving directory /home/bthoven/esp/esp32-ble2mqtt-master/mkspiffs
make: /home/bthoven/esp/esp32-ble2mqtt-master/mkspiffs/mkspiffs: Command not found
make: *** [Makefile:34: /home/bthoven/esp/esp32-ble2mqtt-master/build/spiffs.bin] Error 127

It seems I'm almost there. Please give suggestion...thanks a lot.

shmuelzon, 2019-07-20:
Did you use git to clone the repository? There are submodules that also need to be cloned for everything to work, see the commands in the README.
In addition, I use git to find the version if this app and mkspiffs (which is a submodule) to build the filesystem that holds the configuration.

europacafe, 2019-07-20:
You are right. I use copy into the /home/bthoven/esp/ folder. I delete it and enter into /home/bthoven/esp/ folder then use git clone command to get the ble2mqtt folder. Enter into the ble2mqtt folder, run make menuconfig and no more GIT error. Set serial port to COM3 save and exit. Then I got error
make: *** No rule to make target 'menuconfig'.  Stop.
Is that a problem?

shmuelzon, 2019-07-20:
I'm not sure I understand. You still see a `make: *** No rule to make target 'menuconfig'. Stop.` error but the application builds and you were able to flash it? Does it run correctly on the ESP32?

europacafe, 2019-07-20:
I can see all bluetooth nearby devices in mqtt.fx now. I think it's ok now. Really appreciate you help. As I've never used these kind of command before. Thanks again.

shmuelzon, 2019-07-20:
No problem, glad to hear you got it working.
I'll close this issue for now but feel free to open another if something doesn't work.

closed 2019-07-20 by shmuelzon

cooldesigns, 2019-12-23:
Please specify the note in readme. Esp-idf installation is required! See instructions here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html