shmuelzon/esp32-ble2mqtt #41: error: 'sprintf' may write a terminating nul past the end of the destination [-Werror=format-overflow=]

nardev, 2019-08-05:
Hey guys, what did i do wrong here :D 

`/home/user/Desktop/esp32-ble2mqtt/main/config.c: In function 'config_log_port_get':
/home/user/Desktop/esp32-ble2mqtt/main/config.c:393:12: warning: returning 'void *' from a function with return type 'uint16_t' {aka 'short unsigned int'} makes integer from pointer without a cast [-Wint-conversion]
     return NULL;
            ^~~~
CC build/main/ota.o
CC build/main/wifi.o
CC build/main/ble2mqtt.o
CC build/main/broadcasters.o
/home/user/Desktop/esp32-ble2mqtt/main/broadcasters.c: In function 'ibeacon_metadata_get':
/home/user/Desktop/esp32-ble2mqtt/main/broadcasters.c:96:19: error: 'sprintf' may write a terminating nul past the end of the destination [-Werror=format-overflow=]
     sprintf(s, "%u", be16toh(beacon->major));
                   ^
/home/user/Desktop/esp32-ble2mqtt/main/broadcasters.c:96:5: note: 'sprintf' output between 2 and 6 bytes into a destination of size 5
     sprintf(s, "%u", be16toh(beacon->major));
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Desktop/esp32-ble2mqtt/main/broadcasters.c:98:19: error: 'sprintf' may write a terminating nul past the end of the destination [-Werror=format-overflow=]
     sprintf(s, "%u", be16toh(beacon->minor));
                   ^
/home/user/Desktop/esp32-ble2mqtt/main/broadcasters.c:98:5: note: 'sprintf' output between 2 and 6 bytes into a destination of size 5
     sprintf(s, "%u", be16toh(beacon->minor));
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [/home/user/esp/esp-idf/make/component_wrapper.mk:292: broadcasters.o] Error 1
make: *** [/home/user/esp/esp-idf/make/project.mk:584: component-main-build] Error 2
`

shmuelzon, 2019-08-05:
Hey,

Welcome!
Which ESP-IDF version are you using?

Thanks

closed 2019-08-05 by shmuelzon

nardev, 2019-08-05:
Hello,

I think it's 3.2.2

nardev, 2019-08-06:
p.s. why closing the issue?

shmuelzon, 2019-08-06:
Because it should be fixed, at least the errors you posted.
Have you tried updating and trying again?

nardev, 2019-08-06:
Thank you so much.

It builds now but i have another problem.

`AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/ble2mqtt.elf
esptool.py v2.7-dev
make: *** No targets specified and no makefile found.  Stop.
make: Entering directory /home/user/Desktop/esp32-ble2mqtt/mkspiffs make: Leaving directory /home/user/Desktop/esp32-ble2mqtt/mkspiffs
make: /home/user/Desktop/esp32-ble2mqtt/mkspiffs/mkspiffs: Command not found
make: *** [Makefile:34: /home/user/Desktop/esp32-ble2mqtt/build/spiffs.bin] Error 127
`

shmuelzon, 2019-08-06:
Did you do a recursive clone of this repository? The `mkspiffs` directory is a git submodule and needs to be cloned as well.

nardev, 2019-08-06:
ah.. :( sorry

nardev, 2019-08-06:
Just did that, i hope the only difference from default configuration in menuconfig is maybe my USB port?

nardev, 2019-08-06:
yeeeeeeeeeee

Thank You very much man!.

[image]

shmuelzon, 2019-08-06:
Cool, now let's hope that it runs correctly :)

nardev, 2019-08-07:
It does.. send something to mqtt. Just bit too muchs encapsulated project for me. But great example for the task i'm about to do (i hope so).

thnx

GOGYNAVA, 2022-05-29:
hola buenas, alguien me puede explicar que quiere decir este error?      'sprintf' may write a terminating nul past the end of the destination [-Wformat-overflow=]
   me a salido configurando un marlin para una sk4 1.4 turbo

commit e5417e2d (2019-08-05): Fixed compilation errors/warnings

Fixes #41