shmuelzon/esp32-ble2mqtt #43: Remote logging does not appear to be working

marrold, 2019-09-13:
Hi, I am trying to troubleshoot some issues with a BLE device losing connection. To do so I'd like to enable logging. I've tried various host and port combinations and ran packet captures on both the target and the router, and I can't see this log traffic.

I've done a make clean / make flash etc - any ideas?

Redacted config below:
```
{
  "wifi": {
    "ssid": "ssid",
    "password": "password"
  },
  "mqtt": {
    "server": {
      "host": "host",
      "port": 1883,
      "username": "esp32-ble2mqtt",
      "password": "password"
    },
    "publish": {
      "retain": true
    },
    "topics" :{
      "prefix": "ble2mqtt/",
      "get_suffix": "/get",
      "set_suffix": "/set"
    }
  },
  "log": {
    "host": "224.0.0.200",
    "port": 5000
  }
}
```

shmuelzon, 2019-09-13:
Hey Matthew,

A few things may be at play here.
First off, I’m assuming that the ESP32 is connected to your network and your PC is on the same network as it.
Next, are you using `make remote-monitor` to receive the logs?
Please make sure the is no firewall blocking the traffic.
Since this is a multicast address you might need IGMP snooping enabled on your router.

I think it would be simplest to set the IP address of your LAN PC (probably 192.168.x.x or 10.x.x.x.x) as the logs destination and start from there.

Let me know...

marrold, 2019-09-13:
Hi Assaf, thanks for the quick response.

> First off, I’m assuming that the ESP32 is connected to your network and your PC is on the same network as it.  
Yes that's right

> Next, are you using make remote-monitor to receive the logs?  
I've tried that as well as a packet capture, nothing was recieved

> Please make sure the is no firewall blocking the traffic.  
There's no firewall on this LAN

> Since this is a multicast address you might need IGMP snooping enabled on your router.  
Ah yes, I hadn't thought of that, but I did try the IP address of the machine and did not receive any logs. 

Thanks

marrold, 2019-09-13:
I should have thought of that!

This line looks related:`Log: Failed setting socket as non-blocking`

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6464
ho 0 tail 12 room 4
load:0x40078000,len:14616
ho 0 tail 12 room 4
load:0x40080400,len:4240
entry 0x400806dc
I (32) boot: ESP-IDF v4.1-dev-256-g9f145ff16 2nd stage bootloader
I (33) boot: compile time 16:47:17
I (33) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (91) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (99) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (106) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x33580 (210304) map
I (203) esp_image: segment 1: paddr=0x000435a8 vaddr=0x3ffbdb60 size=0x05c7c ( 23676) load
I (212) esp_image: segment 2: paddr=0x0004922c vaddr=0x40080000 size=0x00400 (  1024) load
I (213) esp_image: segment 3: paddr=0x00049634 vaddr=0x40080400 size=0x069dc ( 27100) load
I (232) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xefe6c (982636) map
I (583) esp_image: segment 5: paddr=0x0013fe8c vaddr=0x40086ddc size=0x103d4 ( 66516) load
I (627) boot: Loaded app from partition at offset 0x10000
I (627) boot: Disabling RNG early entropy source...
I (627) cpu_start: Pro cpu up.
I (631) cpu_start: Application information:
I (636) cpu_start: Project name:     ble2mqtt
I (641) cpu_start: App version:      v0.11.0
I (645) cpu_start: Compile time:     Sep 13 2019 16:47:23
I (652) cpu_start: ELF file SHA256:  ab098a4c9f87be64...
I (658) cpu_start: ESP-IDF:          v4.1-dev-256-g9f145ff16
I (664) cpu_start: Starting app cpu, entry point is 0x400811f4
I (0) cpu_start: App cpu up.
I (674) heap_init: Initializing. RAM available for dynamic allocation:
I (681) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (687) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (693) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (699) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (706) heap_init: At 3FFD3490 len 0000CB70 (50 KiB): DRAM
I (712) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (718) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (724) heap_init: At 400971B0 len 00008E50 (35 KiB): IRAM
I (731) cpu_start: Pro cpu start user code
I (749) spi_flash: detected chip: generic
I (750) spi_flash: flash io: dio
I (750) cpu_start: Chip Revision: 1
I (751) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (825) BLE2MQTT: Version: v0.11.0
I (825) Config: Initializing configuration
I (1035) Config: version: 60f92dd1bda7971027e8798b1f196318ad895a58f84406a9eb66f497d2349e00
I (1035) Log: Initializing remote logging
I (1045) OTA: Initializing OTA
I (1055) wifi: wifi driver task: 3ffd7920, prio:23, stack:3584, core=0
I (1055) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1065) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1085) wifi: wifi firmware version: df55612
I (1085) wifi: config NVS flash: enabled
I (1085) wifi: config nano formating: disabled
I (1095) wifi: Init dynamic tx buffer num: 32
I (1095) wifi: Init data frame dynamic rx buffer num: 32
I (1095) wifi: Init management frame dynamic rx buffer num: 32
I (1105) wifi: Init management short buffer num: 32
I (1105) wifi: Init static rx buffer size: 1600
I (1115) wifi: Init static rx buffer num: 10
I (1115) wifi: Init dynamic rx buffer num: 32
I (1125) wifi: Set ps type: 0

I (1125) Resolve: Initializing resolver
I (1125) BTDM_INIT: BT controller compile version [eec6ee1]
I (1135) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1235) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (1575) WiFi: Connecting to SSID BETA-2.4Ghz
I (1585) wifi: mode : sta (30:ae:a4:22:bd:04)
I (2315) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2955) wifi: state: init -> auth (b0)
I (2965) wifi: state: auth -> assoc (0)
I (2975) wifi: state: assoc -> run (10)
I (2985) wifi: connected with BETA-2.4Ghz, channel 6, BW20, bssid = 6e:3b:6b:27:b2:f1
I (2995) wifi: pm start, type: 0

I (2995) WiFi: Connected
I (3545) tcpip_adapter: sta ip: 10.0.130.130, mask: 255.255.255.0, gw: 10.0.130.1
E (3545) Log: Failed setting socket as non-blocking
I (3545) BLE2MQTT: Connected to WiFi, connecting to MQTT
I (3565) MQTT: Connecting MQTT client
I (3565) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (3575) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (3595) MQTT: MQTT client connected
I (3595) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (3905) BLE2MQTT: Discovered BLE device: 53:3f:64:8d:9a:4a, connecting
```

shmuelzon, 2019-09-13:
According to the log, you’re using ESP-IDF v4.1.
Would you mind testing with the “official” supported version in the README file (v3.2.2).

closed 2019-09-13 by marrold

shmuelzon, 2019-09-13:
No problem, glad to hear it worked out for you!