shmuelzon/esp32-ble2mqtt #44: Feature Request: Passive Scanning

marrold, 2019-09-14:
I have a limited understanding of BLE so apologies in advance, but as I understand it esp32-ble2mqtt is actively connecting to BLE devices.

Would it be possible to make it so it passively "sniffs" beacons and reports them to MQTT, without ever connecting to the device itself?

Thanks

DurandA, 2019-09-15:
This feature is already available. Have a look at [Broadcasters from README](https://github.com/shmuelzon/esp32-ble2mqtt#broadcasters).

Do you need any additional metadata for a particular device (e.g. sensor data)?

shmuelzon, 2019-09-15:
Hey Matthew,

In BLE, when referring to "Passive Scanning" it means the device (ESP32 in our case) will listen for other devices' advertisements without actively sending a "Scan Request".

If you don't want to connect to any detected device, you can simply add an empty whitelist to the configuration file.
```json
{
  "ble": {
    "whitelist": []
}
```

As @DurandA mentioned, there's support for Broadcasters (non-connectable devices that only publish information). We currently support iBeacon, Eddystone, MiJia sensor and the BeeWi.
If you wish to add support for a different type of device, take a look at [broadcasters.c](https://github.com/shmuelzon/esp32-ble2mqtt/blob/master/main/broadcasters.c) for examples on what you need to do.

Let me know if you need anything else.

marrold, 2019-09-15:
For some context, I want to be able to detect the presence of a connectable BLE_ device from its "broadcasts" alone, without connecting to the device. 

This is partially motivated by one of the BLE devices running out of battery after just a couple of days. I suspect these connections may be preventing it from sleeping, or its transmitting excessively and draining the battery  - is this likely? 

I've tried the empty whitelist, but as far as I can tell detected devices aren't then reported via mqtt.

Thanks again

shmuelzon, 2019-09-16:
> This is partially motivated by one of the BLE devices running out of battery after just a couple of days. I suspect these connections may be preventing it from sleeping, or its transmitting excessively and draining the battery - is this likely?

Well, it depends on the how the device is configured and implemented. Generally speaking, BLE peripherals sleep most of the time, even when connected. How often they wake up is defined bbby a few parameters, e.g. connection interval and slave latency.
Is this your own device or a store-bought one?

Anyways, for your use-case, you're going to need to do a bit of work.
The following (not tested in any way) will simply treat all devices as broadcasters:
```patch
diff --git a/main/broadcasters.c b/main/broadcasters.c
index 068d6f5..32b2bb9 100644
--- a/main/broadcasters.c
+++ b/main/broadcasters.c
@@ -462,12 +462,30 @@ static broadcaster_ops_t beewi_smart_door_ops = {
     .metadata_get = beewi_smart_door_metadata_get,
 };
 
+/* All devices */
+static int all_devices_is_broadcaster(uint8_t *adv_data, size_t adv_data_len)
+{
+    return 1;
+}
+
+static void all_devices_metadata_get(uint8_t *adv_data,
+    size_t adv_data_len, int rssi, broadcaster_meta_data_cb_t cb, void *ctx)
+{
+}
+
+static broadcaster_ops_t all_devices_ops = {
+    .name = "All Devices",
+    .is_broadcaster = all_devices_is_broadcaster,
+    .metadata_get = all_devices_metadata_get,
+};
+
 /* Common */
 static broadcaster_ops_t *broadcaster_ops[] = {
     &ibeacon_ops,
     &eddystone_ops,
     &mijia_temp_hum_ops,
     &beewi_smart_door_ops,
+    &all_devices_ops,
     NULL
 };
 
```
This means that on every advertisement or scan response message, the above code will treat it as a broadcaster and publish it via MQTT. It doesn't extract any information from the BLE packet so it will only publish the RSSI by default. If there's something in the advertisement packet you're interested in, you can add that in `all_devices_metadata_get()`. See the other implementations for an example.

marrold, 2019-09-16:
It's a store bought "iTAG" - a cheap and nasty location tracker.

Thanks for the code snippet, that's really helpful. I will try to implement your suggested changes and see where I get to.

shmuelzon, 2019-10-06:
How did it work out for you? Can I close this issue for now?

shmuelzon, 2019-10-15:
I'm closing this issue for now. If there's anything else, you can reopen this issue.

closed 2019-10-15 by shmuelzon