shmuelzon/esp32-ble2mqtt #45: mqtt, can esp32 host its own mqtt?

advancedcurrents, 2019-09-23:
Such that I do not need a mqtt server...

shmuelzon, 2019-09-23:
Hey,

While I’m sure it’s technically possible and was probably already done, it’s not a part of this project’s scope. The idea was to have multiple nodes on premise for redundancy and range while a single (always-on) broker maintains the last know values in case of disconnection.
If you don’t have an always-on computer, you can use an online service or run an MQTT broker on an ESP32 but I wouldn’t suggest doing so on the same device running this project.

Good day

shmuelzon, 2019-10-06:
Hey,

As you haven't responded in two weeks, I'm closing this issue.
Let me know if you need anything else.

Good day

closed 2019-10-06 by shmuelzon