shmuelzon/esp32-ble2mqtt #46: Feature request: Web interface

r00li, 2019-10-05:
Hi!
I've just played around with this project and I think it's awesome. Currently in the middle of setting it up in my home automation system (replacing the need for BLE support in my main HA controller).

During this process I discovered one thing - it's a bit annoying editing the config on the computer, uploading it to the device and then hoping that everything is set up correctly. It would be awesome if this would include a configuration web interface (like a very popular sonoff-tasmota project for instance).

It would be nice if we could see all of the discovered BLE devices in the web interface, then add them to a whitelist/blacklist and configure which of devices services/characteristics we would like published via MQTT, ... It doesn't have to be anything fancy or anything. The first version could just dump the current config and enable us to edit it on the fly and press save. 

I know it's a lot to ask, but that would be one feature that would make this a lot easier to use.

shmuelzon, 2019-10-06:
Hey,

I agree that would be a nice feature. The backend could be a rather simple REST API to just GET and PUT the configuration file. Next step would be exposing the BLE database with a GET and have the UI do the heavy lifting.

A few points worth noting though:
* It would still require (for now) a reboot after the configuration changes
* There are also certificates included as a part of the configuration settings that aren't in the JSON file
* In my mind, you'll have several of these devices to overcome the short range of BLE and I think the configuration should be synced with all of them
  * Editing one device manually might cause some inconsistencies
  * After changing a setting, you would need to download the new configuration file and still upload it to all the other devices

Let me know what you think

r00li, 2019-10-06:
I think that's a good way to start yes. 

Well rebooting after configuration change shouldn't be an issue at all. 

For certificates... I am not sure (haven't used SSL yet), but one idea is to just add a text area where you can paste the content of a certificate... That's something that is often used on some routers for example.

An interface offered by the above mentioned sonoff tasmota firmware should be a good example (since it's running on an ESP8266). There you can download the whole config as a file and upload it to the new module, and then just change the stuff that you need). It doesn't have to be any more complicated than that. 

One other thing that tasmota offers is that they offer pre-built binaries that you can just upload to the esp module. The module then sets up a wifi access point where you can connect and set up your wifi details, hit save and then it reboots and connects to your wifi. That's a more advanced feature, but one that would ultimately be really friendly for someone who just wants an easy to use solution. A more easy to implement solution would be to enter those details via serial for example. But none of that is probably something for the first version.

cooldesigns, 2019-12-24:
Why ssl certs is a problem? Is file uploading difficult? Look here: https://github.com/G6EJD/ESP32-8266-FIle-Upload

For web interface using old way static pages and static forms or rest api is not good at all. Using graphql and any client side javascript framework is more more convenient, but there is no cpp graphql server.

Also I think having autoconnect is comfortable. https://hieromon.github.io/AutoConnect/

shmuelzon, 2019-12-24:
Hey @cooldesigns,

Certificates aren’t a problem, I just mentioned that they are an additional file an not a part of the configuration file (config.json).

I think GraphQL is quite an overkill for this use case. I believe that simple REST APIs and static pages (that include a client side, JS, app using Angular or whatever) would be the simplest to implement and would not use up a lot of CPU time.

Autoconnect is an Arduino library so it won’t help much here. The are alternatives in ESP-IDF but the main problem is that I don’t have time to implement it.

cooldesigns, 2019-12-31:
I can give time for this if you orchestra the tasks. Note: I have web application development skill, I haven't writing c++ project skill.

closed 2020-12-15 by shmuelzon

shmuelzon, 2020-12-15:
Well, it's been a year but I finally got around to adding a web server.
Basically, anything under the `www` folder will be served as is and the backend should be implemented in `main/httpd.c`. Currently only routes for uploading firmware images (`build/ble2mqtt.bin`) and file-system images (`build/fs_0.bin`) are supported.
More can be added as needed.
I would be very beneficial to support uploading only the configuration file and/or manage certificates so one can make modifications without needing to install/compile this project at all.

commit 448496f3 (2020-12-15): Add HTTP server with a (very) basic interface

Closes #46
commit 4c2f3fb6 (2020-12-15): Add HTTP server with a (very) basic interface

Closes #46
commit e009e109 (2020-12-16): Add HTTP server with a (very) basic interface

Closes #46