shmuelzon/esp32-ble2mqtt #47: Force passkey auth?

lukeIam, 2019-10-25:
Is there a way to force the use of the passkey?
Looks like my device no not requests passkey-auth but shows only basic services/characteristics then.
(at least I think so - I'm not a ble expert)

I've added service and characteristics definition to config and gatt.c but they are not discovered.

My config:
```json
{
  "wifi": {
    "ssid": "xxx",
    "password": "xxx"
  },
  "mqtt": {
    "server": {
      "host": "192.168.x.x",
      "port": 1883,
      "username": "user",
      "password": "xxx",
      "client_id": "xxx"
    },
    "publish": {
      "retain": true
    },
    "topics": {
      "prefix": "xxx/",
      "get_suffix": "/get",
      "set_suffix": "/set"
    },
    "ble": {
      "passkeys": {
        "16:0e:72:xx:xx:xx": xxxxxx,
        "0D:7e:32:xx:xx:xx": xxxxxx
      },
      "whitelist": [
        "16:0e:72:xx:xx:xx",
        "0D:7e:32:xx:xx:xx"
      ],
      "services": {
        "definitions": {
          "47e9ee00-47e9-11e4-8939-164230d1df67": {
            "name": "ThermostatService"
          }
        },        
        "whitelist": [
          "47e9ee00-47e9-11e4-8939-164230d1df67"
        ]
      },
      "characteristics": {
        "definitions": {
          "47e9ee2b-47e9-11e4-8939-164230d1df67": {
            "name": "Temperature",
            "types": [
              "CHAR_TYPE_UINT8","CHAR_TYPE_UINT8","CHAR_TYPE_UINT8","CHAR_TYPE_UINT8","CHAR_TYPE_UINT8","CHAR_TYPE_UINT8","CHAR_TYPE_UINT8"
            ]
          }
        },
        "whitelist": [
          "47e9ee2b-47e9-11e4-8939-164230d1df67"
        ]
      }
    }
  }  
}
```

shmuelzon, 2019-10-27:
Hey,

There is currently no way to force pairing. That process is started when a read/write fails due to insufficient authentication or encryption.
From my (limited) experience, devices always expose all of the services/characteristics and only require pairing when reading/writing the actual values.
I would suggest trying to connect to the device from a mobile phone or something using a BLE app, e.g. LightBlue, and see how it behaves.

You can try to force pairing when the device is connected with the following (untested) patch:
```patch
diff --git a/main/ble.c b/main/ble.c
index dca8143..86cfcef 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -673,7 +673,11 @@ static void esp_gattc_cb(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if,
                 param->cfg_mtu.conn_id);
 
             if (dev)
+            {
+                dev->is_authenticating = 1;
+                esp_ble_set_encryption(dev->mac, ESP_BLE_SEC_ENCRYPT_MITM);
                 on_device_connected_cb(dev->mac);
+            }
         }
 
         break;
```

BTW - your characteristic types are wrong and should be `uint8` instead of `CHAR_TYPE_UINT8 `. See [this](https://github.com/shmuelzon/esp32-ble2mqtt/blob/ea67a2338fc6712e380c56625dfa237099f5a8a0/main/ble_utils.c#L165) for the full list.

lukeIam, 2019-10-27:
Hi @shmuelzon,

thanks for your answer!

> BTW - your characteristic types are wrong and should be `uint8` instead of `CHAR_TYPE_UINT8 `. See [this](https://github.com/shmuelzon/esp32-ble2mqtt/blob/ea67a2338fc6712e380c56625dfa237099f5a8a0/main/ble_utils.c#L165) for the full list.

You are right - I changed that.

> From my (limited) experience, devices always expose all of the services/characteristics and only require pairing when reading/writing the actual values.
> I would suggest trying to connect to the device from a mobile phone or something using a BLE app, e.g. LightBlue, and see how it behaves.

And you are right again - the device exposes all services/characteristics without a passkey.
I checked with gatttool:
```
[17:20:19]:~$ gatttool -I
[                 ][LE]> connect 16:0e:72:xx:xx:xx
Attempting to connect to 16:0e:72:xx:xx:xx
Connection successful
[16:0e:72:xx:xx:xx][LE]> characteristics
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x0a, char value handle: 0x0007, uuid: 00002a02-0000-1000-8000-00805f9b34fb
handle: 0x0008, char properties: 0x02, char value handle: 0x0009, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x22, char value handle: 0x000e, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x0011, char properties: 0x02, char value handle: 0x0012, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0013, char properties: 0x02, char value handle: 0x0014, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0015, char properties: 0x02, char value handle: 0x0016, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x0017, char properties: 0x02, char value handle: 0x0018, uuid: 00002a28-0000-1000-8000-00805f9b34fb
handle: 0x001a, char properties: 0x0a, char value handle: 0x001b, uuid: 47e9ee01-47e9-11e4-8939-164230d1df67
handle: 0x001c, char properties: 0x0a, char value handle: 0x001d, uuid: 47e9ee10-47e9-11e4-8939-164230d1df67
handle: 0x001e, char properties: 0x0a, char value handle: 0x001f, uuid: 47e9ee11-47e9-11e4-8939-164230d1df67
handle: 0x0020, char properties: 0x0a, char value handle: 0x0021, uuid: 47e9ee12-47e9-11e4-8939-164230d1df67
handle: 0x0022, char properties: 0x0a, char value handle: 0x0023, uuid: 47e9ee13-47e9-11e4-8939-164230d1df67
handle: 0x0024, char properties: 0x0a, char value handle: 0x0025, uuid: 47e9ee14-47e9-11e4-8939-164230d1df67
handle: 0x0026, char properties: 0x0a, char value handle: 0x0027, uuid: 47e9ee15-47e9-11e4-8939-164230d1df67
handle: 0x0028, char properties: 0x0a, char value handle: 0x0029, uuid: 47e9ee16-47e9-11e4-8939-164230d1df67
handle: 0x002a, char properties: 0x0a, char value handle: 0x002b, uuid: 47e9ee20-47e9-11e4-8939-164230d1df67
handle: 0x002c, char properties: 0x0a, char value handle: 0x002d, uuid: 47e9ee21-47e9-11e4-8939-164230d1df67
handle: 0x002e, char properties: 0x0a, char value handle: 0x002f, uuid: 47e9ee22-47e9-11e4-8939-164230d1df67
handle: 0x0030, char properties: 0x0a, char value handle: 0x0031, uuid: 47e9ee23-47e9-11e4-8939-164230d1df67
handle: 0x0032, char properties: 0x0a, char value handle: 0x0033, uuid: 47e9ee24-47e9-11e4-8939-164230d1df67
handle: 0x0034, char properties: 0x0a, char value handle: 0x0035, uuid: 47e9ee25-47e9-11e4-8939-164230d1df67
handle: 0x0036, char properties: 0x0a, char value handle: 0x0037, uuid: 47e9ee26-47e9-11e4-8939-164230d1df67
handle: 0x0038, char properties: 0x0a, char value handle: 0x0039, uuid: 47e9ee27-47e9-11e4-8939-164230d1df67
handle: 0x003a, char properties: 0x0a, char value handle: 0x003b, uuid: 47e9ee2a-47e9-11e4-8939-164230d1df67
handle: 0x003c, char properties: 0x0a, char value handle: 0x003d, uuid: 47e9ee2b-47e9-11e4-8939-164230d1df67 #<---
handle: 0x003e, char properties: 0x0a, char value handle: 0x003f, uuid: 47e9ee2c-47e9-11e4-8939-164230d1df67
handle: 0x0040, char properties: 0x0a, char value handle: 0x0041, uuid: 47e9ee2d-47e9-11e4-8939-164230d1df67
handle: 0x0042, char properties: 0x0a, char value handle: 0x0043, uuid: 47e9ee2e-47e9-11e4-8939-164230d1df67
handle: 0x0044, char properties: 0x1a, char value handle: 0x0045, uuid: 47e9ee40-47e9-11e4-8939-164230d1df67
handle: 0x0047, char properties: 0x08, char value handle: 0x0048, uuid: 47e9ee30-47e9-11e4-8939-164230d1df67
[16:0e:72:xx:xx:xx][LE]> primary
attr handle: 0x0001, end grp handle: 0x0009 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x000f uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0010, end grp handle: 0x0018 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x0019, end grp handle: 0x0048 uuid: 47e9ee00-47e9-11e4-8939-164230d1df67 #<---
```

So there are definitely there and published - Any idea what I could be missing that I can't see them with esp32-ble2mqtt?

```
I (6404) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (6534) BLE2MQTT: Discovered BLE device: 16:0e:72:xx:xx:xx, connecting
I (10424) BLE2MQTT: Connected to device: 16:0e:72:xx:xx:xx, scanning
I (11064) BLE2MQTT: Publishing: b2m1/16:0e:72:xx:xx:xx/GenericAccess/DeviceName= Comet Blue
I (11154) BLE2MQTT: Publishing: b2m1/16:0e:72:xx:xx:xx/GenericAccess/Appearance= 0
I (11244) BLE2MQTT: Publishing: b2m1/16:0e:72:xx:xx:xx/GenericAccess/PeripheralPrivacyFlag = false
I (11274) BLE2MQTT: Publishing: b2m1/16:0e:72:xx:xx:xx/GenericAccess/PeripheralPreferredConnectionParameters = 80,160,0,300
I (11354) BLE2MQTT: Publishing: b2m1/16:0e:72:xx:xx:xx/GenericAttribute/ServiceChanged = 1,65535
```

> You can try to force pairing when the device is connected with the following (untested) patch:

Anyway I also tested your patch:
`E (8065) BLE: Open failed, status = 0x85``
But as I figured out that the problem is not depending to paring that's not a problem.

shmuelzon, 2019-10-27:
Unfortunately, this looks like an issue I thought I already fixed but can't find anywhere now...
Until I have time to investigate, please add `00002a05-0000-1000-8000-00805f9b34fb` to the characteristics blacklist. It should allow processing all of the discovered characteristics.

lukeIam, 2019-11-16:
sorry was a bit busy the last time.

> `00002a05-0000-1000-8000-00805f9b34fb` to the characteristics blacklist. It should allow processing all of the discovered characteristics.

in combination with fixing my config (problems with the position of some brackets) I can now see the characteristics. But still can't connect: error read 0x02 - but that seems to be a problem with the device and not you lib. (seems that I somehow needs to send the pin to 47e9ee30-47e9-11e4-8939-164230d1df67 bu that one was not discovered)

closed 2019-11-16 by lukeIam