shmuelzon/esp32-ble2mqtt #48: mkspiffs: Permission denied

fabiosoft, 2019-10-29:
I'am able to build after installing the toolchain, but doing `make flash` at the end i get a permission issue.

```
fabio@imacfabio esp32-ble2mqtt % make flash
Toolchain path: /Users/fabio/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/fabio/esp/esp-idf/requirements.txt are satisfied.
make: /Users/fabio/Downloads/esp32-ble2mqtt/mkspiffs: Permission denied
make: *** [/Users/fabio/Downloads/esp32-ble2mqtt/build/spiffs.bin] Error 1
```

Of course mkspiffs directory exists...as i cloned recursively this project
I think somehow it's a problem with Mac os catalina... but how can i solve?

shmuelzon, 2019-10-30:
I've just updated to Catalina to test this, pulled a clean version and built it. I got build error for mkspiffs, specifically:
```
ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
```
I was able to "fix" this by applying the following patch in the mkspiffs directory:
```patch
diff --git a/Makefile b/Makefile
index 4a23f01..87196eb 100644
--- a/Makefile
+++ b/Makefile
@@ -27,9 +27,9 @@ else
 		DIST_SUFFIX := osx
 		CC=clang
 		CXX=clang++
-		TARGET_CFLAGS   = -mmacosx-version-min=10.7 -arch i386 -arch x86_64
-		TARGET_CXXFLAGS = -mmacosx-version-min=10.7 -arch i386 -arch x86_64 -stdlib=libc++
-		TARGET_LDFLAGS  = -arch i386 -arch x86_64 -stdlib=libc++
+		TARGET_CFLAGS   = -mmacosx-version-min=10.7
+		TARGET_CXXFLAGS = -mmacosx-version-min=10.7 -stdlib=libc++
+		TARGET_LDFLAGS  = -stdlib=libc++
 	endif
 	ARCHIVE_CMD := tar czf
 	ARCHIVE_EXTENSION := tar.gz
```

Could you please clean your build, apply the above patch and try again?

Let me know...

fabiosoft, 2019-10-31:
I tried removing only `-arch i386` but leaving x86 and works too.

Thanks

closed 2019-10-31 by fabiosoft