shmuelzon/esp32-ble2mqtt #49: BLE characteristic can only be set once

DurandA, 2019-10-31:
Hi @shmuelzon!

I noticed that setting a characteristic more than once triggers the following error (first time always work):
```Failed writing characteristic, status = 0x3```

https://github.com/shmuelzon/esp32-ble2mqtt/blob/ea67a2338fc6712e380c56625dfa237099f5a8a0/main/ble.c#L744-L750

It seems that 0x03 is [`ESP_GATT_WRITE_NOT_PERMIT`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gatt_defs.html#_CPPv425ESP_GATT_WRITE_NOT_PERMIT).

This error is not a false positive since my device doesn't update its characteristic value.

Do you have any idea about what can cause this issue?

shmuelzon, 2019-10-31:
Do you mean two times in succession/quickly or in general while connected to the device?

DurandA, 2019-10-31:
> Does this happen to you for all writable characteristics?

Yes, I tried with a few ones on my Thingy:52. _WRITE NO RESPONSE_ characteristics silently fail.

> What about different peripherals?

I don't have other BLE peripherals in my office. I will try later with an emulated peripheral (I guess a smartphone can work as a BLE peripheral with the right app).

> Can you dry doing the same with a BLE app on you mobile phone? Does it succeed writing multiple times?

It works fine writing multiple time to the same characteristic using _nRF Connect_.

DurandA, 2019-10-31:
I tried with another ESP32 behaving as BLE peripheral (using the [BLE peripheral example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph)): `BLE: Failed writing characteristic, status = 0x5` which is `ESP_GATT_INSUF_AUTHENTICATION`.

The write operation works just fine with _nRF Connect_. I am not sure what is happening there.

DurandA, 2019-11-01:
> I will try to remove `BLE_GATT_CHR_F_WRITE_ENC` from https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/examples/bluetooth/nimble/bleprph/main/gatt_svr.c#L76

It seems to work when I remove the `BLE_GATT_CHR_F_WRITE_ENC` from the BLE peripheral example. That means that the error is somehow specific to the Thingy:52. Very strange, I will inspect the Thingy:52 firmware and look for clues. I don't expect the Thingy:52 to violate the BLE protocol.

DurandA, 2019-11-03:
The problem was with 8-bit values. The written value was accumulated, causing invalid payloads on my device. See #51.

shmuelzon, 2019-11-03:
Thanks for taking the time to debug and even fix this! Much appreciated!

PR #51 (2019-11-03): Fix writing 8-bit values
Fix #49
commit 83b868c5 (2019-11-03): Fix writing 8-bit values (#51)

Fix #49
closed 2019-11-03 by shmuelzon