shmuelzon/esp32-ble2mqtt #50: Reconnect upon disconnection

DurandA, 2019-10-31:
Hi again @shmuelzon and sorry for spamming issues!

Is there a simple way to reconnect BLE devices if they get disconnected? It typically happens after a few hours with my devices and I have to reset the ESP32 in order for it to reconnect.

Note that I did not investigate on this at all yet. Maybe you already give some thoughts about it?

shmuelzon, 2019-10-31:
This should already happen automatically, as long as the peripheral is broadcasting after the disconnection.
When the device is disconnected, it's removed from the device cache and when it's advertisement packet is received is treated as if it's the first time the device is seen and we connect to it (if allowed by black/white lists).

DurandA, 2019-11-03:
Reconnection works fine with my Thingy:52 if the signal is lost and recovered. However, after a few hours the gateway and my Thingy:52 disconnect and never reconnect. This is most likely due to the Thingy:52 going to sleep and stopping advertising.

I will close this issue for now and investigate.

closed 2019-11-03 by DurandA