shmuelzon/esp32-ble2mqtt #54: Unable to communicate with Comet Blue device

GrumpyMeow, 2019-11-17:
Hi,

I've put many hours of effort into creating a bridge between my Bluetooth radiator thermostat (Comet Blue) and Home Assistant. I think your project is quite mature and expected that i would get it to work.

I do see some attributes being published (for instance DeviceName), but only when i'd whitelisted the device. If i remove the "whitelist" attribute, i don't any attributes being published anymore. This is not what i'd expected. I do see other devices like IMac's being recognized.

Another issue is that i'm unable to retrieve any other attributes by publishing any value to the topic. 

Any hint for me?

This is my current configuration which i build using ESP IDF v3.3:
```
{
  "wifi": {
    "ssid": "mywifi",
    "password": "mypassword"
  },
  "mqtt": {
    "server": {
      "host": "192.168.11.90",
      "port": 1883,
      "username": "homeassistant",
      "password": "anotherpassword",
      "client_id": "ble2mqtt"
    },
    "publish": {
      "retain": true
    },
    "topics" :{
      "device_name": "Address",
      "prefix": "ble/",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "whitelist": ["28:6c:17:96:5e:0a"],
    "services": { 
        "definitions": {
            "47e9ee00-47e9-11e4-8939-164230d1df67": {
                "name": "cb"
            }
        }   
    },
    "characteristics" : {
        "47e9ee2c-47e9-11e4-8939-164230d1df67": {
           "name": "bat",
           "poll": 60,
           "types": [
               "uint8"
           ]
        }
    }
  },
  "passkeys": {
     "28:6c:17:96:5e:0a": 0
  },
  "log": {
    "host": "192.168.11.3",
    "port": 5000
  }
}
```

If i subscribe to MQTT, i see the following data:
```
ble/28:6c:17:96:5e:0a/Connected true
ble/28:6c:17:96:5e:0a/Owner BLE2MQTT-A678
ble/28:6c:17:96:5e:0a/GenericAccess/DeviceName Comet Blue
ble/28:6c:17:96:5e:0a/GenericAccess/Appearance 0
ble/28:6c:17:96:5e:0a/GenericAccess/PeripheralPrivacyFlag false
ble/28:6c:17:96:5e:0a/GenericAccess/PeripheralPreferredConnectionParameters 80,160,0,300
ble/28:6c:17:96:5e:0a/GenericAttribute/ServiceChanged 1,65535
```

But i don't see the ManfacturerNameString attribute for instance.

shmuelzon, 2019-11-17:
Hey,

Unfortunately, this is an issue that has been plaguing me for a while now that I've still haven't been able to get to the root cause of it. It all boils down to the fact the the BLE library doesn't acknowledge that it successfully wrote to the ServiceChanged characteristic and so the app waits forever. I have a ~~fix~~ hack ready to avoid this but I'm not very keen on this solution. In the mean time, you can add `00002a05-0000-1000-8000-00805f9b34fb` to the characteristic blacklist to avoid it altogether.

BTW - Regarding your configuration file, the characteristic definitions also need to be under a `definitions` object. Also, the `poll` field doesn't exist here and was only implemented in the [Node version](https://github.com/shmuelzon/ble2mqtt).

Let me know how it works out for you.

GrumpyMeow, 2019-11-18:
Hi, 
Blacklisting the characteristic indeed worked. 
I also noticed that my confiuration file was incorrect.
thanks!

closed 2019-11-18 by GrumpyMeow

lukeIam, 2019-11-22:
@GrumpyMeow Did you get your comet blues running?
Im failing with the pin auth and getting multiple `BLE: Failed reading characteristic, status = 0x2`

I know that I need to send the encoded pin to `47e9ee30-47e9-11e4-8939-164230d1df67` first, before reading the other characteristics. But I haven't found the correct point in the code for that (currently trying with to inject my code in the beginning of the `ble_foreach_characteristic` function - but that seems not to work)

shmuelzon, 2019-11-22:
@lukeIam, Have you tried going through the [Wiki](https://github.com/shmuelzon/esp32-ble2mqtt/wiki/Comet-Blue-(work-in-progress)) page @GrumpyMeow has written?

GrumpyMeow, 2019-11-23:
Hi, I did manage to get the Comet Blue partially up & running. I’m still doing some optimizations in Home Assistant. I’ll update the wiki-page this weekend.