shmuelzon/esp32-ble2mqtt #57: Wiki pages for devices?

GrumpyMeow, 2019-11-18:
Hi,

is it maybe possible to create a few wiki pages for specific devices. I notice that others are also trying to get the Eurotronics Comet Blue device up & running.

```
wifi": {
    "ssid": "xxxxxx",
    "password": "xxxxxx"
  },
  "mqtt": {
    "server": {
      "host": "192.168.11.90",
      "port": 1883,
      "username": "homeassistant",
      "password": "xxxxxx",
      "client_id": "ble2mqtt"
    },
    "publish": {
      "retain": true,
      "qos" : 0
    },
    "topics" :{
      "prefix": "ble/",
      "device_name": "Address",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "whitelist": ["28:6c:16:96:xx:xx"],
    "services": {
        "definitions": {
            "47e9ee00-47e9-11e4-8939-164230d1df67": {
                "name": "ThermostatService"
            }
        }
    },
    "characteristics" : {
        "blacklist": ["00002a05-0000-1000-8000-00805f9b34fb"],
        "definitions": {
           "47e9ee01-47e9-11e4-8939-164230d1df67": {
                "name": "Time",
                "types": [ "uint8","uint8","uint8","uint8","uint8"]
           },
           "47e9ee10-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointMonday"
           },
           "47e9ee11-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointTuesday"
           },
           "47e9ee12-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointWednesday"
           },
           "47e9ee13-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointThursday"
           },
          },
           "47e9ee13-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointThursday"
           },
           "47e9ee14-47e9-11e4-8939-164230d1df67": {
                "name": "SwitchpointFriday"
           },
           "47e9ee15-47e9-11e4-8939-164230d1df67": {
                "name": "15"
           },
           "47e9ee16-47e9-11e4-8939-164230d1df67": {
                "name": "16"
           },
           "47e9ee17-47e9-11e4-8939-164230d1df67": {
                "name": "17"
           },
           "47e9ee18-47e9-11e4-8939-164230d1df67": {
                "name": "18"
           },
           "47e9ee19-47e9-11e4-8939-164230d1df67": {
                "name": "19"
           },
           "47e9ee1a-47e9-11e4-8939-164230d1df67": {
                "name": "1a"
           },
           "47e9ee1b-47e9-11e4-8939-164230d1df67": {
                "name": "1b"
           },
           "47e9ee1c-47e9-11e4-8939-164230d1df67": {
                "name": "1c"
           },
           "47e9ee1da-47e9-11e4-8939-164230d1df67": {
                "name": "1d"
           },
           "47e9ee1e-47e9-11e4-8939-164230d1df67": {
                "name": "1e"
           },
           "47e9ee1f-47e9-11e4-8939-164230d1df67": {
                "name": "1f"
           },
           "47e9ee20-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday1"
           },
           "47e9ee21-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday2"
           },
           "47e9ee22-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday3"
           },
           "47e9ee23-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday4"
           },
           "47e9ee24-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday5"
           },
           "47e9ee25-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday6"
           },
           "47e9ee26-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday7"
           },
           "47e9ee27-47e9-11e4-8939-164230d1df67": {
                "name": "Holiday8"
           },
           "47e9ee2a-47e9-11e4-8939-164230d1df67": {
                "name": "Flags",
                "types": ["uint24"]
           },
           "47e9ee2b-47e9-11e4-8939-164230d1df67": {
                "name": "Temperature",
                "types": [ "uint8" ,"uint8","uint8","uint8","uint8","uint8","uint8"]
           },
           "47e9ee2c-47e9-11e4-8939-164230d1df67": {
                "name": "Battery",
                "types": [ "uint8"]
           },
           "47e9ee2d-47e9-11e4-8939-164230d1df67": {
                "name": "2d"
           },
           "47e9ee2e-47e9-11e4-8939-164230d1df67": {
                "name": "LcdTimer"
           },
           "47e9ee30-47e9-11e4-8939-164230d1df67": {
                "name": "Pin",
                "types": [ "uint8","uint8","uint8","uint8" ]
           },
           "47e9ee40-47e9-11e4-8939-164230d1df67": {
                "name": "FirmwareVersion"
           }
        }
    }
  },
  "log": {
    "host": "192.168.11.3",
    "port": 5000
  }
}
```
In Home Assistant i can transmit the Pin-code (1234) as:
```ble/28:6c:16:96:xx:xx/ThermostatService/Pin/Set``` with the value: ```1,2,3,4```

shmuelzon, 2019-11-19:
Hey,

I've enabled Wiki for this repository. Feel free to add some notes and a list/menu for devices.
I think each device should include the services characteristics definitions. Home assistant integration is also very welcome, but I'd add the entire device config, not just topic name, e.g.:
```yaml
climate:
  - platform: mqtt
    name: Study
    temperature_command_topic: "xx:xx:xx:xx:xx:xx/ThermostatService/Temperature/Set"
```

@DurandA, Could you please add the definitions for your Thingy52?

Thanks!

closed 2019-11-19 by shmuelzon

GrumpyMeow, 2019-11-19:
Thanks!
I’ll post my configurations when home from work on the wiki.

DurandA, 2019-11-19:
Adding a wiki for devices is such a great idea!

> @DurandA, Could you please add the definitions for your Thingy52?

Sure! However, the wiki is restricted to collaborators only so we cannot edit it.

shmuelzon, 2019-11-19:
> However, the wiki is restricted to collaborators only so we cannot edit it.

Removed the restriction

DurandA, 2019-11-19:
[Thingy:52 wiki page](https://github.com/shmuelzon/esp32-ble2mqtt/wiki/Nordic-Thingy:52).