shmuelzon/esp32-ble2mqtt #59: Disconnect with device

GrumpyMeow, 2019-11-19:
Hi,

Is there an option to forcefully drop the connection with the BLE device ?
I notice that my Comet Blue thermostatic-radiator-valves become non-responsive after a while.
It's also that these device support only a single connection at a time.

Maybe the connection can be dropped when upon sending a specific MQTT-message.

shmuelzon, 2019-11-20:
Hey,

There's currently no way to do that and I think adding it would cause some problems.
Let's say we add a way to disconnect from the peripheral. After it's disconnected, it will be discovered again and then we'll automatically reconnect to it. That would be kind of weird.
So, we could save somewhere to not reconnect to the peripheral again, but then what? Do we limit that in time or indefinitely? And then add some way to force a connect? It becomes a bit of a mess...

My purpose for this project is to be zero-maintenance. Unlike other projects I've seen, I don't want another entity in the network to manage these devices.

As for only a single connection, that's the point of exposing the information over MQTT, to allow multiple entities access to the peripheral.

GrumpyMeow, 2019-11-20:
I understand what you are saying.

What about having a device specific topic with a Boolean flag to determine if the connection is kept-alive?

That makes the logic less messy. Something like “Connected” topic.

shmuelzon, 2019-11-21:
The "Connected" topic is already used to identify if the device is currently connected or not. In fact, if you, or another instance of this project, were to publish "false" on this topic, the instance that's currently connected to the peripheral will sent "true" again to make sure this topic is up-to-date.

This brings up another issue where you would need instance of BLE2MQTT that haven't seen yet the specified peripheral to know to listen on this new topic so when they do discover the peripheral, they'll know wether to connect to it or not.

Have you tried contacting the manufacturer to fix the issue of the device not responding?

GrumpyMeow, 2019-11-22:
I'll have to think about how i can workaround this.
Thanks for now.

closed 2019-11-22 by GrumpyMeow

GrumpyMeow, 2019-11-23:
Hi, I have some ideas how to resolve some issues. I think that creating a Home Assistant automation for publishing the code upon connecting will resolve some issues.