shmuelzon/esp32-ble2mqtt #60: Setup Help

rowdy15, 2019-11-23:
Firstly, thank you for the library, it's just what I need for use with home assistant.

Just wanting help with setup.

How do I manually disable or enable notifications on a service's characteristic.

e.g. the setting the Client Characteristic Configuration Descriptor (CCCD).

With my current setup the esp32 is enabling notifications on all (notification supported) characteristics automatically. I don't want this. I want to control the CCCD of the notification enabled characteristics.

Thanks.

shmuelzon, 2019-11-23:
Hey,

The purpose of this project is to be zero-maintenance, so there is no way to manually register/unregister for notifications.
If a characteristic supports notifications I would rather use that than poll it.

If you’re not interested in a specific service/characteristic at all, you can always make use of the white/black list feature.

rowdy15, 2019-11-23:
OK I see.

The reason for the query is because I have a low power project (using an nRF52832) where I am running a single RTC (using multiple RTCs uses up more power) timer to periodically take battery measurements and another custom service/characteristic for a countdown timer.  when both characteristics are notify enabled the RTC interrupt causes an app error as one of the characteristic's RTC timer is still running.

It would be great if there was a mqtt topic like 

<mac address>/<service>/<characteristic>+<" CCCD">

which you write a 1 (for notify) or 2 (for indicate)

But all good if not feasible.

I also have the below problem:

On the nRF52 debug terminal it is saying "GATT Client Timeout."

Then the esp32 debug terminal is saying the below (and disconnecting before reconnecting):

W (4967424) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
I (4967434) BLE: Connection closed, reason = 0x13

Your thoughts on this would be appreciated also.

rowdy15, 2019-11-24:
Hey shmuelzon,

So just an update on the previous post.

I was able to deal with enabling/disabling notifications on the server (nRF52832 BLE MCU) instead.

And I changed the following in the sdkconfig file which stopped the random disconnections.

From:
CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=y
CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG=y

To:
CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=
CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG=

Thanks again for the great library.

shmuelzon, 2019-11-24:
Glad to hear you got it sorted out.
I’m not sure what those configs do exactly, but as this project isn’t going to be battery operated and assuming it doesn’t impact Wi-Fi performance, it might be worth adding them to the default configuration.

closed 2019-11-24 by shmuelzon