shmuelzon/esp32-ble2mqtt #61: Do characteristics values get cached?

GrumpyMeow, 2019-12-01:
Hi,

I'm still busy on trying to get my Comet Blue thermostatic-radiator-valve to work.
Partially i'm able to control the TRVs through ble2mqtt, but not optimally yet.

I struggle with two issues:
* If i don't use MAC-whitelisting of my TRVs, i get a lot of errors during the scan. This blocks the ble2mqtt to actually get around to discovering my TRVs.
* The other issue i seem to come across is that it seems that when i read some characteristics, i get (what it sseems as) cached values. I continuosly retrieve the time, but don't see it updating.
Is there some caching which i should take into account?

shmuelzon, 2019-12-04:
Hey,

> If i don't use MAC-whitelisting of my TRVs, i get a lot of errors during the scan. This blocks the ble2mqtt to actually get around to discovering my TRVs.

What kind of errors? Is it simply trying to connect to various devices and at some point fails?
This makes sense if there are other BLE devices, including phones and PCs, around. I also whitelist my devices.

> The other issue i seem to come across is that it seems that when i read some characteristics, i get (what it sseems as) cached values. I continuosly retrieve the time, but don't see it updating.
Is there some caching which i should take into account?

Not that I'm aware. There might be some caching mechanism in the ESP BLE stack, but not in my code. If you see the MQTT publication, that means the BLE stack notified the application that data was read.
What happens if you try to do the same with your mobile phone using a BLE app, e.g. LightBlue?

shmuelzon, 2019-12-18:
@GrumpyMeow, Any updates?

GrumpyMeow, 2020-01-18:
@shmuelzon Hi Asaf,
I added simple support for the "Comet Blue" to a project called ESPHome. This gave a much better understanding of Bluetooth, BLE, Python, esp32 and cpp.
Unfortunatly i think that adding support for "Comet Blue" devices to BLE2MQTT is not suitable. There are two issues i see: 
1. BLE2MQTT uses BLE-scanning for discovering devices. The Comet Blue devices do support that, but the range is then very limited. 
2. Keepalive. BLE2MQTT keeps the connection to devices alive. While the Comet Blue device is meant to "Connect, update, disconnect". And then connect again when new data is requested by the controlling device (phone or esp32), thus polling.

To overcome these "issues". The project must add support for initiating a bluetooth connection to configured devices when data is sent to the MQTT-topic and afterwards disconnect. I think it is do-able. But is different than the current intended implementation of your project.
For now i don't think i've got the energy/time to see if i'm able to add this to the project.

shmuelzon, 2020-01-19:
Thanks for the detailed description!

> BLE2MQTT uses BLE-scanning for discovering devices. The Comet Blue devices do support that, but the range is then very limited.

Not sure I understand what you mean here. As far as I know, there's an active scanning and passive scanning. Those are the only two methods of detecting peripherals. There are devices that are non-connectable (broadcasters), but these broadcasts are also detected while the ESP32 is scanning.

> Keepalive. BLE2MQTT keeps the connection to devices alive. While the Comet Blue device is meant to "Connect, update, disconnect". And then connect again when new data is requested by the controlling device (phone or esp32), thus polling.

I'm not sure why the Comet Blue behaves this way. Seems like a waste to me. I don't have any data to back this up, but I believe the maintaining a connection (utilizing slave latency and other features) is more energy saving than creating new connections. There's also no reason to not allow it as you can still disconnect and reconnect if you want, but that's a different issue.

Anyways, like you said, I guess it doesn't exactly suit this project. It's meant to be a deploy-and-forget solution without the need for an external entity managing it, telling it when to connect, disconnect, etc.

I'm closing this issue now but feel free to reopen it, or a new one, if there's anything I can help with.

closed 2020-01-19 by shmuelzon