shmuelzon/esp32-ble2mqtt #62: Wrong negative temperature representation

mibas74, 2019-12-05:
Hi, 
I've placed one of the Xiaomi Temperature and Humidity sensors outdoor and when the temperature is below zero the reading is wrong (like 6550.5)

shmuelzon, 2019-12-05:
Hey,
I don't have that sensor so I can't test myself, but I'm guessing it has something to do with signed/unsigned numbers.

Could you try making the following modifications and see if it does any good?
```patch
diff --git a/main/broadcasters.c b/main/broadcasters.c
index 068d6f5..518a096 100644
--- a/main/broadcasters.c
+++ b/main/broadcasters.c
@@ -362,7 +362,7 @@ static void mijia_temp_hum_metadata_get(uint8_t *adv_data, size_t adv_data_len,
     cb("MessageCounter", s, ctx);
     if (mijia_temp_hum->data_type == MIJIA_TEMP_HUM_DATA_TYPE_TEMP)
     {
-        sprintf(s, "%.1f", le16toh(*(uint16_t *)mijia_temp_hum->data) / 10.0);
+        sprintf(s, "%.1f", (int16_t)le16toh(*(uint16_t *)mijia_temp_hum->data) / 10.0);
         cb("Temperature", s, ctx);
     }
     else if (mijia_temp_hum->data_type == MIJIA_TEMP_HUM_DATA_TYPE_HUM)
@@ -377,7 +377,7 @@ static void mijia_temp_hum_metadata_get(uint8_t *adv_data, size_t adv_data_len,
     }
     else if (mijia_temp_hum->data_type == MIJIA_TEMP_HUM_DATA_TYPE_TEMP_HUM)
     {
-        sprintf(s, "%.1f", le16toh(*(uint16_t *)mijia_temp_hum->data) / 10.0);
+        sprintf(s, "%.1f", (int16_t)le16toh(*(uint16_t *)mijia_temp_hum->data) / 10.0);
         cb("Temperature", s, ctx);
         sprintf(s, "%.1f",
             le16toh(*(uint16_t *)(mijia_temp_hum->data + 2)) / 10.0);
```

shmuelzon, 2019-12-18:
Hey @mibas74,

Could you please try the above patch and let me know if it works for you?
If it does, I'll update the project.

Thanks

mibas74, 2019-12-18:
Hi,
unfortunately i had broken hdd in my computer, so there was necessity to buy new one and restore system. I've applied the changes but in the meantime the weather has changed and it does not seems that will be below zero within couple of days. I'll give an answer as soon as will be possible.

Best Regards
Michal

https://github.com/shmuelzon/esp32-ble2mqtt/issues/62#issuecomment-566877096
-- Wysłane za pomocą K-9 Mail.

shmuelzon, 2019-12-18:
> unfortunately i had broken hdd in my computer

I'm sorry to hear that, hope all is well now.

> it does not seems that will be below zero within couple of days

You could always put the sensor in the freezer :)
Anyways, thanks for the update!

mibas74, 2019-12-29:
Hi,
snow and negative temperatures are back and it seems that everything is Ok now, so we can close this issue.

shmuelzon, 2019-12-29:
Thanks for the update!
I’m reopening the issue until I commit the above patch.

commit f5038145 (2019-12-30): Fix Mijia temperature sign

Fixes #62
closed 2019-12-30 by shmuelzon