shmuelzon/esp32-ble2mqtt #63: stops working after 2-3 minutes

KristianRaue, 2019-12-06:
Hi,

I think this is really a great project. And I was very easily able to get it to work. But unfortunately it only works for a short time and then no more BLE Events arrive, only the uptime and freememory message continue to arrive.

When I use "make monitor" I get some errors during the first minutes but it seems to work. But then once it shows the "E (900892) BT_HCI: btu_hcif_hdl_command_complete opcode 0x200c status 0xc" Error messages, no more BLE messages arrive.

Somehow I think maybe the parallel operation of BLE an WiFi might cause a problem? 

Are there any special settings necessary in the "make menuconfig" ?
[sdkconfig.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/3929858/sdkconfig.txt)

Here is a typical log of the last seconds before the BLE messages stop to appear:

I (241842) BLE2MQTT: Discovered BLE device: 31:ce:6e:74:b6:e6 (RSSI: -38), conne
cting
W (271852) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=
0x0100
E (271862) BLE: Open failed, status = 0x85
I (272032) BLE2MQTT: Discovered BLE device: 31:ce:6e:74:b6:e6 (RSSI: -40), conne
cting
W (302042) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=
0x0100
E (302042) BLE: Open failed, status = 0x85
I (302162) BLE2MQTT: Discovered BLE device: ac:23:3f:25:24:a5 (RSSI: -62), conne
cting
I (306332) BLE2MQTT: Discovered BLE device: 74:7e:d9:21:96:df (RSSI: -74), conne
cting
I (307292) BLE2MQTT: Connected to device: ac:23:3f:25:24:a5, scanning
I (307872) BLE2MQTT: Publishing: ble/ac:23:3f:25:24:a5/GenericAccess/DeviceName
= nRF5x
I (307942) BLE2MQTT: Publishing: ble/ac:23:3f:25:24:a5/GenericAccess/Appearance
= 0
I (307972) BLE2MQTT: Publishing: ble/ac:23:3f:25:24:a5/GenericAccess/PeripheralP
referredConnectionParameters = 16,60,0,400
I (308002) BLE2MQTT: Publishing: ble/ac:23:3f:25:24:a5/GenericAccess/CentralAddr
essResolution = 1
I (308062) BLE2MQTT: Publishing: ble/ac:23:3f:25:24:a5/DeviceInformation/Manufac
W (338182) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=769 reason=0x0100
W (338192) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1025 reason=0x0002
E (338192) BLE: Open failed, status = 0x85
E (338192) BLE: Open failed, status = 0x85
W (338202) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

E (338212) BT_HCI: btu_hcif_hdl_command_complete opcode 0x200c status 0xc
W (338212) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

E (338222) BT_HCI: btu_hcif_hdl_command_complete opcode 0x200c status 0xc
E (338232) BLE: Scanning starting failed, status: 0x1
E (338232) BLE: Scanning starting failed, status: 0x1
W (349832) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
W (617572) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=513 reason=0x0013
I (617582) BLE: Connection closed, reason = 0x13
I (617582) BLE2MQTT: Disconnected from device: 74:7e:d9:21:96:df
E (900892) BT_HCI: btu_hcif_hdl_command_complete opcode 0x200c status 0xc

shmuelzon, 2019-12-06:
Looks like it's trying to connect to two peripherals simultaneously, though that shouldn't be a problem...

Which ESP-IDF version are you using?

I'm not sure it is, but if you thing it's related to Wi-Fi/BT coexistence, there are a few settings in `make menuconfig` you can play with.

KristianRaue, 2019-12-06:
I hope i pulled the newest version when I executed "git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git"

shmuelzon, 2019-12-06:
Could you please try and use version 3.2.2, as is in the README? It's what I use and I don't see these issues.

KristianRaue, 2019-12-07:
I reverted to 3.2.2 but no change.  

PS: I am particularly interested in the Broadcasters. So I have 3 Beacons running in the same room (in addition to a least 5 more BLE devices (watch, smartphone, headphone, etc). The 3 Beacons publish 9 messages per second (3 messages per beacon and second). Maybe this amount of traffic is the problem.

I gave up and I am now using 
https://github.com/thomasleveil/bluetooth-beacon-mqtt-exporter
on a Raspberry Pi 4 under Node.js. Works perfectly.

shmuelzon, 2019-12-08:
I've had ibeacons that broadcast more frequently without any issues.

Anyways, I'm sorry to hear it didn't work out for you. I'm attaching here a binary version of this project, just in case you'd want to try again. You'd still need your own configuration (spiffs.bin) though.

[ble2mqtt.bin.zip](https://github.com/shmuelzon/esp32-ble2mqtt/files/3935961/ble2mqtt.bin.zip)

closed 2019-12-08 by shmuelzon