shmuelzon/esp32-ble2mqtt #65: mac os build error

cooldesigns, 2019-12-24:
```
Toolchain path: /Users/eva/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
LD build/bootloader/bootloader.elf
/Users/eva/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Users/eva/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/Users/eva/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/Users/eva/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/Users/eva/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
make[1]: *** [/Users/eva/Documents/Particle/esp32-ble2mqtt/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/eva/Documents/Particle/esp32-ble2mqtt/build/bootloader/bootloader.bin] Error 2
```
In my .zshrc
`export IDF_PATH=~/esp/esp-idf
export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
export PATH="$IDF_PATH/tools:$PATH"
export PATH="/usr/local/opt/binutils/bin:$PATH"
export PATH="/usr/local/opt/gettext/bin:$PATH"`

```
esp ls
esp-idf          hello_world      xtensa-esp32-elf
```

Also can't compile hello_world app with same errors.

shmuelzon, 2019-12-29:
Hey,

I'm sorry but I can't help you much with that. I'm able to compile on a Mac and have the same toolchain version. The only difference I see is that I haven't switched to zsh on my Mac and keep using Bash.
Have you gone through all of the documentation for setting up a build environment?

cooldesigns, 2019-12-31:
Yes I did 4 times. I have success on Ubuntu, but no lock in os x. I will ask to esp-idf community. I think this steps is so long and difficult if you have an error. Not friendly not meaning errors. 

I think simply downloading and flashing binary file and writing configuration file and compiling on web, after download sending it with bluetooth or usb is more comfortable. If this is possible.

shmuelzon, 2020-01-19:
I can start adding binary images of the application itself in future releases/tags but that would be a problem for the configuration image/partition. Without a valid one, you won't be able to remotely change it. If/when I add an on-boarding process, e.g. publish an access point when no configuration was provided and allow uploading a configuration file, that would be useful. I'll create a separate issue for that and close this one as I can't help much with the build issues.

closed 2020-01-19 by shmuelzon