shmuelzon/esp32-ble2mqtt #66: Add Support for On-Boarding AP

shmuelzon, 2020-01-19:
When no valid configuration is available, or when we fail connecting to the configure Wi-Fi SSID, the ESP32 should create its own access point with a simple web-based interface allowing uploading a SPIFFS configuration image.
Once available, it should be possible to include binary images with future releases.

cooldesigns, 2020-01-19:
That is the solution! Thank you.

ben-barnard, 2020-01-21:
This would be fantastic!

russelltomkins, 2020-09-06:
Hi @shmuelzon , any progress on this feature? Coming from Tasmota (pre-compiled) and OpenMQTTGateway (platform IO), I'm struggling badly (well failing) to get the software toolchain deployed correctly into a blank machine to build this project. 

Pre-compiled with an Onboard-AP would be awesome!

olegvg, 2020-11-15:
Hi @shmuelzon 
Any news from you? How is your PC ?

shmuelzon, 2020-11-16:
Hey, I finally got it set up last weekend and finished porting this project to ESP-IDF v4.1.
I hope to be able to start adding new features next weekend. I will probably start with an HTTP server first for uploading a new image and configuration and then add the access point.
To truly support people wanting to load this application without compiling anything, I will also need to add support for uploading specific files (config.json, certificates, etc.) instead of a complete SPIFFS image, but that would come later.
Maybe once the basic building blocks are ready, someone might be able to help out with that.

commit 7e0e7c82 (2020-12-16): Add softAP in case configuration failed loading

Closes #66
closed 2020-12-16 by shmuelzon