shmuelzon/esp32-ble2mqtt #67: Duplicated value responses after reconnects

bssstudio, 2020-02-03:
Greetings!

I am having an issue with duplicated responses or duplicated published messages to MQTT (can't test) when requesting a value from a BLE device. After reboot of ESP32 and when the device is connected the first time, everything works as expected i.e. I get one response to a request. 
```
BLE2MQTT/00:1a:22:14:xx:xx/Connected true
BLE2MQTT/00:1a:22:14:xx:xx/Owner BLE2MQTT-3930
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/DeviceName CC-RT-BLE
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/Appearance 0
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/PeripheralPrivacyFlag false
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/PeripheralPreferredConnectionParameters 0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/GenericAttribute/ServiceChanged 0,0
BLE2MQTT/00:1a:22:14:xx:xx/DeviceInformation/ManufacturerNameString eq-3
BLE2MQTT/00:1a:22:14:xx:xx/DeviceInformation/ModelNumberString CC-RT-BLE
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/9e5d1e47-5c13-43a0-8635-82ad38a1386f/347f7608-2e2d-47eb-913b-75d4edc4de3b 0,16,3,2

BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09/Set 128,00

BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
```

After some time the BLE device gets disconnected and reconnected and after requesting a value, there are multiple responses. 

```
BLE2MQTT/00:1a:22:14:xx:xx/Connected false
BLE2MQTT/00:1a:22:14:xx:xx/Connected true
BLE2MQTT/00:1a:22:14:xx:xx/Owner BLE2MQTT-3930
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/DeviceName CC-RT-BLE
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/Appearance 0
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/PeripheralPrivacyFlag false
BLE2MQTT/00:1a:22:14:xx:xx/GenericAccess/PeripheralPreferredConnectionParameters 0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/GenericAttribute/ServiceChanged 0,0
BLE2MQTT/00:1a:22:14:xx:xx/DeviceInformation/ManufacturerNameString eq-3
BLE2MQTT/00:1a:22:14:xx:xx/DeviceInformation/ModelNumberString CC-RT-BLE
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/9e5d1e47-5c13-43a0-8635-82ad38a1386f/347f7608-2e2d-47eb-913b-75d4edc4de3b 0,16,3,2

BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09/Set 128,00

BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/d0e8434d-cd29-0996-af41-6c90f4e0eb2a 2,1,8,0,4,10,0,0,0,0,24,3,42,34,7
BLE2MQTT/00:1a:22:14:xx:xx/3e135142-654f-9090-134a-a6ff5bb77046/3fa4585a-ce4a-3bad-db4b-b8df8179ea09 128,0,2,1,13,44,55,126,40,144,0,0,0,0,0,0
```

I am speculating that some housekeeping is not done when device is disconnected, however I have been unable to track it in the code.

Thank you!

shmuelzon, 2020-02-03:
Hey,

I'm assuming those are the MQTT published messages. Could you please post the (preferably debug) logs from the ESP32?
You can see them by either connecting to the serial interface or via the remote logging mechanism described in the README file.

That should help make it clearer if those values were read multiple times, published multiple times or were actually received as notifications from the BLE device.
Also, please make sure you're using the ESP-IDF version described in the README, that's the only version I work and test with.

Thanks

bssstudio, 2020-02-03:
Hi!

I have managed to trace it down to internal subscription_list. When connecting to the BLE device, the mqtt subscription list is updated - a topic and callback is added regardless if the topic already exists.
I have to check if there this list is not properly updated when the device is disconnected.
One other possible solution would be to check if the list already contains the topic in question and just update the callback rather then appending a new topic to the subscription_list (I am testing this solution and for now it is working). 

The ESP-IDF version is 3.2.2. 

I will post logs soon, just to compile something that makes sense.

Thanks!

shmuelzon, 2020-02-04:
Hey,

The topic subscription is removed when the device is disconnected. [ble_on_device_disconnected()](https://github.com/shmuelzon/esp32-ble2mqtt/blob/f50381451c045eaf009d4cb6470c83f56e81e4b0/main/ble2mqtt.c#L340) is called when a BLE device is disconnected and the MQTT subscriptions are removed for each of its characteristics in [ble_on_characteristic_removed()](https://github.com/shmuelzon/esp32-ble2mqtt/blob/f50381451c045eaf009d4cb6470c83f56e81e4b0/main/ble2mqtt.c#L322).
In any case, your log shows that the ESP32 publishes multiple topics and also characteristics that are different than the one you're setting, so I don't think the subscription list is at fault here.

Debug logs will tell us if the publications were caused from incoming notifications or not so we can tell if there is really a problem or if the BLE peripheral actually sent the data.

Thanks

shmuelzon, 2020-07-09:
Hey,

Is still an issue or can I close it?

Thanks

shmuelzon, 2020-11-18:
Closing due to inactivity

closed 2020-11-18 by shmuelzon