shmuelzon/esp32-ble2mqtt #69: Errors in compilation ble2mqtt

cjogo, 2020-02-18:


cjogo, 2020-02-18:
VMWARE running UBUNTU :-

Followed the installation guide from espressif then used “make flash”

Any idea as to how to overcome the 2 errors seen at the bottom?

.>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

CC build/lwip/port/esp32/netif/dhcp_state.o
CC build/lwip/port/esp32/netif/ethernetif.o
CC build/lwip/port/esp32/debug/lwip_debug.o
AR build/lwip/liblwip.a
CC build/main/mqtt.o
CC build/main/resolve.o
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/resolve.c: In function 'resolve_host':
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/resolve.c:42:38: warning: passing argument 3 of 'mdns_query_a' from incompatible pointer type [-Wincompatible-pointer-types]
         if (!mdns_query_a(buf, 5000, (struct ip4_addr *)&ip.sin_addr))
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/resolve.c:3:
/home/david/esp/esp-idf/components/mdns/include/mdns.h:340:83: note: expected 'esp_ip4_addr_t *' {aka 'struct esp_ip4_addr *'} but argument is of type 'struct ip4_addr *'
 esp_err_t mdns_query_a(const char * host_name, uint32_t timeout, esp_ip4_addr_t * addr);
                                                                  ~~~~~~~~~~~~~~~~~^~~~
CC build/main/config.o
CC build/main/ble_utils.o
CC build/main/gatt.o
CC build/main/log.o
CC build/main/ble.o
CC build/main/ota.o
CC build/main/wifi.o
In file included from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:3:
/home/david/esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c: In function 'event_handler':
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:77:5: warning: 'mdns_handle_system_event' is deprecated [-Wdeprecated-declarations]
     mdns_handle_system_event(ctx, event);
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:7:
/home/david/esp/esp-idf/components/mdns/include/mdns.h:365:11: note: declared here
 esp_err_t mdns_handle_system_event(void *ctx, system_event_t *event) __attribute__((deprecated));
           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c: In function 'wifi_connect':
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:110:5: error: unknown type name 'esp_wpa2_config_t'; did you mean 'esp_eth_config_t'?
     esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
     ^~~~~~~~~~~~~~~~~
     esp_eth_config_t
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:110:32: error: implicit declaration of function 'WPA2_CONFIG_INIT_DEFAULT'; did you mean 'WIFI_INIT_CONFIG_DEFAULT'? [-Werror=implicit-function-declaration]
     esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
                                ^~~~~~~~~~~~~~~~~~~~~~~~
                                WIFI_INIT_CONFIG_DEFAULT
In file included from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:2:
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:151:25: error: too many arguments to function 'esp_wifi_sta_wpa2_ent_enable'
         ESP_ERROR_CHECK(esp_wifi_sta_wpa2_ent_enable(&config));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:6:
/home/david/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant/esp_wpa2.h:36:11: note: declared here
 esp_err_t esp_wifi_sta_wpa2_ent_enable(void);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c: In function 'wifi_initialize':
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:162:5: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
     tcpip_adapter_init();
     ^~~~~~~~~~~~~~~~~~
In file included from /home/david/esp/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from /home/david/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/david/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/david/esp/esp-idf/components/esp_event/include/esp_event_loop.h:3,
                 from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:3:
/home/david/esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:163:5: warning: 'esp_event_loop_init' is deprecated [-Wdeprecated-declarations]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
     ^~~~~~~~~~~~~~~
In file included from /home/david/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/david/esp/esp-idf/components/esp_event/include/esp_event_loop.h:3,
                 from /home/david/esp/esp-idf/ble2mqtt/esp32-ble2mqtt-master/main/wifi.c:3:
[... cut ...] idf/components/esp_event/include/esp_event_legacy.h:227:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
           ^~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/home/david/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'wifi.o' failed
make[1]: *** [wifi.o] Error 1
/home/david/esp/esp-idf/make/project.mk:604: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

shmuelzon, 2020-02-18:
Hey,

I would start by using the ESP-IDF version specified in the README file.
Things have changed in v4.0 and I’ve yet yo update.

Good day

valioiv, 2020-03-07:
> Hey,
> 
> I would start by using the ESP-IDF version specified in the README file.
> Things have changed in v4.0 and I’ve yet yo update.
> 
> Good day

What is your estimation on the needed efforts? I'm open to help if you have an idea...

shmuelzon, 2020-03-07:
Do you have any specific reason for using ESP-IDF v4.0? v3.2.2 is very stable and runs for weeks on my devices without reboots or disconnections.

If you must use v4.0, then it seems that the mDNS API has changed and so has WiFi in general. My usage of them was mainly taken from the examples in ESP-IDF so I would start by comparing this repo with the updated examples and update the code accordingly. I doubt there has been a substantial change and there's probably some guide somewhere on how to update applications to the new API. At first glance, I don't think it should take very long but, unfortunately, I don't have time to invest in it right now.

valioiv, 2020-03-07:
> Do you have any specific reason for using ESP-IDF v4.0? v3.2.2 is very stable and runs for weeks on my devices without reboots or disconnections.
>
> If you must use v4.0, then it seems that the mDNS [...]

Actually personally I don't need latest and greatest ESP-IDF. I just need working BLE-to-MQTT bridge for a PoC for something else. I just compiled and flashed your code and it works great for the half an hour I'm playing with it! And yes, your approach is completely correct - I'd also suggest getting the new ESP-IDF and implement the "app" BLE bridge "logic" on top of it considering the updated APIs. If have some free time soon I can try.

shmuelzon, 2020-03-07:
@cjogo, have you tried working with ESP-IDF v3.2.2? Can I close this issue for now?
I have plans to update to v4.0 but it, probably, won't be any time soon.

cjogo, 2020-03-08:
Have tried to compile v3.2.2 but to no avail. Will try again but I suggest you close this topic.
Will persevere nevertheless.

shmuelzon, 2020-03-09:
OK. If you need any help with compiling against v3.2.2, let me know

closed 2020-03-09 by shmuelzon

cjogo, 2020-03-15:
Hi

My apologies for disturbing you again. However I've compiled the system as
per your instructions all ok but.... I'm suffering the same fate as
stops working after 2-3 minutes #63
However this topic has been closed without an obvious solution.

There has been a reference by you to configurations in make menuconfig. Any
idea where I can start?

Thanks again for the excellent effort much appreciated.

Kind regards

Compton

shmuelzon, 2020-03-15:
Hey,

The configuration I mentioned relate to WiFi/BLE coexistence, but I don’t think that should be an issue. You could also make use if the binary attached to that issue and flash it on your device to rule that problem out.

If nothing works, please post debug logs and I’ll see if anything jumps out to me.

Good day