shmuelzon/esp32-ble2mqtt #70: iBeacon reception

iw2lsi, 2020-04-29:
Hi,

  I'm running the gateway but I do not see any iBeacon informations... and I'm sure there are at least 3 around me...

For testing, I'm subscribing to the topic where I expect the gateway to push its data using a raspberry... I see that something is published (ie uptime and the version below)...

Am I doing something wrong ? do I need to set something special in order to forward iBeacons to the mqtt server ?

   thanks 

           Giampaolo

mosquitto_sub -d -h 192.168.1.100 -t BLE2MQTT-F8D4/#
Client mosqsub|6135-piBeaconGW sending CONNECT
Client mosqsub|6135-piBeaconGW received CONNACK (0)
Client mosqsub|6135-piBeaconGW sending SUBSCRIBE (Mid: 1, Topic: BLE2MQTT-F8D4/#, QoS: 0)
Client mosqsub|6135-piBeaconGW received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|6135-piBeaconGW received PUBLISH (d0, q0, r1, m0, 'BLE2MQTT-F8D4/Version', ... (18 bytes))
v0.12.0-2-gf503814
Client mosqsub|6135-piBeaconGW received PUBLISH (d0, q0, r1, m0, 'BLE2MQTT-F8D4/ConfigVersion', ... (64 bytes))
90deb01b7f1b9adf92207b727126199aa45f85071fbff7e7ea07c9290eb90372
Client mosqsub|6135-piBeaconGW received PUBLISH (d0, q0, r1, m0, 'BLE2MQTT-F8D4/Uptime', ... (3 bytes))
360
Client mosqsub|6135-piBeaconGW received PUBLISH (d0, q0, r1, m0, 'BLE2MQTT-F8D4/FreeMemory', ... (5 bytes))
73380

shmuelzon, 2020-04-30:
Hey Giampaolo,

Out of curiosity, how are you sure there are iBeacons near you? Do you have other device that does detect them?

Anyways, there are a few possible reasons I can think of:
1. The iBeacons are out of range and are not detected by the esp32
2. The iBeacons are connectable (this is against the standard) and the esp32 has connected to them and they no longer broadcast their information
3. A bug that causes the devices to not be correctly identified as iBeacons

I would suggest taking a look at the logs, either via serial interface or remotely, and see if there are any clues there.
To overcome the second point, I would create an empty whitelist so the esp32 won’t connect to any device, this is just to make sure it does identify the beacons.
If there’s a bug, the we need to add some debug messages in `main/broadcasters.c`, specifically to `ibeacon_is_broadcaster()` to understand why it’s no detected as an iBeacon.

Please post your logs and I’ll see if I can find any clues there.

Good day

iw2lsi, 2020-04-30:
Hello,

> Hey Giampaolo,
>
> Out of curiosity, how are you sure there are iBeacons near you? Do you
> have other device that does detect them?
>
I'm using my Iphone with iLocate and eBeacon applications for iOS... below
are the screenshots:

[image: IMG_7780.PNG]
[image: IMG_7781.PNG]

> Anyways, there are a few possible reasons I can think of:
>
>    1. The iBeacons are out of range and are not detected by the esp32
>
> this is for sure not the case... I can see them even on raspberry pi
running node-red with BeaconScanner plugin
from node-red-contrib-beacon-scanner module:

Below is the node-red debug print:

id: "e5ac9675168a"
address: "e5:ac:96:75:16:8a"
localName: null
txPowerLevel: null
rssi: -61
beaconType: "iBeacon"
iBeacon: object
advertisement: object
manufacturerData: buffer[26]raw <http://192.168.1.100:1880/#>
[0 … 9]
0: 0x4c
1: 0x0
2: 0x2
3: 0x15
4: 0xeb
5: 0xef
6: 0xd0
7: 0x83
8: 0x70
9: 0xa2
[10 … 19]
10: 0x47
11: 0xc8
12: 0x98
13: 0x37
14: 0xe7
15: 0xb5
16: 0x63
17: 0x4d
18: 0xf5
19: 0x24
[20 … 25]
20: 0x8a
21: 0x16
22: 0x75
23: 0x96
24: 0xcb
25: 0x64
serviceData: array[0]
serviceUuids: array[0]
solicitationServiceUuids: array[0]
serviceSolicitationUuids: array[0]
uuid: "EBEFD083-70A2-47C8-9837-E7B5634DF524"
major: 35350
minor: 30102txPower: 203

>    1. The iBeacons are connectable (this is against the standard) and the
>    esp32 has connected to them and they no longer broadcast their information
>
> I'm using two different beacon (iB002M and iB004N-Plus) from
http://www.ankhmaway.com.cn... attached are the user guides.
Both of them can be put in connectable mode for configuring purposes... but
I'm sure they are not during the tests.

>
>    1. A bug that causes the devices to not be correctly identified as
>    iBeacons
>
> this might be the problem... using ibeacon scanner script from Radius
Network on a raspberry pi zero I had to change a byte from 2A to 2B as
shown below in order to see something... not sure what this byte is
representing.

#!/bin/bash
# iBeacon Scan by Radius Networks

if [[ $1 == "parse" ]]; then
  packet=""
  capturing=""
  count=0
  while read line
  do
    count=$[count + 1]
    if [ "$capturing" ]; then
      if [[ $line =~ ^[0-9a-fA-F]{2}\ [0-9a-fA-F] ]]; then
        packet="$packet $line"
      else
#       if [[ $packet =~ ^04\ 3E\ 2A\ 02\ 01\ .{26}\ 02\ 01\ .{14}\ 02\ 15
]]; then
        if [[ $packet =~ ^04\ 3E\ 2B\ 02\ 01\ .{26}\ 02\ 01\ .{14}\ 02\ 15
]]; then

          UUID=`echo $packet | sed 's/^.\{69\}\(.\{47\}\).*$/\1/'`
          MAJOR=`echo $packet | sed 's/^.\{117\}\(.\{5\}\).*$/\1/'`
          MINOR=`echo $packet | sed 's/^.\{123\}\(.\{5\}\).*$/\1/'`
          POWER=`echo $packet | sed 's/^.\{129\}\(.\{2\}\).*$/\1/'`
          UUID=`echo $UUID | sed -e 's/\ //g' -e
's/^\(.\{8\}\)\(.\{4\}\)\(.\{4\}\)\(.\{4\}\)\(.\{12\}\)$/\1-\2-\3-\4-\5/'`
          MAJOR=`echo $MAJOR | sed 's/\ //g'`
          MAJOR=`echo "ibase=16; $MAJOR" | bc`
          MINOR=`echo $MINOR | sed 's/\ //g'`
          MINOR=`echo "ibase=16; $MINOR" | bc`
          POWER=`echo "ibase=16; $POWER" | bc`
          POWER=$[POWER - 256]
          if [[ $2 == "-b" ]]; then
            echo "$UUID $MAJOR $MINOR $POWER"
          else
            echo "UUID: $UUID MAJOR: $MAJOR MINOR: $MINOR POWER: $POWER"
          fi
        fi
        capturing=""
        packet=""
      fi
    fi

    if [ ! "$capturing" ]; then
      if [[ $line =~ ^\> ]]; then
        packet=`echo $line | sed 's/^>.\(.*$\)/\1/'`
        capturing=1
      fi
    fi
  done
else
  sudo hcitool lescan --duplicates 1>/dev/null &
  if [ "$(pidof hcitool)" ]; then
   sudo hcidump --raw | ./$0 parse $1
  fi
fi

this is what the script is now printing:

UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 40951 MINOR: 353 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 40951 MINOR: 353 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 40951 MINOR: 353 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 40951 MINOR: 353 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 35350 MINOR: 30102 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 40951 MINOR: 353 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53
UUID: EBEFD083-70A2-47C8-9837-E7B5634DF524 MAJOR: 29586 MINOR: 38295 POWER:
-53

> I would suggest taking a look at the logs, either via serial interface or
> remotely, and see if there are any clues there.
> To overcome the second point, I would create an empty whitelist so the
> esp32 [...]
thanks for the hints... I'll try to investigate on ESP32 right now... but
I'm totally new to this platform...

  best regards and thanks for the prompt support

       Giampaolo

shmuelzon, 2020-04-30:
Hey,

There are 3 conditions that must be met for the BLE peripheral to be detected as an iBeacon and they are validated [here](https://github.com/shmuelzon/esp32-ble2mqtt/blob/f50381451c045eaf009d4cb6470c83f56e81e4b0/main/broadcasters.c#L75).

The Apple company ID and beacon type seem to line up with the advertisement data you posted. The third validation is the length of the data. Assuming I counted the size of `ibeacon_t’ correctly, it should be 25 bytes but the payload you posted has 26.
I don’t know if that’s OK as far as the iBeacon standard goes but I followed Apple’s documentation when I first implemented it and it worked with the beacons I had.

Maybe you can change the length validation to be equal or larger than the ‘ibeacon_t’ structure and let me know if that helps?
I can send a patch later, if needed, when I’ll be at my computer.

Thanks!

iw2lsi, 2020-04-30:
Hello Assaf,

I've done some investigations... it seems that my beacons are really
26-byte length... I do not understand why...

Also... the application seems stuck trying to connect to one of the nearby
devices... by commenting non-advertising stuff I was able to see my first
beacons published on MQTT as reported.

Below are also the (dirty!) changes I made on your code so far...

   thanks

          Giampaolo

I (2124) wifi:AP's beacon interval = 100352 us, DTIM period = 1
I (2134) WiFi: Connected
I (4884) event: sta ip: 192.168.1.238, mask: 255.255.255.0, gw: 192.168.1.1
I (4884) BLE2MQTT: Connected to WiFi, connecting to MQTT
I (4884) MQTT: Connecting MQTT client
I (4884) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (4994) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (5004) MQTT: MQTT client connected
I (5004) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (5104) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
E (5104) Broadcaster: Unexpected ibeacon len: 29
I (5124) Broadcaster: checking for ibeacon on a 0-byte advertising frame:
'03039ffe17169ffe0000000000000000000000000000000000000000'
E (5134) Broadcaster: Unexpected ibeacon len: 0
I (5134) Broadcaster: checking for ibeacon on a 26-byte advertising frame:
'0201041bff4c000215ebefd08370a247c89837e7b5634df52473929597cb5a'
I (5144) Broadcaster: ibeacon found!
I (5154) BLE2MQTT: Discovered iBeacon broadcaster
I (5214) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
E (5214) Broadcaster: Unexpected ibeacon len: 29
I (5294) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092002281e61443d8a2a7864f09b637489d5fd695d8c81bc58b4'
E (5294) Broadcaster: Unexpected ibeacon len: 29
I (5314) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
E (5314) Broadcaster: Unexpected ibeacon len: 29
I (5594) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092002281e61443d8a2a7864f09b637489d5fd695d8c81bc58b4'
E (5594) Broadcaster: Unexpected ibeacon len: 29
I (5624) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
E (5624) Broadcaster: Unexpected ibeacon len: 29
I (5814) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092002281e61443d8a2a7864f09b637489d5fd695d8c81bc58b4'
E (5814) Broadcaster: Unexpected ibeacon len: 29
I (5824) Broadcaster: checking for ibeacon on a 0-byte advertising frame:
'03039ffe17169ffe0000000000000000000000000000000000000000'
E (5834) Broadcaster: Unexpected ibeacon len: 0
I (5894) Broadcaster: checking for ibeacon on a 0-byte advertising frame:
'03039ffe17169ffe0243466f5a3736537266583800000171ca99a99c'
E (5894) Broadcaster: Unexpected ibeacon len: 0
I (5904) Broadcaster: checking for ibeacon on a 0-byte advertising frame:
'03039ffe17169ffe0000000000000000000000000000000000000000'
E (5914) Broadcaster: Unexpected ibeacon len: 0
I (5934) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
E (5934) Broadcaster: Unexpected ibeacon len: 29
I (6014) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092002281e61443d8a2a7864f09b637489d5fd695d8c81bc58b4'
E (6014) Broadcaster: Unexpected ibeacon len: 29
I (6034) Broadcaster: checking for ibeacon on a 29-byte advertising frame:
'1eff060001092006a416ca61c574e717006239bbc757943a1bc20e05ce8198'
[... 211 lines cut ...]
    uint8_t len;
    ibeacon_t *beacon = ibeacon_data_get(adv_data, adv_data_len, &len);

char temp[5];
char buffer[255];
memset(buffer, 0x00, 255);
for(int i=0; i<adv_data_len; i++){
   sprintf(temp, "%02x", adv_data[i]);
   strcat(buffer, temp);
   }
    ESP_LOGI(TAG, "checking for ibeacon on a %d-byte advertising frame:
'%s'", len, buffer);
    if (!beacon || ( len != sizeof(ibeacon_t) && len !=
1+sizeof(ibeacon_t)) ){
    ESP_LOGE(TAG, "Unexpected ibeacon len: %d",  len);
        return 0;
        }

    ESP_LOGI(TAG, "ibeacon found!");
    /* Technically, we should also check the device is BLE only and
     * non-connectable, but most iBeacon simulators don't advertise as such
*/
    return le16toh(beacon->company_id) == 0x004C /* Apple */ &&
        le16toh(beacon->beacon_type) == 0x1502;
}

iw2lsi, 2020-04-30:
Hello Assaf,

  I wonder if my beacons are actually altbeacons instead of iBeacons...

looking at https://github.com/AltBeacon/spec , AD LENGTH [MFG SPECIFIC] is
0x1b as in my frames...

     Giampaolo

AltBeacon Protocol Format

The AltBeacon advertisement makes use of the Manufacturer Specific
Advertising Data structure as defined in *Bluetooth Specification Version
4.0, Volume 6, Part B, Section 2.3 Advertising Channel PDU*.

The AltBeacon advertisement is made up of a 1-byte length field, 1-byte
type field and two-byte company identifier, as prescribed by the
Manufacturer Specific Advertising Data structure format, followed by 24
additional bytes containing the beacon advertisement data.

[image: Exploded View]
<https://github.com/AltBeacon/spec/blob/master/altbeacon-spec-exploded-view.png>

See the AltBeacon Protocol Data and AltBeacon Protocol Fields as described
below for information on the specific fields, their descriptions and
accepted values.
<https://github.com/AltBeacon/spec#altbeacon-protocol-diagram>AltBeacon
Protocol Diagram

[image: AltBeacon Protocol Format]
<https://github.com/AltBeacon/spec/blob/master/altbeacon-protocol-diagram.png>
<https://github.com/AltBeacon/spec#altbeacon-protocol-fields>AltBeacon
Protocol Fields
Field NameDescriptionAccepted Values
AD LENGTH [MFG SPECIFIC] Length of the type and data portion of the
Manufacturer Specific advertising data structure. 0x1B
AD TYPE [MFG SPECIFIC] Type representing the Manufacturer Specific
advertising data structure. 0xFF
MFG ID The beacon device manufacturer's company identifier code. The little
endian representation of the beacon device manufacturer's company code as
maintained by the Bluetooth SIG assigned numbers database
BEACON CODE The AltBeacon advertisement code The big endian representation
of the value 0xBEAC
BEACON ID A 20-byte value uniquely identifying the beacon The big endian
representation of the beacon identifier. For interoperability purposes, the
first 16+ bytes of the beacon identifier should be unique to the
advertiser's organizational unit. Any remaining bytes of the beacon
identifier may be subdivided as needed for the use case.
REFERENCE RSSI A 1-byte value representing the average received signal
strength at 1m from the advertiser A signed 1-byte value from 0 to -127
MFG RESERVED Reserved for use by the manufacturer to implement special
features A 1-byte value from 0x00 to 0xFF. Interpretation of this value is
to be defined by the manufacturer and is to be evaluated based on the MFG ID
 value

shmuelzon, 2020-04-30:
Hey,

The MFG ID doesn’t match the AltBeacon spec (0xBEAC) so it’s not that exactly. It might be sone weird hybrid complements of the far east :)

It does match Apple’s ID and includes the iBeacon type so someone tried to publish as iBeacon, but it’s a bit off.

Good day

iw2lsi, 2020-04-30:
Hello Assaf...

  I agree with you... I wonder how can it be that my iPhone apps are
actually showing these as iBeacons...

Anyway, I wrote to the manufacturer... hope that they can clarify their
frames...

   thanks for the great code and for your support!

    Best regards from Italy

          Giampaolo

shmuelzon, 2020-04-30:
Hey,

I’m guessing the only validate the advertisement length is at least 25 just ignore the rest.
Don’t know if that’s OK or not.

I’m wondering if the rest of the fields are OK or not. Are the UUID and major/minor values what you expect to see? What about the detected distance from the beacons?

Good day

iw2lsi, 2020-04-30:
> I’m guessing they only validate the advertisement length is at least 25

Hello Assaf,

   yes... all the other fields are consistent...

      Giampaolo

shmuelzon, 2020-05-05:
Hey Giampaolo,

I modified the code to allow a larger payload for iBeacon.
Let me know if you still have issues detecting your beacons.

Thanks!

closed 2020-05-05 by shmuelzon

iw2lsi, 2020-05-05:
Hello Assaf,

  I'm having some IDF environment issues right now... I'll test it
tomorrow...

             giampaolo

iw2lsi, 2020-05-06:
Hello Assaf,

  am I wrong or the mkspiffs folder is gone on your latest commit ?

           Best Regards

              Giampaolo

shmuelzon, 2020-05-06:
Hey,

I think (and hope) you're wrong. It's still there but it's a submodule. If you didn't use the `--recursive` flag when cloning this repository, you'll need to execute `git submodule update --init --recursive` to pull it.

Good day

iw2lsi, 2020-05-06:
Hello Assaf...

  ok... my bad.... I forgot the recursive flag...

 I've tested the new release and it does not seem to work for me...
iBeacons are now detected but not forwarded to MQTT.

In order to see something on my nodered, I had to comment

//    ble_set_on_device_discovered_cb(_ble_on_device_discovered);
//    ble_set_on_device_connected_cb(_ble_on_device_connected);
//    ble_set_on_device_disconnected_cb(_ble_on_device_disconnected);
//    ble_set_on_device_services_discovered_cb(
//        _ble_on_device_services_discovered);
//    ble_set_on_device_characteristic_value_cb(
//        _ble_on_device_characteristic_value);
//    ble_set_on_passkey_requested_cb(ble_on_passkey_requested);

as previously...

In my case I'm interested in advertising-only messages... so disabling the
other features is good for me as it's also much faster in detecting beacons.

   below is the log with your original code and with the commented out lines

           Giampaolo

original

I (164) BLE2MQTT: Version: v0.12.0-3-g478cb92
I (164) Config: Initializing configuration
I (374) Config: version:
96a3dd69ee4267d1bfb78ac6e7222a9fe1ddc2af1bc30cbe8e8005853bf04ba2
I (374) Log: Initializing remote logging
I (374) OTA: Initializing OTA
I (394) wifi:wifi driver task: 3ffd7ff0, prio:23, stack:3584, core=0
I (394) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (394) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (424) wifi:wifi firmware version: 6fd589c
I (424) wifi:config NVS flash: enabled
I (424) wifi:config nano formating: disabled
I (424) wifi:Init dynamic tx buffer num: 32
I (434) wifi:Init data frame dynamic rx buffer num: 32
I (434) wifi:Init management frame dynamic rx buffer num: 32
I (444) wifi:Init management short buffer num: 32
I (444) wifi:Init static rx buffer size: 1600
I (454) wifi:Init static rx buffer num: 10
I (454) wifi:Init dynamic rx buffer num: 32
I (454) wifi:Set ps type: 0

I (464) Resolve: Initializing resolver
I (464) BTDM_INIT: BT controller compile version [0ab6383]
I (474) system_api: Base MAC address is not set, read default base MAC
address from BLK0 of EFUSE
I (574) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (954) WiFi: Connecting to SSID fastPanky
I (964) wifi:mode : sta (84:0d:8e:0c:f8:d4)
I (1084) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2074) wifi:state: init -> auth (b0)
I (2074) wifi:state: auth -> assoc (0)
I (2074) wifi:state: assoc -> run (10)
I (2094) wifi:connected with fastPanky, aid = 9, channel 1, BW20, bssid =
b4:fb:f9:93:df:48
I (2094) wifi:security type: 3, phy: bgn, rssi: -77
I (2104) wifi:pm start, type: 0

I (2104) WiFi: Connected
I (2184) wifi:AP's beacon interval = 100352 us, DTIM period = 1
[... 136 lines cut ...]
I (105674) BLE2MQTT: Discovered iBeacon broadcaster
I (114844) BLE2MQTT: Discovered iBeacon broadcaster
I (117714) BLE2MQTT: Discovered iBeacon broadcaster
I (118024) BLE2MQTT: Discovered iBeacon broadcaster
I (119024) BLE2MQTT: Discovered iBeacon broadcaster
I (120034) BLE2MQTT: Discovered iBeacon broadcaster
I (120864) BLE2MQTT: Discovered iBeacon broadcaster
I (123044) BLE2MQTT: Discovered iBeacon broadcaster
I (123724) BLE2MQTT: Discovered iBeacon broadcaster
I (124874) BLE2MQTT: Discovered iBeacon broadcaster
I (125054) BLE2MQTT: Discovered iBeacon broadcaster
I (126054) BLE2MQTT: Discovered iBeacon broadcaster
I (126874) BLE2MQTT: Discovered iBeacon broadcaster
I (127064) BLE2MQTT: Discovered iBeacon broadcaster
I (128874) BLE2MQTT: Discovered iBeacon broadcaster
I (130884) BLE2MQTT: Discovered iBeacon broadcaster
I (131084) BLE2MQTT: Discovered iBeacon broadcaster
I (133104) BLE2MQTT: Discovered iBeacon broadcaster
I (134104) BLE2MQTT: Discovered iBeacon broadcaster
I (136894) BLE2MQTT: Discovered iBeacon broadcaster
I (138904) BLE2MQTT: Discovered iBeacon broadcaster
I (139114) BLE2MQTT: Discovered iBeacon broadcaster
I (140124) BLE2MQTT: Discovered iBeacon broadcaster
I (141134) BLE2MQTT: Discovered iBeacon broadcaster
I (142924) BLE2MQTT: Discovered iBeacon broadcaster

shmuelzon, 2020-05-06:
Hey,

If you don't want to connect to any device, you don't need to modify the code, you can just use an empty whitelist:
```json
{
  "ble": {
    "whitelist": []
}
```
I see the iBeacons are detected correctly so you should get something via MQTT.
Just to be sure, are you connected to the same broker as the ESP32?
If you try to listen on all topics, e.g.
```bash
mosquitto_sub -h <host> -t "#" -v
```
Do you see anything? Do you see the bookeeping topics published by the ESP32, uptime and free memory?

Thanks

iw2lsi, 2020-05-06:
I'm testing it... I'll be back ASAP

   Giampaolo

iw2lsi, 2020-05-06:
Hello,

  it works with original code plus the whitelist... that's fine!

  without the whitelist it stuck somewhere...

  below is the log with the commented code, with the original code + the
whitelist, with the original code without the whitelist

     Giampaolo

code commented:

mosquitto_sub -h 192.168.1.100 -t "#" -v
boot connection estabilished
owntracks/test/owtest
{"batt":21,"lon":9.6633142791778432,"acc":10,"p":98.689964294433594,"bs":1,"vel":0,"vac":3,"lat":45.69189453125,"inregions":["bEmma-1582378867","bRedTag-1585528286"],"t":"b","conn":"w","tst":1588790264,"alt":247,"_type":"location","tid":"ow"}
test 1584091893059
tele/tasmota/LWT Online
tele/portone/LWT Online
BLE2MQTT-F8D4/Version v0.12.0-3-g478cb92
BLE2MQTT-F8D4/ConfigVersion
96a3dd69ee4267d1bfb78ac6e7222a9fe1ddc2af1bc30cbe8e8005853bf04ba2
BLE2MQTT-F8D4/Uptime 2100
BLE2MQTT-F8D4/FreeMemory 77912
65:6f:69:10:96:bd/Connected false
65:6f:69:10:96:bd/Owner BLE2MQTT-F8D4
65:6f:69:10:96:bd/GenericAccess/DeviceName iPhone
65:6f:69:10:96:bd/GenericAccess/Appearance 64
4c:9c:87:d2:3d:9f/Connected false
4c:9c:87:d2:3d:9f/Owner BLE2MQTT-F8D4
4c:9c:87:d2:3d:9f/GenericAccess/DeviceName iPhone
4c:9c:87:d2:3d:9f/GenericAccess/Appearance 64
14:bb:6e:ce:cb:f2/Connected false
14:bb:6e:ce:cb:f2/Owner BLE2MQTT-F8D4
14:bb:6e:ce:cb:f2/GenericAccess/DeviceName [TV] pankyTV
14:bb:6e:ce:cb:f2/GenericAccess/Appearance 0
4c:d8:2f:da:8c:02/Connected false
4c:d8:2f:da:8c:02/Owner BLE2MQTT-F8D4
4c:d8:2f:da:8c:02/GenericAccess/DeviceName iPhone
4c:d8:2f:da:8c:02/GenericAccess/Appearance 64
6a:60:82:5e:33:e6/Connected false
6a:60:82:5e:33:e6/Owner BLE2MQTT-F8D4
6a:60:82:5e:33:e6/GenericAccess/DeviceName iPhone
6a:60:82:5e:33:e6/GenericAccess/Appearance 64
5f:ac:7b:76:db:0b/Connected false
5f:ac:7b:76:db:0b/Owner BLE2MQTT-F8D4
5f:ac:7b:76:db:0b/GenericAccess/DeviceName iPhone
5f:ac:7b:76:db:0b/GenericAccess/Appearance 64
6b:71:15:73:41:9e/Connected false
6b:71:15:73:41:9e/Owner BLE2MQTT-F8D4
6b:71:15:73:41:9e/GenericAccess/DeviceName iPhone
6b:71:15:73:41:9e/GenericAccess/Appearance 64
BLE2MQTT-F8D4/d3:a5:92:73:97:95/Type iBeacon
BLE2MQTT-F8D4/d3:a5:92:73:97:95/RSSI -87
BLE2MQTT-F8D4/d3:a5:92:73:97:95/UUID ebefd083-70a2-47c8-9837-e7b5634df524
BLE2MQTT-F8D4/d3:a5:92:73:97:95/Major 29586
BLE2MQTT-F8D4/d3:a5:92:73:97:95/Minor 38295
BLE2MQTT-F8D4/d3:a5:92:73:97:95/Distance 50.12
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Type iBeacon
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/RSSI -59
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/UUID ebefd083-70a2-47c8-9837-e7b5634df524
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Major 40951
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Minor 353
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Distance 2.00
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Type iBeacon
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/RSSI -64
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/UUID ebefd083-70a2-47c8-9837-e7b5634df524
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Major 40951
BLE2MQTT-F8D4/d5:97:61:01:f7:9f/Minor 353
[... 502 lines cut ...]
object
original: buffer[13]raw <http://192.168.1.100:1880/#>
[0 … 9]
0: 0xa9
1: 0xef
2: 0x7b
3: 0xc
4: 0xd7
5: 0xa3
6: 0xba
7: 0x41
8: 0x0
9: 0x5a
[10 … 12]
10: 0x32
11: 0x42
12: 0x8
payload: object
tempval: 23.329999923706055
humival: 44.587890625
topic: "callsign"
datetime: "06 May 2020 20:22:12"
caller: "THLS01"
isvalid: true
_msgid: "56363fe4.f114d"

shmuelzon, 2020-05-06:
Hey,

The `ף▒AZ2B` payload isn't coming from the ESP23 but another entity on the MQTT bus.
I see both the Uptime and FreeMemory topics are published so I don't think the application is stuck. It may try to connect to a device that's not really connectable and that causes the BLE stack to wait before trying to receive anything else, like the iBeacon.
At least that's what it looks like in your [previous comment](https://github.com/shmuelzon/esp32-ble2mqtt/issues/70#issuecomment-624808407).
I'm not sure it's really a bug, but feel free to open a new issue with updated logs from the ESP32 and I'll take another look.

Thanks

iw2lsi, 2020-05-06:
Hello Assaf

  you are right... the raw payload is coming from a LoRa temperature
device... the message is then forwarded using a LoRa to MQTT gateway
running on a raspberry...

The strange is that as soon this payload comes into the flow, nothing else
is shown from your side... I do not understand why...

 Anyway... I'm pretty fine with the whitelist behaviour... and ble2mqtt is
exactly doing what I was looking for!

     thanks a lot Assaf and best Regards from Italy...

         Giampaolo

shmuelzon, 2020-05-06:
Great, I’m glad it worked out for you!

commit 478cb925 (2020-05-05): Allow larger iBeacons payloads

Some iBeacon devices advertise additional payload after the iBeacon data. This
seems to be allowed and still detected by other applications.

Fixes #70