shmuelzon/esp32-ble2mqtt #71: Help Setting Value - nRF8001

mukundv, 2020-05-04:
Hi There
What you've got here is awesome. With this, I'm `almost` able to control the [Nordic NRF8001](https://github.com/adafruit/Adafruit_nRF8001) based switch called [SwitchRobot](https://www.adaprox.com.au/switchbot).
This is my Config
```
  "ble": {
    "whitelist": [],
    "services": {
      "whitelist": [
        "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
      ],
      "definitions": {
        "6E400001-B5A3-F393-E0A9-E50E24DCCA9E": {
          "name": "Nordic UART Service"
        }
      }
    },
    "characteristics": {
      "whitelist": [
        "6E400003-B5A3-F393-E0A9-E50E24DCCA9E",
        "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
      ],
      "definitions": {
        "6E400003-B5A3-F393-E0A9-E50E24DCCA9E": {
          "name": "WriteChar",
          "types": ["uint8", "uint8", "uint8", "uint8", "uint8", "uint8", "uint8"]
        },
        "6E400002-B5A3-F393-E0A9-E50E24DCCA9E": {
          "name": "NotifyChar"
        }
      }
    }
  }
```
From MQTT, I can see the value when the button is pressed like below:
`<<MAC>>/Nordic UART Service/WriteChar - 42,66,83,36,13,10,0`
If I set the same the same by publishing to the topic `<<MAC>>/Nordic UART Service/WriteChar/Set` I don't get any response. 

However, If I set the same via [nrfConnect App](https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Connect-for-mobile) I'm able to control the switch.

Any help in this space is appreciated.

Mukund

shmuelzon, 2020-05-04:
Hey Mukund,

Could you please provide debug logs of when the device connects and when you publish the MQTT topic for writing to the characteristic?
That should give me a better insight as to what’s going on.

BTW - if your characteristic is a simple byte array, you don’t need to define its types. That’s the default.

Let me know...

mukundv, 2020-05-05:
Hi @shmuelzon 
Thanks for the quick reply. Here's the log, but unfortunately none of the MQTT-->BLE messages seem to be appearing on this. All the `BLE2MQTT` messages  you see here are when the physical button is pressed on the device. 
I did a `make remote-monitor` to capture these. Is there any other way to get the debug logs?

```
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Listening on port 5000
2020-05-05 17:11:43.693678 (192.168.86.88): D (4188521) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:43.696705 (192.168.86.88): D (4188531) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:49.632202 (192.168.86.88): [0;32mI (3400) BLE2MQTT: Connected to WiFi, connecting to MQTT[0m
2020-05-05 17:11:49.834381 (192.168.86.88): D (3410) MQTT_CLIENT: MQTT client_id=esp32-ble2mqtt[0m
2020-05-05 17:11:49.835328 (192.168.86.88): D (3420) MQTT_CLIENT: Core selection disabled[0m
2020-05-05 17:11:49.835862 (192.168.86.88): D (3420) TRANS_TCP: [sock=55],connecting to server IP:192.168.86.165,Port:1883...[0m
2020-05-05 17:11:49.836484 (192.168.86.88): D (3460) MQTT_CLIENT: Transport connected to mqtt://192.168.86.165:1883[0m
2020-05-05 17:11:49.836977 (192.168.86.88): [0;32mI (3470) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000[0m
2020-05-05 17:11:49.837362 (192.168.86.88): D (3480) MQTT_CLIENT: Connected[0m
2020-05-05 17:11:50.039644 (192.168.86.88): D (3480) BLE2MQTT: Queuing event MQTT_CONNECTED[0m
2020-05-05 17:11:50.040471 (192.168.86.88): [0;32mI (3480) BLE2MQTT: Connected to MQTT, scanning for BLE devices[0m
2020-05-05 17:11:50.041926 (192.168.86.88): D (3490) MQTT: Subscribing to BLE2MQTT-52FC/OTA/Firmware[0m
2020-05-05 17:11:50.043006 (192.168.86.88): D (3490) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful[0m
2020-05-05 17:11:50.043877 (192.168.86.88): D (3500) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT-52FC/OTA/Firmware, id: 33933, type=8 successful[0m
2020-05-05 17:11:50.044576 (192.168.86.88): D (3510) MQTT: Subscribing to BLE2MQTT/OTA/Firmware[0m
2020-05-05 17:11:50.045428 (192.168.86.88): D (3510) MQTT_CLIENT: mqtt_enqueue id: 33933, type=8 successful[0m
2020-05-05 17:11:50.046471 (192.168.86.88): D (3520) OUTBOX: ENQUEUE msgid=33933, msg_type=8, len=33, size=33[0m
2020-05-05 17:11:50.245301 (192.168.86.88): D (3540) OUTBOX: ENQUEUE msgid=36815, msg_type=8, len=28, size=61[0m
2020-05-05 17:11:50.246513 (192.168.86.88): D (3550) MQTT_CLIENT: msg_type=9, msg_id=33933[0m
2020-05-05 17:11:50.248069 (192.168.86.88): D (3560) MQTT_CLIENT: msg_type=9, msg_id=33933[0m
2020-05-05 17:11:50.249069 (192.168.86.88): D (3560) MQTT_CLIENT: pending_id=22914, pending_msg_count = 3[0m
2020-05-05 17:11:50.249728 (192.168.86.88): D (3560) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT-52FC/OTA/Config, id: 22914, type=8 successful[0m
2020-05-05 17:11:50.250440 (192.168.86.88): I (3570) wifi: ampdu: ignore deleting tx BA0
2020-05-05 17:11:50.251062 (192.168.86.88): D (3570) OUTBOX: DELETED msgid=33933, msg_type=8, remain size=28[0m
2020-05-05 17:11:50.449946 (192.168.86.88): D (3600) MQTT_CLIENT: msg_type=9, msg_id=36815[0m
2020-05-05 17:11:50.450810 (192.168.86.88): D (3600) MQTT_CLIENT: msg_type=9, msg_id=36815[0m
2020-05-05 17:11:50.451348 (192.168.86.88): D (3600) OUTBOX: ENQUEUE msgid=22914, msg_type=8, len=31, size=59[0m
2020-05-05 17:11:50.451778 (192.168.86.88): D (3610) MQTT_CLIENT: pending_id=2045, pending_msg_count = 3[0m
2020-05-05 17:11:50.452160 (192.168.86.88): D (3620) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT/OTA/Config, id: 2045, type=8 successful[0m
2020-05-05 17:11:50.654140 (192.168.86.88): D (3650) MQTT_CLIENT: msg_type=9, msg_id=22914[0m
2020-05-05 17:11:50.655406 (192.168.86.88): D (3650) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)[0m
2020-05-05 17:11:50.656376 (192.168.86.88): D (3650) MQTT_CLIENT: pending_id=2045, pending_msg_count = 2[0m
2020-05-05 17:11:50.657225 (192.168.86.88): D (3660) OUTBOX: DELETED msgid=22914, msg_type=8, remain size=0[0m
2020-05-05 17:11:50.657938 (192.168.86.88): D (3670) MQTT_CLIENT: Subscribe successful[0m
2020-05-05 17:11:50.658449 (192.168.86.88): D (3690) MQTT_CLIENT: msg_type=9, msg_id=2045[0m
2020-05-05 17:11:50.658894 (192.168.86.88): D (3690) MQTT_CLIENT: msg_type=9, msg_id=2045[0m
2020-05-05 17:11:50.858361 (192.168.86.88): D (4050) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (e0:21:1d:6f:25:34, -64)[0m
2020-05-05 17:11:50.859171 (192.168.86.88): [0;32mI (4060) BLE2MQTT: Discovered BLE device: e0:21:1d:6f:25:34 (RSSI: -64), not connecting[0m
2020-05-05 17:11:50.859754 (192.168.86.88): D (4140) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:50.860784 (192.168.86.88): D (4140) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (76:20:c7:6d:0d:f1, -61)[0m
2020-05-05 17:11:50.861817 (192.168.86.88): [0;32mI (4140) BLE2MQTT: Discovered BLE device: 76:20:c7:6d:0d:f1 (RSSI: -61), not connecting[0m
2020-05-05 17:11:50.862868 (192.168.86.88): D (4240) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:50.863850 (192.168.86.88): D (4320) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:50.864918 (192.168.86.88): D (4420) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:11:51.063164 (192.168.86.88): D (4940) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
[... cut ...]
2020-05-05 17:11:58.640959 (192.168.86.88): D (12340) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-05 17:12:02.738944 (192.168.86.88): D (16620) BLE: Received GATTC event 10 (ESP_GATTC_NOTIFY_EVT), gattc_if 1[0m
2020-05-05 17:12:02.739902 (192.168.86.88): D (16620) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (ef:ff:d9:bc:b9:e5, 6e400003-b5a3-f393-e0a9-e50e24dcca9e, 0x3ffed684, 7)[0m
2020-05-05 17:12:02.740656 (192.168.86.88): [0;32mI (16630) BLE2MQTT: Publishing: ef:ff:d9:bc:b9:e5/Nordic UART Service/WriteChar = 42,65,83,36,13,10,0[0m

```

shmuelzon, 2020-05-05:
Hey Mukund,

From your log, the `6e400003-b5a3-f393-e0a9-e50e24dcca9e` characteristic supports only notifications while the `6e400002-b5a3-f393-e0a9-e50e24dcca9e` service supports writing.
I think you mixed the two in your configuration file.

Try to switch the two names in your configuration file and you should get the UART data from the `NotifyChar` topic and set to the `WriteChar` topic.

Let me know

mukundv, 2020-05-07:
I think that is one step closer. Now I can see the MQTT --> BLE comms in the logs, but still no joy in getting the switch to flip.
Just to avoid confusion, I removed the `WriteChar` and `NotifyChar` config leaving the guid as is.
Here's the logs.
```

Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Listening on port 5000
2020-05-07 07:43:32.550796 (192.168.86.88): [0;32mI (3380) BLE2MQTT: Connected to WiFi, connecting to MQTT[0m
2020-05-07 07:43:32.560336 (192.168.86.88): [0;32mI (3390) MQTT: Connecting MQTT client[0m
2020-05-07 07:43:32.561065 (192.168.86.88): D (3390) MQTT_CLIENT: MQTT client_id=esp32-ble2mqtt[0m
2020-05-07 07:43:32.561791 (192.168.86.88): D (3400) MQTT_CLIENT: Core selection disabled[0m
2020-05-07 07:43:32.753835 (192.168.86.88): D (3400) TRANS_TCP: [sock=55],connecting to server IP:192.168.86.165,Port:1883...[0m
2020-05-07 07:43:32.755091 (192.168.86.88): D (3430) MQTT_CLIENT: Transport connected to mqtt://192.168.86.165:1883[0m
2020-05-07 07:43:32.755976 (192.168.86.88): [0;32mI (3430) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000[0m
2020-05-07 07:43:32.756703 (192.168.86.88): D (3440) MQTT_CLIENT: Connected[0m
2020-05-07 07:43:32.757382 (192.168.86.88): [0;32mI (3440) MQTT: MQTT client connected[0m
2020-05-07 07:43:32.758014 (192.168.86.88): D (3440) BLE2MQTT: Queuing event MQTT_CONNECTED[0m
2020-05-07 07:43:33.164099 (192.168.86.88): D (3460) MQTT: Subscribing to BLE2MQTT-52FC/OTA/Firmware[0m
2020-05-07 07:43:33.165089 (192.168.86.88): D (3460) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful[0m
2020-05-07 07:43:33.165748 (192.168.86.88): D (3470) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT-52FC/OTA/Firmware, id: 52888, type=8 successful[0m
2020-05-07 07:43:33.166195 (192.168.86.88): D (3470) MQTT: Subscribing to BLE2MQTT/OTA/Firmware[0m
2020-05-07 07:43:33.166583 (192.168.86.88): D (3480) MQTT_CLIENT: mqtt_enqueue id: 52888, type=8 successful[0m
2020-05-07 07:43:33.166955 (192.168.86.88): D (3490) OUTBOX: ENQUEUE msgid=52888, msg_type=8, len=33, size=33[0m
2020-05-07 07:43:33.369597 (192.168.86.88): D (3500) MQTT: Subscribing to BLE2MQTT-52FC/OTA/Config[0m
2020-05-07 07:43:33.371525 (192.168.86.88): D (3510) MQTT_CLIENT: mqtt_enqueue id: 4666, type=8 successful[0m
2020-05-07 07:43:33.372611 (192.168.86.88): D (3510) MQTT_CLIENT: msg_type=9, msg_id=52888[0m
2020-05-07 07:43:33.374323 (192.168.86.88): D (3520) OUTBOX: ENQUEUE msgid=4666, msg_type=8, len=28, size=61[0m
2020-05-07 07:43:33.375113 (192.168.86.88): D (3520) MQTT_CLIENT: msg_type=9, msg_id=52888[0m
2020-05-07 07:43:33.573505 (192.168.86.88): D (3540) OUTBOX: DELETED msgid=52888, msg_type=8, remain size=28[0m
2020-05-07 07:43:33.574771 (192.168.86.88): D (3550) MQTT_CLIENT: Subscribe successful[0m
2020-05-07 07:43:33.575880 (192.168.86.88): D (3550) MQTT_CLIENT: mqtt_enqueue id: 59711, type=8 successful[0m
2020-05-07 07:43:33.576873 (192.168.86.88): D (3560) OUTBOX: ENQUEUE msgid=59711, msg_type=8, len=31, size=59[0m
2020-05-07 07:43:33.577858 (192.168.86.88): D (3570) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT/OTA/Config, id: 54175, type=8 successful[0m
2020-05-07 07:43:33.578792 (192.168.86.88): D (3580) BLE: Starting BLE scan[0m
2020-05-07 07:43:33.579713 (192.168.86.88): D (3580) MQTT_CLIENT: msg_type=9, msg_id=4666[0m
2020-05-07 07:43:33.779406 (192.168.86.88): D (3600) OUTBOX: DELETED msgid=4666, msg_type=8, remain size=31[0m
2020-05-07 07:43:33.780498 (192.168.86.88): D (3610) MQTT_CLIENT: Subscribe successful[0m
2020-05-07 07:43:33.781687 (192.168.86.88): D (3610) MQTT_CLIENT: msg_type=9, msg_id=59711[0m
2020-05-07 07:43:33.783398 (192.168.86.88): D (3620) MQTT_CLIENT: msg_type=9, msg_id=59711[0m
2020-05-07 07:43:33.785539 (192.168.86.88): D (3620) MQTT_CLIENT: pending_id=54175, pending_msg_count = 2[0m
2020-05-07 07:43:33.982869 (192.168.86.88): D (3660) MQTT_CLIENT: pending_id=54175, pending_msg_count = 1[0m
2020-05-07 07:43:33.984534 (192.168.86.88): D (3670) MQTT_CLIENT: Subscribe successful[0m
2020-05-07 07:43:33.986202 (192.168.86.88): D (4040) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-07 07:43:33.986954 (192.168.86.88): D (4040) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (ef:ff:d9:bc:b9:e5, -60)[0m
2020-05-07 07:43:33.987826 (192.168.86.88): [0;32mI (4040) BLE2MQTT: Discovered BLE device: ef:ff:d9:bc:b9:e5 (RSSI: -60), connecting[0m
2020-05-07 07:43:33.988553 (192.168.86.88): D (4060) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)[0m
2020-05-07 07:43:33.989089 (192.168.86.88): D (4620) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1[0m
2020-05-07 07:43:33.989598 (192.168.86.88): D (4620) BLE: GATTC event 40 wasn't handled[0m
2020-05-07 07:43:34.187670 (192.168.86.88): D (4630) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)[0m
2020-05-07 07:43:34.597936 (192.168.86.88): D (5350) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-07 07:43:34.598998 (192.168.86.88): D (5350) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (d7:27:56:9a:6a:01, -60)[0m
2020-05-07 07:43:34.600296 (192.168.86.88): [0;32mI (5350) BLE2MQTT: Discovered BLE device: d7:27:56:9a:6a:01 (RSSI: -60), not connecting[0m
2020-05-07 07:43:34.602096 (192.168.86.88): D (5450) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
[... cut ...]
2020-05-07 07:43:54.464522 (192.168.86.88): D (25330) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)[0m
2020-05-07 07:43:54.667936 (192.168.86.88): D (25440) BLE: Queue timer expired[0m
2020-05-07 07:43:54.668751 (192.168.86.88): D (25440) BLE: Dequeue: type: 1, device: ef:ff:d9:bc:b9:e5, char: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, len: 7, val: 0x3ffed094[0m
2020-05-07 07:43:55.077534 (192.168.86.88): D (25840) BLE: Received GATTC event 4 (ESP_GATTC_WRITE_CHAR_EVT), gattc_if 1[0m

 
```

shmuelzon, 2020-05-07:
Hey Mukund,

It looks like everything should be working. The Set topic is received and sent to the BLE stack and a write confirmation (from the stack) is received.
I can think of 3 possible issues:
1. The data is sent correctly but not what the device expects. You say you’ve sent the exact same thing via mobile app and it works?
2. The data is incorrectly converted from the MQTT payload to the binary data. All I currently have in the logs is the payload length (7 in this case) which corresponds to what you want to send. I’ll try to send you a patch tomorrow that prints the actual data that’s going to be written
3. The device says is supports both write and write without response and this project gives write (with response) a priority. Maybe the device only support write without response? It should be simple enough to test this with another patch

I’ll keep you posted when I can prepare those.

Good day

mukundv, 2020-05-08:
Thanks for the response.

> 1. The data is sent correctly but not what the device expects. You say you’ve sent the exact same thing via mobile app and it works?

Well not exactly; I should've been clear in the earlier message. I did some more digging around and here's the clipping of `btsnoop_hci.log` that was captured while using the [app](https://play.google.com/store/apps/details?id=com.release.adaprox.controller) for another switchbot. 
[image]
As you can see, the client sends the following commands first
`*G$`, `*E$`, `*H$`, `*C$` 
Then sends the actual command `*A$` that actuates the switches.
I've tried to summarise the comms in the picture below
[image]

I don't understand how `*A$` translates to `42,65,83,36,13,10,0`

Hope this shines some light into the investigation.

Thanks again.

Mukund

shmuelzon, 2020-05-08:
Hey,

The numbers are ASCII values. 42 is `*`, 65 is `A` and so on. Just look up an ASCII table and you’ll find all of the values. 13 and 10 are `\r\n` and zero is a NULL terminate, i.e. end of string.
From your image, when turning on and off, the client doesn’t send the \r\n so you should remove that as well. I can’t tell if it sends the NULL termination or not, I’ll need the capture file for that.
Anyways, if you don’t need to send any special characters like the `\r\n` or null terminator, just define the characteristic type as a string and you can publish `*A$` directly on the MQTT bus.

Let me know how it works out for you

mukundv, 2020-05-09:
Thank you so much for your support. I'm now able to control them via MQTT.
The final config is as follows.
```
    "publish": {
      "retain": false
    },
    "topics": {
      "prefix": "",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "log": {
    "host": "224.0.0.200",
    "port": 5000
  },
  "ble": {
    "whitelist": [
      "EF:21:E7:1F:10:AD",
      "EF:FF:D9:BC:B9:E5"
    ],
    "services": {
      "whitelist": [
        "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
      ],
      "definitions": {
        "6E400001-B5A3-F393-E0A9-E50E24DCCA9E": {
          "name": "Nordic UART Service"
        }
      }
    },
    "characteristics": {
      "whitelist": [
        "6E400003-B5A3-F393-E0A9-E50E24DCCA9E",
        "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
      ],
      "definitions":{
        "6E400003-B5A3-F393-E0A9-E50E24DCCA9E":{
          "name":"NotifyChar"
        },
        "6E400002-B5A3-F393-E0A9-E50E24DCCA9E":{
          "name":"WriteWithoutNotifyChar"
        }
      }
    }
  }
```
To actuate the switch>
Publish `42,66,36,0` to  `<<MAC>>/Nordic UART Service/WriteWithoutNotifyChar` to make the switch go from bottom to top

Publish `42,65,36,0` to  `<<MAC>>/Nordic UART Service/WriteWithoutNotifyChar` to make the switch go from top to bottom

Other useful info:
Switch sends `42,71,67,36,13,10,0` when plugged in and charging
Switch sends `42,71,56,56,36,0` when unplugged

Todo: Work out how to get the battery level.

Thank you very much for the help.

Mukund

closed 2020-05-09 by mukundv

shmuelzon, 2020-05-09:
Glad you got it working for you!
Regarding the battery level, there are only 4 things the app queries the device. Based on the responses, `*G$` would be my first guess since the response is a dual figure value