shmuelzon/esp32-ble2mqtt #73: BLE heart rate sensor (Garmin Dual) frequently disconnects and reconnects while broadcasting to ESP32

schnodda, 2020-06-19:
Hey,
I managed to set up my ESP32 to receive BLE heart rate data via my Garmin Dual sensor. Unfortunately, while broadcasting the BLE sender frequently disconnects and reconnects from the ESP32 receiver. As a result, I receive around 4 measurements per 20sec which is not acceptable for my use case.

I have also tried it with the service data from a smart trainer Bike (Bike Power service) and in this case, the ESP32 receives the data continuously without issue. The HRM sensor works perfectly fine with other devices. 

So I assume the issue is inherent in the configurations of ESP32 and the HRM sensor. I hope, there is something in your great code, that could be adjusted in order to allow some more time for a notification receiption before the time out is called.

This is the monitor log from the BLE recipient indicating that a 0x08 error (BLE_HCI_CONNECTION_TIMEOUT) is the fault. The error is displayed ca. on the 20th line from the bottom (bold):

> Python requirements from /home/baruch/esp/esp-idf-v3.2.2/requirements.txt are satisfied.
> MONITOR
> --- idf_monitor on /dev/ttyUSB0 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:6176
> ho 0 tail 12 room 4
> load:0x40078000,len:10180
> load:0x40080400,len:6660
> entry 0x40080764
> I (30) boot: ESP-IDF v3.2.2 2nd stage bootloader
> I (30) boot: compile time 23:50:31
> I (30) boot: Enabling RNG early entropy source...
> I (35) boot: SPI Speed      : 40MHz
> I (39) boot: SPI Mode       : DIO
> I (43) boot: SPI Flash Size : 4MB
> I (47) boot: Partition Table:
> I (50) boot: ## Label            Usage          Type ST Offset   Length
> I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> I (80) boot:  3 ota_0            OTA app          00 10 00010000 00180000
> I (88) boot:  4 ota_1            OTA app          00 11 00190000 00180000
> I (95) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
> I (103) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
> I (110) boot: End of partition table
> I (114) boot: No factory image, trying OTA 0
> I (119) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x381c4 (229828) map
> I (209) esp_image: segment 1: paddr=0x000481ec vaddr=0x3ffbdb60 size=0x0594c ( 22860) load
> I (218) esp_image: segment 2: paddr=0x0004db40 vaddr=0x40080000 size=0x00400 (  1024) load
> 0x40080000: _WindowOverflow4 at /home/baruch/esp/esp-idf-v3.2.2/components/freertos/xtensa_vectors.S:1779
> 
> I (219) esp_image: segment 3: paddr=0x0004df48 vaddr=0x40080400 size=0x020c8 (  8392) load
> I (230) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xf0f90 (987024) map
> 0x400d0018: _flash_cache_start at ??:?
> 
> I (581) esp_image: segment 5: paddr=0x00140fb0 vaddr=0x400824c8 size=0x141b8 ( 82360) load
> 0x400824c8: ets_timer_arm at /home/baruch/esp/esp-idf-v3.2.2/components/esp32/ets_timer_legacy.c:86
> 
> I (629) boot: Loaded app from partition at offset 0x10000
> I (629) boot: Disabling RNG early entropy source...
> I (630) cpu_start: Pro cpu up.
> I (634) cpu_start: Starting app cpu, entry point is 0x40081020
> 0x40081020: call_start_cpu1 at /home/baruch/esp/esp-idf-v3.2.2/components/esp32/cpu_start.c:246
> 
> I (0) cpu_start: App cpu up.
> I (644) heap_init: Initializing. RAM available for dynamic allocation:
> I (651) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
> I (657) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
> I (663) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
> I (669) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
> I (675) heap_init: At 3FFD33D0 len 0000CC30 (51 KiB): DRAM
> I (681) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (688) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (694) heap_init: At 40096680 len 00009980 (38 KiB): IRAM
[... 61 lines cut ...]
> I (17884) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/BatteryService/BatteryLevel = 85
> I (17904) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/HeartRate/BodySensorLocation = 1
> I (26894) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/HeartRate/HeartRateMeasurement = 0,76
**> I (29894) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/HeartRate/HeartRateMeasurement = 16,75,833
> I (29894) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/HeartRate/HeartRateMeasurement = 16,74,847
> W (35894) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0008
> I (35894) BLE: Connection closed, reason = 0x8
> I (35894) BLE2MQTT: Disconnected from device: dc:5b:41:b4:1b:c9
> I (38954) BLE2MQTT: Discovered BLE device: dc:5b:41:b4:1b:c9 (RSSI: -63), connecting
> I (41444) BLE2MQTT: Connected to device: dc:5b:41:b4:1b:c9, scanning**
> I (42064) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/GenericAccess/DeviceName = HRM-Dual:341261
> I (42094) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/GenericAccess/Appearance = 833
> I (42174) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/GenericAccess/PeripheralPreferredConnectionParameters = 200,400,0,400
> I (42204) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/GenericAccess/CentralAddressResolution = 1
> I (42264) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/ManufacturerNameString = Garmin
> I (42294) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/ModelNumberString = 3299
> I (42364) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/SerialNumberString = 3333764365
> I (42394) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/HardwareRevisionString = 1
> I (42484) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/FirmwareRevisionString = 2.30.0
> I (42604) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/DeviceInformation/SoftwareRevisionString = 2.30.0
> I (42684) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/BatteryService/BatteryLevel = 85
> I (42694) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/BatteryService/BatteryLevel = 85
> I (42714) BLE2MQTT: Publishing: dc:5b:41:b4:1b:c9/HeartRate/BodySensorLocation = 1

Thank you so much for your help! :)

shmuelzon, 2020-06-24:
Hey,

> The HRM sensor works perfectly fine with other devices

When you say it works fine with other devices, do you mean other BLE Central devices?
Have you tried using an app like LightBlue and connect to the sensor? Does it keep the connection active?

I'm wondering if it may be related to the connections parameters (interval, latency, timeout) that are configured between the sensor and ESP32. I don't really modify those and keep the default (I think those are whatever the sensor publishes) but I can try to look at what APIs ESP-IDF provide.

Thanks

schnodda, 2020-06-24:
> Hey,
> 
> > The HRM sensor works perfectly fine with other devices
> 
> When you say it works fine with other devices, do you mean other BLE Central devices?
> Have you tried using an app like LightBlue and connect to the sensor? Does it keep the connection active?
> 
> I'm wondering if it may be related to the connections parameters (interval, latency, timeout) that are configured between the sensor and ESP32. I don't really modify those and keep the default (I think those are whatever the sensor publishes) but I can try to look at what APIs ESP-IDF provide.
> Perhaps `esp_ble_gap_set_prefer_conn_params()` is what's needed here, but I would still prefer to adhere to the peripheral's defaults as it would know best what it can do.
> 
> Thanks

Thank you so much for your reply!

I have previously tried to connect to the device to my cell phone via the nRF connect app which worked without issues. My amateurish assumption is that because heart rate sensors measure the HR by "extending" the circuit through your body, they are more prone to disconnects than devices on a constant energy supply (eg. My smart trainer).

Not an expert but your mentioned values were exactly also what I suspected need adjusting. But I struggled to find the relevant values in the code that needed changing. 

Obviously don't expect you adding these modifications to your code, so that my requirements are met. I am perfectly happy with info on what code snippet I would have to add where to implement these changes.

I noticed in the file `components/protocomm/src/simple_ble/simple_ble.c` in the ESP-IDF (3.2.2) that these values are mentioned. Is that what I need to adjust? I couldn't try it out in your code, since this file is already compiled in yours. ( Not sure how to compile that) Maybe that helps you. 

Thanks again!

shmuelzon, 2020-06-24:
Hey,

The file you mentioned is for a peripheral (e.g. BLE sensor), we need the functionality of other side, central.
The following (untested) patch would print whenever these parameters are updated:
```patch
diff --git a/main/ble.c b/main/ble.c
index fd9b919..ec5793d 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -613,6 +613,15 @@ static void gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
         }
         break;
     }
+    case ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT:
+         ESP_LOGI(GATTC_TAG, "update connection params status = %d, min_int = %d, max_int = %d,conn_int = %d,latency = %d, timeout = %d",
+                  param->update_conn_params.status,
+                  param->update_conn_params.min_int,
+                  param->update_conn_params.max_int,
+                  param->update_conn_params.conn_int,
+                  param->update_conn_params.latency,
+                  param->update_conn_params.timeout);
+        break;
     default:
         ESP_LOGD(TAG, "GAP event %d wasn't handled", event);
         break;
```
It should let you know which values were chosen for each connection.

According to the example at `examples/bluetooth/ble_throughput/throughput_client/main/example_ble_client_throughput.c`, the following (also untested) patch should be able to set it:
```patch
diff --git a/main/ble.c b/main/ble.c
index fd9b919..ec5793d 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -272,6 +272,7 @@ int ble_connect(mac_addr_t mac)
     if (!dev)
         return -1;

+    esp_ble_gap_set_prefer_conn_params(mac, 32, 32, 0, 600);
     /* Stop scanning while attempting to connect */
     esp_ble_gap_stop_scanning();
     return esp_ble_gattc_open(g_gattc_if, mac, dev->addr_type, true);
```

I hope this helps.
We could add some field to the configuration file to allow setting, per device, the connection parameters in a generic way.
If you reach some understanding regarding these parameters, or perhaps find we're missing some functionality like setting default values, please let me know and I'll update this project.

Thanks!

shmuelzon, 2020-11-18:
@schnodda, are there any updates on this? You could also try the update version of this project and it now uses ESP-IDF v4.1 and hopefully some things would be fixed.