shmuelzon/esp32-ble2mqtt #76: Error: Invalid JSON in configuration file.

mtzro2003, 2020-09-12:
I'm getting `Invalid JSON` error when trying to `make flash`.
I can't see what's wrong with the below config.
JSONLint says it's valid... 
What am I missing? 

```
{
  "wifi": {
    "ssid": "XXXXXXXXXXX",
    "password": "XXXXXXXXXXX"
  },
  "mqtt": {
    "server": {
      "host": "XX.XX.XX.XX",
      "port": 1883,
      "username": "XXXXXX",
      "password": "XXXXXXXXXXXXXXX"
    },
    "publish": {
      "retain": true
    },
    "topics" :{
      "prefix": "ble2mqtt",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
    "ble": {
    "whitelist": [ "FF:FF:3D:11:47:B7" ],
    "services": {
      "definitions": {}
    },
    "characteristics": {
      "definitions": {
        "00002a19-0000-1000-8000-00805f9b34fb": {
          "name": "battery-level"
        }
      }
    },
    "passkeys": {}
  }
}
```

shmuelzon, 2020-09-12:
Hey,

Yes, it seems valid. I check it by passing it to `python -m json.tool`, see https://github.com/shmuelzon/esp32-ble2mqtt/blob/f5012b119f3df750a83ee7df8228da229751b672/Makefile#L70-L72

Could you running it manually in the command line as see the output?
`cat data/config.json | python -m json.tool`.

Let me know if how works out.

mtzro2003, 2020-09-12:
Well, ahem, looks like the posted config was not the same as the one in the file 😊 
It was a comma missing, evidently...
Silly me!
Sorry to disturb with such a stupid mistake 😆 
Keep uo the good work!

closed 2020-09-12 by mtzro2003

shmuelzon, 2020-09-12:
No problem, glad it works for you now.
As you suggested, I will try to modify it so it will print where the problematic part in the JSON file is.

commit 6ae729e4 (2020-09-12): Don't hide the JSON parser's error description

Closes #76