shmuelzon/esp32-ble2mqtt #77: Can't get it worked with Switchbot

shady2k, 2020-09-14:
When I publish message to ble2mqttgw/<MAC>/Switchbot/Push/Set nothing happens. Switchbot waiting for {0x57, 0x01, 0x00} command, so I send 57,1,0 to MQTT bus. What am I doing wrong?

Here is my config:
```
{
  "wifi": {
    "ssid": "xxxxxxxx",
    "password": "xxxxxxxx"
  },
  "mqtt": {
    "server": {
      "host": "192.168.0.xx",
      "port": 1883
    },
    "publish": {
      "retain": false
    },
    "topics" :{
      "prefix": "ble2mqttgw/",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "log": {
      "host": "192.168.0.xx",
      "port": 5000
  },
  "ble": {
    "whitelist": [
      "d8:98:27:4c:ad:f3"
    ],
    "services": {
      "whitelist": [
        "cba20d00-224d-11e6-9fb8-0002a5d5c51b"
      ],
      "definitions": {
        "cba20d00-224d-11e6-9fb8-0002a5d5c51b": {
          "name": "Switchbot"
        }
      }
    },
    "characteristics": {
      "whitelist": [
        "cba20002-224d-11e6-9fb8-0002a5d5c51b"
      ],
      "definitions":{
        "cba20002-224d-11e6-9fb8-0002a5d5c51b":{
          "name":"Push"
        }
      }
    }
  }
}
```

<details>
<summary>Log</summary>
Toolchain path: /opt/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc</br>
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a</br>
Compiler version: 5.2.0</br>
Listening on port 5000</br>
2020-09-14 19:33:27.610106 (192.168.0.222): D (4895) MQTT_CLIENT: MQTT client_id=ESP32_7593B1</br>
2020-09-14 19:33:27.619574 (192.168.0.222): D (4905) MQTT_CLIENT: Core selection disabled</br>
2020-09-14 19:33:27.621745 (192.168.0.222): D (4905) TRANS_TCP: [sock=55],connecting to server IP:192.168.0.45,Port:1883...</br>
2020-09-14 19:33:27.623600 (192.168.0.222): D (5375) MQTT_CLIENT: Transport connected to mqtt://192.168.0.45:1883</br>
2020-09-14 19:33:27.625101 (192.168.0.222): [0;32mI (5375) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000</br>
2020-09-14 19:33:27.629306 (192.168.0.222): D (5385) MQTT_CLIENT: Connected</br>
2020-09-14 19:33:27.631944 (192.168.0.222): [0;32mI (5385) MQTT: MQTT client connected</br>
2020-09-14 19:33:27.633529 (192.168.0.222): D (5395) BLE2MQTT: Queuing event MQTT_CONNECTED</br>
2020-09-14 19:33:27.635496 (192.168.0.222): [0;32mI (5395) BLE2MQTT: Connected to MQTT, scanning for BLE devices</br>
[... 125 lines cut ...]
2020-09-14 19:33:32.200322 (192.168.0.222): D (9955) MQTT_CLIENT: deliver_publish, message_length_read=57, message_length=57</br>
2020-09-14 19:33:32.204598 (192.168.0.222): D (9955) MQTT_CLIENT: Get data len= 6, topic len=47</br>
2020-09-14 19:33:32.207901 (192.168.0.222): D (9965) MQTT: Received: ble2mqttgw/d8:98:27:4c:ad:f3/Switchbot/Push/Set => 57,1,0 (6)
</br>
2020-09-14 19:33:32.215691 (192.168.0.222): D (9975) BLE2MQTT: Got write request: ble2mqttgw/d8:98:27:4c:ad:f3/Switchbot/Push/Set, len: 6</br>
2020-09-14 19:33:32.225792 (192.168.0.222): D (9985) BLE: Enqueue: type: 1, device: d8:98:27:4c:ad:f3, char: cba20002-224d-11e6-9fb8-0002a5d5c51b, len: 3, val: 0x3ffbb538</br>
2020-09-14 19:33:32.367862 (192.168.0.222): D (10115) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:32.371971 (192.168.0.222): D (10125) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:32.576166 (192.168.0.222): D (10325) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:32.734048 (192.168.0.222): D (10495) BLE: Queue timer expired</br>
2020-09-14 19:33:32.738269 (192.168.0.222): D (10495) BLE: Dequeue: type: 1, device: d8:98:27:4c:ad:f3, char: cba20002-224d-11e6-9fb8-0002a5d5c51b, len: 3, val: 0x3ffbb538</br>
2020-09-14 19:33:32.747713 (192.168.0.222): D (10495) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:32.768645 (192.168.0.222): D (10525) BLE: Received GATTC event 4 (ESP_GATTC_WRITE_CHAR_EVT), gattc_if 1</br>
2020-09-14 19:33:32.883429 (192.168.0.222): D (10635) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:32.975590 (192.168.0.222): D (10725) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:33.092038 (192.168.0.222): D (10845) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:33.574231 (192.168.0.222): D (11325) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:33.601881 (192.168.0.222): D (11355) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:33.784238 (192.168.0.222): D (11535) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:34.113055 (192.168.0.222): D (11865) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:34.298406 (192.168.0.222): D (12055) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:34.898354 (192.168.0.222): D (12655) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
2020-09-14 19:33:35.223934 (192.168.0.222): D (12975) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)</br>
</code>
</details>

shmuelzon, 2020-09-14:
Hey,

The values published on the MQTT bus are decimal. So, if you want to write `0x57, 0x01, 0x00`, you should publish `87,1,0`.
I'm not familiar with the Switchbot's characteristics, so if that doesn't work, make sure you're using the correct endianess where the lowest byte is sent first. Maybe it should be `0,1,87`.

Let me know...

shady2k, 2020-09-14:
Thank you, 87,1,0 works!

closed 2020-09-14 by shady2k

shmuelzon, 2020-09-14:
Great, glad to hear it worked out.
BTW - if you believe your Switchbot configuration would be helpful for others, please consider adding it to the [Wiki](https://github.com/shmuelzon/esp32-ble2mqtt/wiki).