shmuelzon/esp32-ble2mqtt #78: Getting Started Help

rowanbradley, 2020-10-03:
I need an Ethernet MQTT to BLE gateway that will enable me to control Bluetooth (BLE) radiator valves using MQTT. I anticipate that I will need two or three of these to cover my whole house. My current thoughts are that the device does not need WiFi (it will communicate with my main house controller via wired Ethernet), and that it will be powered using micro-USB, just using micro-USB chargers plugged into the nearest mains socket. It does not need battery backup, since if the power fails everything will stop working, but when the power is restored, the main controller can re-establish contact with all the BLE devices (assuming that they have rebooted successfully).  Will esp32-ble2mqtt do this? Will it do it off the shelf, or will I need to do some development? What if some of the BLE devices send messages or require command inputs that the standard software does not translate to MQTT? How do I add such capabilities?

What is the cheapest/best hardware for this? Can I buy the hardware to do it? Or do I need to make it from an existing design? Or do I need to design something myself? 

Thanks - Rowan

shmuelzon, 2020-10-03:
Hey,

Those are a lot of questions I'll try to address but let me know if I missed anything or if you have any follow ups...
> Will esp32-ble2mqtt do this?

Yes. That's basically its only purpose.

> Will it do it off the shelf, or will I need to do some development?

Off the shelf. By default it will connect to any BLE device it discovers and expose every service and characteristic offered by the BLE peripheral over MQTT. I don't believe you'll need to develop anything, but you may want to modify the (JSON) configuration file to suit your needs, e.g. define which devices to connect to.

> What if some of the BLE devices send messages or require command inputs that the standard software does not translate to MQTT?

BLE devices have a GATT server that expose a list of services and each service has one or more characteristics. There are many [services](https://www.bluetooth.com/specifications/gatt/services/) and [characteristics](https://www.bluetooth.com/specifications/gatt/characteristics/) that were standardized by the Bluetooth SIG. These were added to this project and should be exposed in a user-friendly manner over MQTT (UUIDs are converted to names and values are published in their parsed form). Anything else is exposed over MQTT using the service's and characteristic's UUID and the values are exposed as raw byte values. Assuming you know which services and characteristics your radiator valves expose and what are the values for each characteristic, you can add those to the configuration file so they will be exposed nicely over MQTT.
You can try this out with various mobile phone apps (e.g. LightBlue) to verify you know which characteristics need to be written/read and their values.

> How do I add such capabilities?

As described above, everything will be exposed over MQTT. Custom services / characteristics can be defined in the configuration file, please see the README file.

> What is the cheapest/best hardware for this?

This project only support ESP32. As, by definition, it's a WiFi + BLE SoC, your cheapest option would be to purchase a device which incorporates it and includes USB for power and programming. You can get [those](https://www.aliexpress.com/item/32858054775.html) rather cheaply, ~4$. As you require Ethernet, that means extra hardware (and cost). I found [this](https://www.aliexpress.com/item/4001122992446.html) (~16$) as an example but can't really vouch for it. This project currently supports 3 Ethernet chips: LAN8720, TLK110 and IP101. Take into consideration that ESP-IDF v4.1 which support is currently supported on the [esp-idf_v4.1_support](https://github.com/shmuelzon/esp32-ble2mqtt/tree/esp-idf_v4.1_support) branch, supports these Ethernet chips: LAN8720, IP101, RTL8201 and DP83848. If your WiFi coverage allows it, perhaps you should start with a WiFi-only solution and if/when you feel confident about it, invest in Ethernet.

> Can I buy the hardware to do it? Or do I need to make it from an existing design? Or do I need to design something myself?

ESP32 is a fairly common device, You can find it on AliExpress, Banggood and the likes as well as Adafruit and other hacker-friendly sites. There are many variations of it with additional modules, e.g. screens, cameras, etc., but, to run this project, you just need a plain module.

Hope that answers everything, let me know...

CWGSM3V0, 2020-10-22:
I came across this project while looking for an alternative to openmqttgateway -- I have a m5stickc which is a esp32-pico. I'm just getting into these esp devices and was wondering if my unit will work with this? I am very interested in this project because it supports passkeys for BLE (which my Xiaomi BLE temp sensors use) and to tie into my home assistant project. Sorry this is out of nowhere but figured I'd ask first before trying to get it going.

shmuelzon, 2020-10-22:
Hey,

I see no reason why it wouldn’t work with the M5StickC. This project requires only an ESP32 and won’t make use of the display and other sensors you have, but should work nonetheless.
Note that it’s built on ESP-IDF directly and not on the Arduino wrappers. There are plenty of instructions on line to set that up, just make sure to use the ESP-IDF version written in the README file.

Let me know if you run into any issues.

shmuelzon, 2020-11-18:
@rowanbradley @CWGSM3V0, are there any additional questions/problems or may I close this issue?

CWGSM3V0, 2020-11-19:
I'm all good here, thanks!

rowanbradley, 2020-11-19:
Assaf,

No, the issue for me is not yet resolved. I have not got back to you 
because my computer crashed, and will not reboot, so I am concentrating 
on recovering that first. There are a number of questions that I still 
need answering:

 1. How much RAM will I need in the ESP32 system to run esp32-ble2mqtt?
 2. How much flash will I need?
 3. I am trying to use Eclipse as my IDE, but I'm not sure that I have
    the correct toolchain installed to build esp32-ble2mqtt. How do I
    check this?
 4. How do I install the toolchain if this is needed?
 5. How do I install the files that I downloaded from GitHub into
    Eclipse so that they can be built into an ESP32 application? I.e.
    what files do I need to copy, to where? And what else do need to do?
 6. Anything else that I am going to need to do to successfully build
    your application and load it into the ESP card that I have bought
    (which is the LILYGO® TTGO T-Internet-POE ESP32-WROOM LAN8720A with
    its download daughterboard.

I hope that you may be able to answer these, or to tell me what I need 
to read to get these answers.

Thank you very much - Rowan

rowanbradley, 2020-11-20:
Assaf,

No, please don't close this issue. I will need more help. My PC crashed 
and now won't boot so I am currently concentraing on recovering from 
this. Once I've got the PC running again I will ask more questions about 
building esp32-ble2mqtt. For now my questions are:

 1. How much RAM do I need in my ESP32 board to run esp32-ble2mqtt?
 2. How much flash do I need?
 3. I could not get ESP-IDF to install on my Windows 8.1 PC, so I am now
    trying to use Eclipse to build the project. I am not sure whether I
    have successfuly installed the right tool chain. How do I check this?
 4. If I have not got the right toolchain, how do Install this into my
    installation of Eclipse?
 5. How do I copy the esp32-ble2mqtt project that i have downloaded from
    Github into Eclipse, so that it will build with the right toolchain
    into an ESP32 application? I.e. which files and directories do I
    have to copy to where, and what else do have to do before trying to
    build the project?

I hope that you can answer these questions, or tell me what I need to 
read in order to get answers to these.

Many thanks for yiour help - Rowan

shmuelzon, 2020-11-23:
> 1. How much RAM do I need in my ESP32 board to run esp32-ble2mqtt?

I use the ESP-WROOM-32 modules and those come with 520KB RAM

> 2. How much flash do I need?

The project uses 4MB flash for dual-image OTA upgrades. Technically, it can fit in 2MB.

> 3 +4 If I have not got the right toolchain, how do Install this into my installation of Eclipse?
The newer ESP-IDF versions (please use v4.1) make sure all the dependencies are installed and the correct toolchain is used by running their install script.

As for Eclipse, I can't help you much as I don't use it myself. Try following [Espressif's installation instructions](https://docs.espressif.com/projects/esp-idf/en/v4.1/get-started/index.html) and verify you can compile a sample application.
If that works, then this project should compile without any issues.

shmuelzon, 2021-07-17:
Hey @rowanbradley,

I haven't heard back from you in a while so I'm closing this ticket for now.
If you're still having issues, please either reopen this ticket or create a new one.

Thanks

closed 2021-07-17 by shmuelzon