shmuelzon/esp32-ble2mqtt #79: Working

tcpipchip, 2020-11-04:
Hi, 

will it run on 2Mb flash ESP32 modules ?

If yes, i will want to test in the NINA W102

shmuelzon, 2020-11-04:
Hey,

The partition table is configured for 4MB to allow 2 applications images and 2 file systems for configuration, certificates, etc.
This is needed to support OTA firmware and configuration updates.
You might be able to modify the partition table to have only 1 of each and that would fit. You'll need to modify the Makefile to only flash the first SPIFFS partition. The rest of the code might be OK if you don't try to initiate OTA updates.

Hope that helps

tcpipchip, 2020-11-04:
Hi @shmuelzon 

How are you ?

Nice, i will try to port to W102.

How i disable to conenct to ble, i mean, only capture the broadcasting data! ?

shmuelzon, 2020-11-04:
If you don't want to actively connect to any BLE peripherals, you can configure an empty whitelist in your configuration file:
```json
{
  "ble": {
    "whitelist": []
  }
}
```

tcpipchip, 2020-11-04:
like this

{
  "ble": {
    "whitelist": []
  },
  "wifi": {
    "ssid": "Andreia Oi Miguel 2.4G",
    "password": "mcs80251"
  },
  "mqtt": {
    "server": {
      "host": "broker.hivemq.com",
      "port": 1883
    },
    "publish": {
      "retain": true
    }
  }
  
}

in config.json, right ?

shmuelzon, 2020-11-04:
Yes, that seems correct

tcpipchip, 2020-11-04:
strange that now not show more the broadcasting data of my BLE
[image]

tcpipchip, 2020-11-04:
if i remove that json

i can see my  broadcast ble,  but is tryng to connect it :(
[image]

shmuelzon, 2020-11-04:
In the first screenshot it seems like it is detected as an iBeacon. That means you should see, on the MQTT bus, the major and minor IDs, distance and such. In the second screenshot it's connecting to the device and you'll see, on the MQTT bus, the different GATT services and characteristics the device exposes.
Do you not see anything on the MQTT bus?

tcpipchip, 2020-11-04:
@shmuelzon 

First of all, thanks to help me!!!!

Let me explain better,

I have a two bles transmiting data (Broadcasting)

One is connectable and other is not connectable!

Is it possible with make this software (add option)

1) to not connect to connectable BLE
2) only take the BLE manufacturer data ?

Looks that only publishing data to MQTT for those BLE that are connectable! :(

For those not connectable, no transmiting data do MQTT, inclusive the Manufacturer Data

tcpipchip, 2020-11-04:
if not clear, i can rewrite!

shmuelzon, 2020-11-04:
In order to avoid connecting to a connectable BLE device, use the whitelist/blacklist I described below.
Regarding broadcasted data, not all type of broadcasters are published on MQTT, only the supported ones. They are defined in `main/broadcasters.c` and currently support iBeacons, Eddystone beacons, etc.
If you want to publish data from other types, you'll need to add support for it there.

Anyway, the whitelist/blacklist configuration is only for connecting to devices, it's not used for broadcasters.
The iBeacon you showed before should work regardless

tcpipchip, 2020-11-06:
Hi, can you help to change the software to publish the MANUFACTURER DATA for all ble that are in the WHITELIST ? 
And that never connects to them ?

tcpipchip, 2020-11-06:
i have here 20 ble modules that are broadcasting the MANUFACTURER DATA, wich includes the temp, humidity, etc.

shmuelzon, 2020-11-06:
Hey,

I don't know which devices you have or how the sensor readings are formatted so I can't help much.
You can take a look at the Mijia temperature + humidity sensor at: https://github.com/shmuelzon/esp32-ble2mqtt/blob/6ae729e4fec72253eafa36348ffd18e5b7a99973/main/broadcasters.c#L389
You need to implement two functions to add support for a new type of broadcasters. One that validates the specific broadcaster type and another that parses the data and returns it for publishing.
If you know the format of the manufacturer data, it should be pretty straight forward to adapt to your devices.
Keep the empty whitelist in the configuration to avoid connecting to the peripherals.

Hope that helps

tcpipchip, 2020-11-06:
Hi @shmuelzon 

Thanks for always answer my questions!

tcpipchip, 2020-11-08:
Hi, this is the info that i got
{"mac":"282c02403d9b","payload":"0201060d094566656e746f53656e736f721bff6c0202050901000005600003000102064408003d0e3500008cd2","name":"EfentoSensor","rssi":-81}

shmuelzon, 2020-11-08:
OK, that seems like the raw payload broadcasted. Note that it's a length-type-value format so, in your case:
```
Length: 02
Type: 01 -> Flags
Value: 06 (0b00000110) -> LE general discoverable mode + BR/EDR supported (is this a dual mode device?)

Length: 0d (13)
Type: 09 -> Complete local name
Value: 45 66 65 6e 74 6f 53 65 6e 73 6f 72-> EfentoSensor

Length: 1b (27)
Type: ff -> Manufacturer specific data
Value: 6c 02 02 05 09 01 00 00 05 60 00 03 00 01 02 06 44 08 00 3d 0e 35 00 00 8c d2
```
The last part is what you actually need to parse yourself as it's specific to your sensor. You can also see in the [Mijia](https://github.com/shmuelzon/esp32-ble2mqtt/blob/master/main/broadcasters.c#L328) example how to get the manufacturer specific data from the entire advertisement payload.
Do you have any documentation as to what the format is?

tcpipchip, 2020-11-08:
Good Sunday for you!!!!

That info i got from Arduino
{"mac":"282c02403d9b","payload":"0201060d094566656e746f53656e736f721bff6c0202050901000005600003000102064408003d0e3500008cd2","name":"EfentoSensor","rssi":-81}
But i aborted the use of Arduino because, that payload complete only shows in the first scan, then drop the parser data!

I forget to tell that 
{"mac":"282c02403d9b","payload":"0201060d094566656e746f53656e736f721bff6c0202050901000005600003000102064408003d0e3500008cd2","name":"EfentoSensor","rssi":-81}
i got from Arduino BLE SCAN, i aborted it because that parse that only shows in the first SCAN...
the late shows
{"time":"2020-11-08T00:04:53Z","mac":"282c02403d9b","payload":"0201060d094566656e746f53656e736f72","name":"EfSensor","rssi":-67}, cust the parser

Working on your code!

In esp-idf, which is the similar function to sprint/print on for me debug ?

shmuelzon, 2020-11-08:
> Working on your code!

Let me know if you run into any issues.

> In esp-idf, which is the similar function to sprint/print on for me debug ?

You can use the `ESP_LOGI` macros, e.g. `ESP_LOGI(TAG, "my format: %d", 5);`.
Those will show up both on the serial console and the remote logging mechanism, if you're using it.

tcpipchip, 2020-11-08:
yes, i will need your help soon!

tcpipchip, 2020-11-08:
btw, i dont need to parser data, only send to mqtt the payload

tcpipchip, 2020-11-08:
still working! no too much success

shmuelzon, 2020-11-18:
@tcpipchip, any luck? Is there anything else that's blocking you?

tcpipchip, 2020-11-18:
Hi @shmuelzon 

I implemented all on ESP32 Arduino. It´s working now! I had 3 only days to finish. Thank you so much!

shmuelzon, 2020-11-18:
OK, no problem. Closing this issue then.